nordicsemi/Android-DFU-Library #214: Android-DFU-App DFU SERVICE NOT FOUND

mschuetz93, 2020-01-27:
Hello!

I hope someone can help me out with this issue or give me a hint:
I am using the Android DFU App (https://github.com/runtimeco/Android-DFU-App) to test updating my BLE device. I start the application, go to DFU section, upload a zip-file and connect to my device. After I hit UPLOAD, the peripheral disconnects from the client and the application shows the Toast "DFU SERVICE NOT FOUND". After researching in Google, I found the nRF Connect for Mobile application. If I update the BLE device there, everything works fine. Why is it not working in the Android DFU application?

Sadly, the source code for nRF Connect is not available so I can't check whats different there. Could it be the problem that the Android DFU App is not automatically switching the BLE device to DFU mode? What else could be the issue that the DFU service is not found? If I check all services within the nRF Connect app, the service "Secure DFU Service" is available.

Thank you in advance!

Best regards,
Markus

philips77, 2020-01-27:
After an update using nRF Connect, the other app should work as well, depending on when this fork was created. I'm updating nRF Toolbox on our GitHub at the moment and a new version should be available tomorrow or on Wednesday.

mschuetz93, 2020-01-28:
What could be the problem for this to happen?
Is the new version targeting this problem or is it including other features/bug fixes?

mschuetz93, 2020-01-29:
Ok. I didn't know that there exists an "official" repository of the nRF Toolbox. 
For people running into the same issues, here is the maintained repository:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox

closed 2020-02-12 by philips77