nordicsemi/Android-DFU-Library #216: IntentService is deprecated

AimFirst, 2020-02-05:
DfuBaseService extends IntentService. However, IntentService is deprecated. There's a few known issues with them not working reliably (for example on a galaxy J7). Is there any plan to update the base service to use something newer (JobIntentService perhaps)?

philips77, 2020-02-12:
Hello,
`IntentService` is not deprecated. The `JobIntentService` has slightly different purpose, it doesn't have to be started immediately, and this is the behavior we want in DFU library. `IntentService` is just a service, that starts a thread to perform a task and stops itself when the task is complete.
There are of course some restrictions related to [background execution limits on Android 8+](https://developer.android.com/about/versions/oreo/background), but that's why the [DfuServiceInitiator](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java) has [setForeground(boolean)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/1aa9ebbc3d95890a7a96794b7f889e598acdeef4/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L147) method, which defaults to *true*. This will show a notification to the user, but that is intended. The notification can be customized by extending [updateForegroundNotification(NotificationCompat.Builder builder)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/1aa9ebbc3d95890a7a96794b7f889e598acdeef4/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1821) method in the Dfu Service implementation.

Besides, what issues are you talking about?

AndresQuiVal, 2020-05-27:
Official Android documentation tells android IntentService is deprecated on API Level 26+, refer to:
https://developer.android.com/reference/android/app/IntentService

philips77, 2020-05-27:
It's deprecated in Android R, that is API 30. The lib starts the service in foreground to avoid background execution limits problems. Usually, when user clicks "update", they want the job done now, not in some future... 
I'll think about it. In mcumgr lib, for updates of devices running firmware based on nRF Connect SDK, we provide lower level implementation, where it is the user that needs to implement the service or a job.