nordicsemi/Android-DFU-Library #217: Callback uses wrong mac address

dkijkuit, 2020-02-08:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 29
 - Bonding used: no
 - Library version: 1.9.0

**Device information (please complete the following information):**
 - Device: Oneplus 6
 - OS: Android 10

During an update the bootloader could advertise with a different macaddress incremented by 1.  All callbacks use the temporary address as a parameter. This makes it impossible to always correctly identify the original device from the calling activity. Please use the original mac address in the callbacks.

trix0, 2020-08-26:
Do we have workaround for this?

philips77, 2020-08-26:
Hello,
No, so far there is no workaround. Indeed the returned `BluetoothDevice` is the current one, so in DFU mode it may have MAC address incremented by 1, if such is being used. You may compare 5 most significant bytes to compare and skip the last one, as only the last one gets changed.

rouddy, 2021-05-25:
Hi
Does comparing 5 most significant bytes work well?
How about mac address ends with FF?

dinesharjani, 2026-08-21:
Closing since Filip provided an answer. If the issue is still important, please reopen or file as a new Issue.

closed 2026-08-21 by dinesharjani