nordicsemi/Android-DFU-Library #219: abort() DFU can take very long when not yet connected

bobatsar, 2020-02-14:
**DFU Bootloader version (please complete the following information):**
 - SDK version: SDK 15.0
 - Bonding used: no
 - Library version: 1.10.0

**Device information (please complete the following information):**
 - Device: HTC U11 Life
 - OS: Android 9

**Description**
When calling abort() if a device was not yet connected it can take up to 20 s until the service gets stopped.

abort() while uploading a firmware return immediately.

philips77, 2020-02-14:
Yes, in the current implementation the abort only is executed when the library is suppose to do something. It checks that flag and aborts. But if a connection was requested, the library just waits for a callback, either with state connected or an error, before taking any action.

bobatsar, 2020-02-26:
Is there any way, that the callback is executed earlier?

philips77, 2020-02-27:
Let's call it a feature request. Unfortunately, I have no time to implement it at the moment, so you'll have to wait. But as the priority for this is very low, I don't think we can fix it in 2020, unless I'm done with all other tasks. Sorry for that.

philips77, 2020-07-28:
Hi, I've started implementing fixing this issue. I'll release a version today.

philips77, 2020-07-28:
Released in 1.11.0. Please, test if you have time.

closed 2020-07-28 by philips77

bobatsar, 2020-08-04:
@philips77  I just did a quick test and it seems to work. Thanks for the fix.