nordicsemi/Android-DFU-Library #220: Calling 'Start' During An Existing Update Causes Problems

saadsaifse, 2020-02-18:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16] 28
 - Bonding used: [yes/no] no
 - Library version: [e.g. 1.9.1] 1.9.1

**Device information (please complete the following information):**
 - Device: [e.g. Samsung S10] Nokia TA-1024
 - OS: [e.g. Android 9] 9

**Describe the bug**
I am using this library through a Xamarin.Android project after creating a binding for it. I got it working and created a `DfuBaseService` in my app. I have noticed a strange behavior that only occurs upon sending the `Start` call to the library during an existing update in progress.

In this case, the existing update, finishes as expected but when the subsequent queued update starts, its status changes as below and then no progress event occurs.

1. Previous update finishes with `Completed` status
1. Next update starts with status `Connected` -> `Starting` -> `EnablingDFUMode` -> `Connecting` -> `Connected` -> `Starting` -> `Started` -> `Progress` (only once) -> `Disconnecting` -> (after long time) -> `Disconnected` -> `Completed`
1. The logs can be seen below:

**Logs**

 These logs also contain my custom log:

```
[0:] [18:08:51 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Progress: '99'
[0:] [18:08:52 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Progress: '100'
[0:] [18:08:52 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Disconnecting
02-18 18:08:53.137 D/BluetoothGatt(22465): onClientConnectionState() - status=19 clientIf=7 device=FD:51:D5:3E:73:DF
02-18 18:08:53.139 D/BluetoothGatt(22465): refresh() - device: FD:51:D5:3E:73:DF
02-18 18:08:53.144 D/BluetoothGatt(22465): close()
02-18 18:08:53.144 D/BluetoothGatt(22465): unregisterApp() - mClientIf=7
[0:] [18:08:54 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Disconnected
[0:] [18:08:54 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Completed
02-18 18:08:56.588 D/BluetoothGatt(22465): connect() - device: FD:51:D5:3E:73:DE, auto: false
02-18 18:08:56.588 D/BluetoothGatt(22465): registerApp()
02-18 18:08:56.588 D/BluetoothGatt(22465): registerApp() - UUID=7901a5ea-9f9e-4a6e-8fc0-fe88896f20c8[0:] [18:08:56 INF] DFU update received for device with address: FD:51:D5:3E:73:DE, Message: Connecting

02-18 18:08:56.593 D/BluetoothGatt(22465): onClientRegistered() - status=0 clientIf=7
02-18 18:08:58.344 D/BluetoothGatt(22465): onClientConnectionState() - status=0 clientIf=7 device=FD:51:D5:3E:73:DE
02-18 18:08:58.346 D/BluetoothGatt(22465): discoverServices() - device: FD:51:D5:3E:73:DE
02-18 18:08:58.924 D/BluetoothGatt(22465): onConnectionUpdated() - Device=FD:51:D5:3E:73:DE interval=6 latency=0 timeout=500 status=0
02-18 18:08:59.391 D/BluetoothGatt(22465): onSearchComplete() = Device=FD:51:D5:3E:73:DE Status=0
[0:] [18:08:59 INF] DFU update received for device with address: FD:51:D5:3E:73:DE, Message: Connected
[0:] [18:08:59 INF] DFU update received for device with address: FD:51:D5:3E:73:DE, Message: Starting
02-18 18:08:59.450 D/BluetoothGatt(22465): onConnectionUpdated() - Device=FD:51:D5:3E:73:DE interval=39 latency=0 timeout=500 status=0
02-18 18:09:00.397 D/BluetoothGatt(22465): setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
[0:] [18:09:01 INF] DFU update received for device with address: FD:51:D5:3E:73:DE, Message: Enabling DFU Mode
02-18 18:09:06.612 D/BluetoothGatt(22465): onClientConnectionState() - status=8 clientIf=7 device=FD:51:D5:3E:73:DE
02-18 18:09:06.613 D/BluetoothGatt(22465): refresh() - device: FD:51:D5:3E:73:DE
02-18 18:09:06.615 D/BluetoothGatt(22465): close()
02-18 18:09:06.616 D/BluetoothGatt(22465): unregisterApp() - mClientIf=7
02-18 18:09:06.637 D/BluetoothAdapter(22465): isLeEnabled(): ON
02-18 18:09:06.645 D/BluetoothLeScanner(22465): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
02-18 18:09:06.682 D/BluetoothAdapter(22465): isLeEnabled(): ON
[0:] [18:09:08 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Connecting
02-18 18:09:08.717 D/BluetoothGatt(22465): connect() - device: FD:51:D5:3E:73:DF, auto: false
02-18 18:09:08.717 D/BluetoothGatt(22465): registerApp()
02-18 18:09:08.717 D/BluetoothGatt(22465): registerApp() - UUID=bd427147-186d-4e2d-b3d7-8ad6c0b7f6c0
02-18 18:09:08.721 D/BluetoothGatt(22465): onClientRegistered() - status=0 clientIf=7
02-18 18:09:08.822 D/BluetoothGatt(22465): onClientConnectionState() - status=0 clientIf=7 device=FD:51:D5:3E:73:DF
02-18 18:09:08.825 D/BluetoothGatt(22465): discoverServices() - device: FD:51:D5:3E:73:DF
02-18 18:09:09.259 D/BluetoothGatt(22465): onConnectionUpdated() - Device=FD:51:D5:3E:73:DF interval=12 latency=0 timeout=600 status=0
02-18 18:09:09.464 D/BluetoothGatt(22465): onConnectionUpdated() - Device=FD:51:D5:3E:73:DF interval=6 latency=0 timeout=500 status=0
02-18 18:09:09.558 D/BluetoothGatt(22465): onSearchComplete() = Device=FD:51:D5:3E:73:DF Status=0
[0:] [18:09:09 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Connected
[0:] [18:09:09 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Starting
02-18 18:09:09.619 D/BluetoothGatt(22465): onConnectionUpdated() - Device=FD:51:D5:3E:73:DF interval=12 latency=0 timeout=600 status=0
02-18 18:09:10.564 D/BluetoothGatt(22465): configureMTU() - device: FD:51:D5:3E:73:DF mtu: 517
02-18 18:09:10.589 D/BluetoothGatt(22465): onConfigureMTU() - Device=FD:51:D5:3E:73:DF mtu=247 status=0
02-18 18:09:10.590 D/BluetoothGatt(22465): setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
[0:] [18:09:11 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Started
[0:] [18:09:11 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Progress: '0'
[0:] [18:09:12 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Disconnecting
02-18 18:11:17.934 D/BluetoothGatt(22465): onClientConnectionState() - status=8 clientIf=7 device=FD:51:D5:3E:73:DF
02-18 18:11:17.935 D/BluetoothGatt(22465): refresh() - device: FD:51:D5:3E:73:DF
02-18 18:11:17.936 D/BluetoothGatt(22465): close()
02-18 18:11:17.936 D/BluetoothGatt(22465): unregisterApp() - mClientIf=7
[0:] [18:11:19 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Disconnected
[0:] [18:11:19 INF] DFU update received for device with address: FD:51:D5:3E:73:DF, Message: Completed
```

philips77, 2020-02-20:
In your `DfuService` class (one that extends `DfuBaseService`) override `isDebug()` method and return *true* from it to get more logs.

Prathik-Jain, 2020-05-29:
@philips77 

```
2020-05-29 17:26:20.340 22781-22820/co.iotready.iotreadyscale D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(98627029) in 56 ms
2020-05-29 17:26:20.344 22781-22872/co.iotready.iotreadyscale I/DfuBaseService: Starting DFU service in foreground
2020-05-29 17:26:20.391 22781-22781/co.iotready.iotreadyscale I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='CE:4E:5E:26:46:84', name=IoTReadyDFU}, rssi=-44, timestampNanos=418778907834262, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[02, 01, 06, 03, 02, 59, FE, 0C, 09, 49, 6F, 54, 52, 65, 61, 64, 79, 44, 46, 55, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]}
2020-05-29 17:26:20.392 22781-22853/co.iotready.iotreadyscale I/RxBle#CancellableDisposable: Scan operation is requested to stop.
2020-05-29 17:26:20.412 22781-22853/co.iotready.iotreadyscale D/BluetoothAdapter: isLeEnabled(): ON
2020-05-29 17:26:20.440 22781-22872/co.iotready.iotreadyscale I/DfuBaseService: Connecting to the device...
2020-05-29 17:26:20.443 22781-22872/co.iotready.iotreadyscale D/BluetoothGatt: connect() - device: CE:4E:5E:26:46:84, auto: false
2020-05-29 17:26:20.443 22781-22872/co.iotready.iotreadyscale D/BluetoothGatt: registerApp()
2020-05-29 17:26:20.443 22781-22872/co.iotready.iotreadyscale D/BluetoothGatt: registerApp() - UUID=7df8f08d-3d9b-45d5-96df-ab803740706f
2020-05-29 17:26:20.447 22781-22793/co.iotready.iotreadyscale D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-05-29 17:26:20.588 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=CE:4E:5E:26:46:84
2020-05-29 17:26:20.589 22781-22781/co.iotready.iotreadyscale I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2020-05-29 17:26:20.588 22781-22847/co.iotready.iotreadyscale I/DfuBaseService: Connected to GATT server
2020-05-29 17:26:20.591 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: discoverServices() - device: CE:4E:5E:26:46:84
2020-05-29 17:26:20.594 22781-22847/co.iotready.iotreadyscale I/DfuBaseService: Attempting to start service discovery... succeed
2020-05-29 17:26:21.074 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=12 latency=0 timeout=600 status=0
2020-05-29 17:26:21.280 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=6 latency=0 timeout=500 status=0
2020-05-29 17:26:21.368 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onSearchComplete() = Device=CE:4E:5E:26:46:84 Status=0
2020-05-29 17:26:21.369 22781-22847/co.iotready.iotreadyscale I/DfuBaseService: Services discovered
2020-05-29 17:26:21.376 22781-22872/co.iotready.iotreadyscale W/DfuImpl: Secure DFU bootloader found
2020-05-29 17:26:21.378 22781-22781/co.iotready.iotreadyscale I/System.out: ConnectedCE:4E:5E:26:46:84
2020-05-29 17:26:21.379 22781-22781/co.iotready.iotreadyscale I/System.out: StartingCE:4E:5E:26:46:84
2020-05-29 17:26:21.421 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=12 latency=0 timeout=600 status=0
2020-05-29 17:26:22.378 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Requesting MTU = 517
2020-05-29 17:26:22.379 22781-22872/co.iotready.iotreadyscale D/BluetoothGatt: configureMTU() - device: CE:4E:5E:26:46:84 mtu: 517
2020-05-29 17:26:22.417 22781-22847/co.iotready.iotreadyscale D/BluetoothGatt: onConfigureMTU() - Device=CE:4E:5E:26:46:84 mtu=23 status=0
2020-05-29 17:26:22.418 22781-22847/co.iotready.iotreadyscale I/DfuImpl: MTU changed to: 23
2020-05-29 17:26:22.418 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Enabling notifications...
2020-05-29 17:26:22.419 22781-22872/co.iotready.iotreadyscale D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
2020-05-29 17:26:23.464 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Resume feature disabled. Performing fresh DFU
2020-05-29 17:26:23.464 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
2020-05-29 17:26:23.507 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
2020-05-29 17:26:23.510 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
2020-05-29 17:26:23.550 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 141)
2020-05-29 17:26:23.595 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending 141 bytes of init packet...
2020-05-29 17:26:23.595 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 12-8A-01-0A-44-08-01-12-40-08-01-10-02-1A-02-A7-01-20-00-28)
2020-05-29 17:26:23.604 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 00-30-00-38-F0-FE-02-42-24-08-03-12-20-3F-BA-A9-FB-03-CE-78)
2020-05-29 17:26:23.615 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 00-4E-4D-BA-E1-8F-2D-E8-71-91-66-CD-35-31-32-E8-56-5F-CA-05)
2020-05-29 17:26:23.623 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 6E-D1-EF-5A-3D-48-00-52-04-08-01-12-00-10-00-1A-40-B1-93-F0)
[... cut ...]
2020-05-29 17:26:23.858 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Setting object to Data (Op Code = 6, Type = 2)
2020-05-29 17:26:23.897 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
2020-05-29 17:26:23.902 22781-22872/co.iotready.iotreadyscale I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/12)
2020-05-29 17:26:23.903 22781-22781/co.iotready.iotreadyscale I/System.out: StartedCE:4E:5E:26:46:84
2020-05-29 17:26:23.903 22781-22781/co.iotready.iotreadyscale I/System.out: DFU0
```

philips77, 2020-06-24:
Hi, did you manage to solve your issue?

saadsaifse, 2020-06-26:
Hi, sorry, I was not able to work on that anymore. I guess the problem did not appear again so far.

closed 2020-06-26 by saadsaifse

zmoshansky, 2021-07-02:
I had a similar issue where successive DFUs would not emit progress. If updating multiple devices concurrently, ensure you call `registerProgressListener(activity, progressListener, deviceAddress)` with the `deviceAddress`. Without passing the device address, it registers as a global listener and caused problems for me.