nordicsemi/Android-DFU-Library #221: Nested private ProgressListener interface causing problems with Xamarin bindings

anthonys123, 2020-03-04:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 1.9.1]

**Device information (please complete the following information):**
 - Device: [e.g. Samsung A20]
 - OS: [e.g. Android 6]

**Your question**
Hello - I'm attempting to generate C#-based Xamarin bindings for the Android DFU and am having trouble with the binding generator accessing the private ProgressListener interface within the private class DfuProgressInfo.  I had to change the visibility of DfuProgressInfo and ProgressListener to public to create successful bindings of the two.  Unfortunately, the generator treats DfuProgressInfo as part of the package name and the Java compiler cannot locate the interface at build time.

Would it be possible to move the ProgressListener definition outside the DfuProgressInfo class?  Also, would there be any side effects of assigning public visibility to DfuProgressInfo and ProgressListener?

Thanks in advance.

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-03-05:
Hello, sure, it can be done. It looks like extracting the interface may break backwards compatibility, but the good should be easy. I'll do this in the next release.

anthonys123, 2020-03-05:
Many thanks Aleksander for the fast reply - much appreciated!

May I ask when the next release is scheduled?  As possible workarounds, I have several questions:

- I can remove specific bindings through the metadata.xml file with the Xamarin generator.  Would it be safe for me to remove the ProgressListener interface binding from the DfuProgressInfo class or to remove the DfuProgressInfo class entirely?

- If I wanted to create a custom build of the AAR prior to the official release, is it a straightforward Gradle project build?

Many thanks again for your assistance.

- Anthony

philips77, 2020-03-06:
Hello, 
I can do a release today or early next week. I have never worked with Xamarin, so I can't help you here, I'm afraid.
If you want to build it on your own, you need to comment out the following lines:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/cb39f55085c9b1d2dbf1b5c0a19ee9236dda038a/dfu/build.gradle#L28
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/cb39f55085c9b1d2dbf1b5c0a19ee9236dda038a/dfu/build.gradle#L10 <- type the version you want instead of VERSION_NAME, e.g. "1.2.3".
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/cb39f55085c9b1d2dbf1b5c0a19ee9236dda038a/build.gradle#L11-L12

The **local.properties** file should be created automatically when you import the project. In my case it just contains the following line:
```
sdk.dir=C\:\\Users\\[USERNAME]\\AppData\\Local\\Android\\sdk
```

The files you mentioned were added much earlier. From GitHub logs I see them being added in 2016, around version 0.6.3.

anthonys123, 2020-03-09:
Hi Aleksander - quick update.  I was able to get the latest version, 1.10.1, to build under Xamarin and start without the Java compiler throwing the earlier exceptions above.  I started with this project:

https://github.com/AlejandroRuiz/Xamarin.Nordic.Dfu.Android

which used version 1.7.0 and advanced one version at a time, adding necessary Xamarin dependencies when I reached 1.9.0 which was a migration to AndroidX.  I suspect I was missing specific Xamarin dependencies in my original AAR binding project.

Thanks for the guidance above for building the AAR.  I'll retry that today or tomorrow.

Quick question about the DfuServiceInitiator::Start() function.  Per your examples, it appears that the second parameter to the function should be a NotificationActivity class.  Is this parameter optional (i.e., can NULL be passed to it)?  We don't have to show update progress in our app.

- Anthony

philips77, 2020-03-09:
Then you may disable notifications by calling this method:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/cb39f55085c9b1d2dbf1b5c0a19ee9236dda038a/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L124-L134
The second parameter in `start(Context, Class<? extends DfuBaseService>)` is the actual DFU service, that extends `DfuBaseService`. You must have it in your app. Also, remember to add it to the Android Manifest. But in this Dfu service class you may safely return *null* form this method:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/cb39f55085c9b1d2dbf1b5c0a19ee9236dda038a/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1877

philips77, 2020-03-10:
Hi, do you still want the changes to be done? I created a PR for you to look at.

anthonys123, 2020-03-10:
Many thanks for creating the PR.  I think I finally was able to communicate with our firmware last night (though the firmware doesn't seem to have the DFU Service running).  I'm not yet certain whether I'll need the changes, but would these changes be worthwhile to implement in any case?  I'll take a look at the PR now.

Here is the log from the DFU library - would you agree that it's indicating our firmware is not running the DFU Service?

03-10 04:59:50.400 D/BluetoothAdapter(17510): STATE_ON
03-10 04:59:50.402 D/BluetoothGatt(17510): connect() - device: E1:32:46:D7:F3:5B, auto: false
03-10 04:59:50.402 D/BluetoothAdapter(17510): isSecureModeEnabled
03-10 04:59:50.402 D/BluetoothGatt(17510): registerApp()
03-10 04:59:50.402 D/BluetoothGatt(17510): registerApp() - UUID=e366d5f8-d7a8-4ca6-b71a-8ff00be0350b
03-10 04:59:50.409 D/BluetoothGatt(17510): onClientRegistered() - status=0 clientIf=9
03-10 04:59:50.413 D/BluetoothGatt(17510): onClientConnectionState() - status=0 clientIf=9 device=E1:32:46:D7:F3:5B
03-10 04:59:52.017 D/BluetoothGatt(17510): discoverServices() - device: E1:32:46:D7:F3:5B
03-10 04:59:52.030 D/BluetoothGatt(17510): onSearchComplete() = Device=E1:32:46:D7:F3:5B Status=0
**03-10 04:59:52.033 W/DfuBaseService(17510): DFU Service not found.**
03-10 04:59:52.033 D/BluetoothGatt(17510): cancelOpen() - device: E1:32:46:D7:F3:5B
03-10 04:59:52.042 D/BluetoothGatt(17510): onClientConnectionState() - status=0 clientIf=9 device=E1:32:46:D7:F3:5B
03-10 04:59:52.048 D/BluetoothGatt(17510): close()
03-10 04:59:52.052 D/BluetoothGatt(17510): unregisterApp() - mClientIf=9

philips77, 2020-03-10:
But did you manage to get the Xamarin binding with DFU library version 1.10.1 rom jcenter, or you had to modify the code and added it as a module?

Regarding the **DFU Service not found**, this may happen in 2 cases: 1 old services were cached (we recommend connecting with nRF Connect and selecting menu->Refresh services to clear the cache), or you device doesn't have DFU Service nor Buttonless DFU service (a service that can trigger jump to DFU bootloader mode).

anthonys123, 2020-03-10:
I had to modify the Xamarin **Metadata.xml** file used to modify the generated code per the following:

<?xml version="1.0" encoding="UTF-8"?>
<metadata>
    <attr path="/api/package[@name='no.nordicsemi.android.dfu']/class[@name='DfuBaseService']" name="extends">mono.android.app.IntentService</attr>
</metadata>

I'm not sure what you mean by "rom jcenter" above.  I was able to create a Xamarin binding assembly of the current AAR, 1.10.1, without serious warnings by using another template project from Git and have it work with our Xamarin app attempting to connect to the DFU Service.  I'm still not certain why my original binding project had the Java compiler errors locating the internalized interface, though it may have something to do with the .NET references I was using.

This being said, I may still run into issues when we get the DFU Service running on our firmware, so the changes might be needed then.

I'm hoping to make further progress with our firmware this week.  I'll let you know when I get things working and we can decide if the changes should move forward.

Many thanks for your feedback about the **DFU Service not found** message.

philips77, 2020-03-10:
I meant "from jcenter", updated my comment.
jcenter is a repository for dependencies, like maven.

anthonys123, 2020-03-11:
Hi Aleksander - we had success today updating firmware on our device with the Xamarin app on Android with version 1.10.1!  Many thanks again for all your help.

I saw the Android notifications coming through on the system tray.  Now I'm determining what would be necessary to subscribe to local broadcast events within our app.  I think this may require the changes in the PR above for the private interface.

Looking at your documentation, it indicates that I need to create a **DfuProgressListener** instance based off a **DfuProgressListenerAdapter** and then override a number of functions.  Due to the following error being logged by the Xamarin code generator, the DfuProgressListener mapping is not created, preventing me from instantiating the instance (refer to attached screenshot showing the mapping missing):

1>BINDINGSGENERATOR : warning BG8C00: For type NO.Nordicsemi.Android.Dfu.DfuBaseService, base interface no.nordicsemi.android.dfu.DfuProgressInfo.ProgressListener is invalid.

Fortunately, Xamarin is generating the **DfuProgressListenerAdapter** mapping.

I think I will need the PR changes for Xamarin to generate the DfuProgressListener mapping.  Would it be possible to spin a test build with the changes that I could try in my dev environment?

Also, it's unclear to me what I do with the instantiated DfuProgressListener?  Is that passed to the DfuServiceInitiator instance?

- Anthony

[image]

philips77, 2020-03-13:
You don't need the `DfuProgressListener`, to be honest. It's just a helper class. Normally, you should register it using [DfuProgressListenerHelper](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceListenerHelper.java) which just registers local broadcast listener, and whenever a broadcast is received, a corresponding method is called. You can just register to the same broadcasts on your own, like the helper does.

anthonys123, 2020-03-13:
Many thanks again for the guidance.  I created a class inheriting from DfuProgressListenerAdapter and IDfuProgressListener, registered an instance of it through the DfuServiceListenerHelper::RegisterProgressListener() static, and overrode the appropriate functions.  All breakpoints were hit!

philips77, 2020-03-15:
Great. Closing?

anthonys123, 2020-03-16:
Will do - thanks again.

closed 2020-03-16 by anthonys123