nordicsemi/Android-DFU-Library #223: DFU fail at error: 3 & error: 133

rickbsr, 2020-03-11:
**Information:**

Hello, I am using your product to develop an android app that for Ble-RCU ota dfu. Then, I encountered some problems and would like to ask you. 

Details as follow:

The goal of my project is to develop an app specifically for Android STB, it function is to perform OTA DFU for the Ble-RCU of STB.

Due to the demand of my products, in our code,I got the BluetoothDevice  from which is Bonded List instead of Scan.

The following is a link to my code: [Link](https://github.com/rickbsr/DFU_Sample/blob/master/app/src/main/java/com/nrf/dfu_sample/MainActivity.java)

**DFU Bootloader version:**
SDK version: 15
Bonding used: yes
Library version: 1.10.1

**Device information**
Device: Android STB (Set Top Box)
OS: Android 9, SDK 28

**Question:**
I couldn’t run DFU successfully. In fact, there are two different failures in the same code. The first is in
:Descriptor write error: 3
:Unable to set notifications state (error 3)
And the other is in
:Descriptor read error: 133
:Unable to read Service Changed CCCD (error 133)

**Log**
[Linked to error 3 Log](https://github.com/rickbsr/DFU_Sample/blob/master/app/error_log_state_1.txt)
[Linked to error 133 Log](https://github.com/rickbsr/DFU_Sample/blob/master/app/error_log_state_2.txt)

philips77, 2020-03-11:
Hello,
The error 3 means: GATT_WRITE_NOT_PERMITTED. Most probably it's a caching issue. As you are using "Buttonless service with bond sharing", the device, upon switching from app mode to bootloader mode, should send Service Change indication to indicate that services have changed. Android should invalidate services and perform fresh service discovery after connecting to the bootloader. The bond information should be retained. In our case it looks like there is an issue with that. Let me perform some tests here and come back to you.

rickbsr, 2020-03-11:
Hello,
In fact, I have tried to use nRF-Tools, it works, but it started from Scan, I tried  to trace the code, I guess, am I missing any steps in my sample, which caused it Does not work.
However, I can't find which is step that I missed. Therefore I made sample and follow on your documents, but it still couldn't work.

rickbsr, 2020-03-11:
By the way, nRF52832 is the product we used.

closed 2020-05-25 by rickbsr