nordicsemi/Android-DFU-Library #226: Doing buttonless DFU  without sharing bonding information when device is bonded

maliru, 2020-03-27:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: 16.0[e.g. SDK 16]
 - Bonding used: yes [yes/no]
 - Library version: latest[e.g. 1.9.1]

**Device information (please complete the following information):**
 - Device: set top box[e.g. Samsung A20]
 - OS: Android 9[e.g. Android 6]

**Your question**
What do you have problem with?
Customer are perform a DFU  for remote controller with set-top-box.
The RC must be bonded with set-top-box and we select buttonless without sharing bonding information.
However , when perform DFU,  the log show it is bonded so perform buttonless with sharing bonding information. The result lead the device trying to connecting the bonded address , not adderss+1. Then time-out happen and DFU  failure.
I use the same remote FW to test with nRF connect APP at android 8 cell phone.
It could be workable which APP search  address +1 for new address.

Is there any thing customer missing or something they should do at APP  side?

**Logs**
Add any other context or logs describing the problem here.
[log.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/4393272/log.txt)

philips77, 2020-03-27:
Hello,
With bonded devices you should use Buttonless service with bond sharing. If this doesn't work for you, we can try to debug it and find a solution, but switching to buttonless service without bond sharing may not be the right option. Officially, DFU for bonded devices wasn't supported for SDK 12.x and 13, so behavior when using buttonless service without bonds in this case is unknown.

From the log I can see that the service was scanning after jump, but failed without finding any. It means that the device didn't advertise neither with normal, not +1 address, as the scanner looks for any of those addresses. Or, your app does not have ACCESS_FINE_PERMISSION granted (runtime permission). In that case the scanner would not return any devices, nor return any error.

maliru, 2020-03-27:
Hi Philips,

Actually , at before , we use buttonless with sharing bond information but we got another issue.
The RC lost function after DFU. Checking at the sniffer and hci log at set-top-box. It seems that RC is working but android system did not deal with the HID report anymore. It force us to use the DFU without bonding information to test it.
I modified the RC fw and test with android mobile is working fine but not with customer's APP.
It is the feature nRF connect APP could provide but we can not do only use DFU lib.

BR
Brian

closed 2020-04-13 by maliru

maliru, 2020-05-19:
Hi Philips,
After using sharing bond information , we met too much issue at andriod TV system.
We want to use  without sharing bond information for this bonded device.
Is it possible to achieve that ? 
When  i work at andriod 8 with nrf connect app , sometime  it is not work.
APP could not find the device which is the address +1.
I saw that is a similar open case now.

BR
Brian

philips77, 2020-05-19:
In bootloader you may not advertise with direct advertising of you're not bonded (that is if you're using address +1).