nordicsemi/Android-DFU-Library #227: Disconnect during DFU cause app to crash

Strife91, 2020-04-09:
So I tested to see what happens when you turn off the bluetooth device during a device firmware upgrade and it caused my app to crash and the Bluetooth device defaults to bootloader mode(dfutarg).

That is of course not the behavior I want, I want the app to try to reconnect and redo the process.

- How do I stop the app from crashing?
- Can I reconnect to a device in bootloader mode to try and reinstall the firmware?

 - SDK version: 29
 - Bonding used: No
 - Library version: 1.10.1

**Device information (please complete the following information):**
 - Device: OnePlus 7T Pro
 - OS: Android 10

**Logs**
```
2020-04-09 10:32:49.575 7770-8213/com.app.app E/AndroidRuntime: FATAL EXCEPTION: IntentService[DfuBaseService]
    Process: com.app.app, PID: 7770
    java.lang.IllegalArgumentException: E8:BF:33:6A:88:C is not a valid Bluetooth address
        at android.bluetooth.BluetoothDevice.<init>(BluetoothDevice.java:1082)
        at android.bluetooth.BluetoothAdapter.getRemoteDevice(BluetoothAdapter.java:744)
        at no.nordicsemi.android.dfu.DfuBaseService.connect(DfuBaseService.java:1509)
        at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1288)
        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:78)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

philips77, 2020-04-09:
Does it happen every time? The stack trace says that the address is not valid (truncated). This is weird, as the exception originates in the Android stack.

1. You may connect to the bootloader and continue, but remember that you need to scan for it. First of all, the device in bl mode is using address +1. Second - you need to scan a device to connect to it if it has random address (not public).

Strife91, 2020-04-10:
> 
> 
> Does it happen every time? The stack trace says that the address is not valid (truncated). This is weird, as the exception originates in the Android stack.
> 
> You may connect to the bootloader and continue, but remember that you need to scan for it. First of all, the device in bl mode is using address +1. Second - you need to scan a device to connect to it if it has random address (not public).

Yes it crashes every time. Now that I think about it, it probably has to do with the app trying to reconnect to the bluetooth device after a dfu has "finished". I will take a closer look at it.

I assume the mac address is not incremented but simply a 1 is added to the end of the address?

Strife91, 2020-04-19:
I managed to stop it from crashing.

closed 2020-04-19 by Strife91

zwells, 2020-05-15:
How did you fix the issue? @Strife91