nordicsemi/Android-DFU-Library #228: Can't connect to device if show "DfuImpl : Scanning for new address finished with: null"

Joey780606, 2020-04-15:
**DFU Bootloader version (please complete the following information):**
 - SDK version: [SDK 15.2.0]
 - Bonding used: [no]
 - Library version: [1.9.1]

**Device information (please complete the following information):**
 - Device: [Nexus 7(M0B30x)]
 - OS: [Android 6]

I have two question about library v1.9.1.

Q1: I have tested many Android device with dfu library, like Pixel 3, Nokia 6, Asus max pro... it all seems OK to update FW.
But when I tested with Nexus 7, it often show up "DfuImpl : Scanning for new address finished with: null",
After this message, it will try to connect to MAC that doesn't +1 (See attached file, it should connect with FD:2D:EA:FA:9E:AA, but it connect to FD:2D:EA:FA:9E:A9).
This process will wait for 30 seconds, fail, retrying 3 times but still can't connect device.
Then it give me error code -133.

Can I check is this a normal work flow?  Maybe Nexus 7 is a little old (Other devices feedback is very quickly and can find device that MAC with +1).
But it's too weird so I want to know what's happen? (It will spend a lot of time to do OTA.)

Q2: I have observered if dfu operate time over 2 minutes, it sometimes will try to re-scan one times.
In my test, if re-scan can find device and update firmware successfully, or fail to udpate firmware and tell me -133 error code,
The other device I connected with the BLE would be disconnected soon.
But device can find at first time and update success, other device won't be disconnected.

Do anyone also meet similar question? 
This situation happen to many devices, not only in Nexus 7.
[Nordic_question_log.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/4479045/Nordic_question_log.txt)

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-06-24:
Reg 1.: I don't know why Nexus 7 would not discover your device when in DFU mode, but does find it in application mode. Detailed investigation would be required, but as this is a very old device, this device can just not be supported for DFU.

Reg 2.: I didn't understand this question.

philips77, 2020-07-28:
Closing due to inactivity.

closed 2020-07-28 by philips77