nordicsemi/Android-DFU-Library #230: Some problem on HuaWei Devices

zhangwm1219, 2020-04-17:
**Information**
private void DoUpdateDFU(String address,String devicename,File xfile){
        mProBar.setVisibility(View.VISIBLE);

        final DfuServiceInitiator starter = new DfuServiceInitiator(address)
                .setDeviceName(devicename)
                .setDisableNotification(true)
                .setPacketsReceiptNotificationsEnabled(true)
                .setPacketsReceiptNotificationsValue(10)
                .setPrepareDataObjectDelay(400)
                .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);//false

        filePath = xfile.getAbsolutePath();
        fileStreamUri = null;
        if(xfile.isFile() && xfile.exists()){
            if (fileType == DfuService.TYPE_AUTO) {
                starter.setZip(fileStreamUri, filePath);
            }
            starter.start(this, DfuService.class);
        }
}

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 12.x]
 - Bonding used: [yes/no]
 - Library version: [e.g. 1.10.0]

**Device information (please complete the following information):**
 - Device: [e.g. HUAWEI P30]
 - OS: [e.g. Android 10]

**Your question**
After I complete the upgrade once, I cannot continue to upgrade. I have to restart Bluetooth once

**Logs**
Add any other context or logs describing the problem here.
 W/libEGL: EGLNativeWindowType 0x705852c650 disconnect failed
 I/DfuBaseService: DFU service created. Version: 1.10.0
 I/DfuBaseService: Starting DFU service in foreground
 I/DfuBaseService: Connecting to the device...
 I/BluetoothDevice: connectGatt
 I/DfuBaseService: Connected to GATT server
I/DfuBaseService: Attempting to start service discovery... succeed
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
 I/DfuBaseService: Services discovered
 W/DfuImpl: Secure DFU bootloader found
I/DfuImpl: Requesting MTU = 517
 I/DfuImpl: MTU changed to: 23
I/DfuImpl: Enabling notifications...

philips77, 2020-06-11:
Hello, 
Sorry for the delayed response. Did you manage to solve your issue?
This looks like a caching issue. Restarting Bluetooth clears cache.

zhangwm1219, 2020-06-16:
> Hello,
> Sorry for the delayed response. Did you manage to solve your issue?
> This looks like a caching issue. Restarting Bluetooth clears cache.

Restarting Bluetooth works fine, but only succeeds once，next time I must restart bt again

philips77, 2020-06-16:
Yes, because on each connection the services are cached. In theory you should have Service Changed characteristic in Generic Attribute service to turn the caching off, but Android incorrectly either always caches (Android 4.3-7), it never caches (8-9? or so 10?). I didn't check recently, so I can't tell you precisely.

closed 2020-06-24 by philips77