nordicsemi/Android-DFU-Library #231: Best practice device stuck in dfu mode

Strife91, 2020-04-20:
Hi,

The rare scenario I'm imagining is if the app crashes or is destroyed during a dfu and the MAC-address is lost from the app.

Ideally I would like the app to scan for these devices at launch and ask the user for permission to load it with the correct firmware again.

However I have to make sure that it is our product and not any other kind of Bluetooth device. Is there a way to identify for sure that it is our product and that it's stuck in dfu mode?

I know the most common way to handle these kind of scenarios is to default to the previous firmware installation, but since we are out of storage capacity and can't fit two copies of firmware we have to find another solution.

So far I have thought of two possible solutions,

1. When I receive an onError I save the new mac-address(address+1) on the phone.  When I launch the app I scan for devices with those mac-addresses and initiate a dfu. If dfu is successful remove the mac-address from the list. The only issue with this is that it can't be done from any other phone.

2. Have some kind of recovery function in the app that scan for devices with "dfutarg" in the name and let the user confirm that they have connected to the correct device then initiate dfu. Not very intuitive and risk of user trying to load the firmware on incompatible device.

Thanks in advance.

philips77, 2020-04-22:
By default, the dfu bootloader advertise with name DfuTarg and with dfu service uuid. You may modify this to make it more unique. Remember, that you'll need to modify the default DfuPeripheralSelector in iOS, if you also have an iOS app. If you're using dfu from Sdk 14+, you may use the alternative name, the feature that iOS dfu lib is using (as Mac addresses are not available there), but in Android Mac addresses are equally good.

I would just modify the default adv name and look for that name. It's very unlikely someone will have a divorce with the same name. And mitm is always possible.

philips77, 2020-05-27:
Can we close the issue now?

closed 2020-05-27 by Strife91

tompi, 2020-05-27:
We are doing alt 2 a lot(lack of memory). We make sure the device advertises manufacturer id AND our custom "device id" both in normal operating mode and in dfu mode. Works fine and feels slightly more robust than relying on the name

Strife91, 2020-05-27:
> We are doing alt 2 a lot(lack of memory). We make sure the device advertises manufacturer id AND our custom "device id" both in normal operating mode and in dfu mode. Works fine and feels slightly more robust than relying on the name

What do you mean by custom device id? Are you talking about a custom service?

tompi, 2020-05-27:
We are sending this in the manufacturer data section of the scan response (together with some other stuff) We also use a bit there to indicate "operation mode" of the device, so we can easily identify devices stuck in dfu mode)

Strife91, 2020-05-27:
> We are sending this in the manufacturer data section of the scan response (together with some other stuff) We also use a bit there to indicate "operation mode" of the device, so we can easily identify devices stuck in dfu mode)

Ok, thanks so much for the explanation.