nordicsemi/Android-DFU-Library #234: Android 10 issue

plutocon, 2020-05-12:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: nrf51822 / nRF51_SDK_9.0.0_2e23562
 - Bonding used: no
 - Library version: 1.10.1

**Device information (please complete the following information):**
 - Device: Samsung S10
 - OS: Android 10

**Describe the bug**
DFU does not work on Android 10 devices.
After booting with the boot loader, I cannot connect the device.

The same source code works fine on the previous Android.

DFU does not work even with nrfConnect

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-05-12:
Hello,
Yes, we are aware of this issue. Have a look at this comment: 
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/368#issuecomment-619066196

Unfortunately, nothing can be done about this on the library side.

plutocon, 2020-05-13:
> Hello,
> Yes, we are aware of this issue. Have a look at this comment:
> [NordicSemiconductor/IOS-Pods-DFU-Library#368 (comment)](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/368#issuecomment-619066196)
> 
> Unfortunately, nothing can be done about this on the library side.

Thank you for answer.

Since there are products that are already being sold, I would like a solution.

philips77, 2020-05-13:
The problem is, that iOS and Android device stopped reporting direct advertising to not bonded devices to apps. Direct advertising is what was using in Legacy DFU from SDK 7 until the end when buttonless service was used. To be clear, that does not violate spec and is still allowed, but perhaps there is a security issue related to it, as we have seen similar fixes on Android and iOS made in around the same time.

The buttonless service cannot be used anymore with new mobile phones, at least not until the bootloader will be fixed. The device needs to be switched to DFU mode in some other way, like with a button (if possible). On device that have a button and supported this feature, it is usually pressing the button while toggling power on, or some other combination. For devices without a button, an old iPhone or older Android phone is needed. On Android side, I think only the latest models were updated, so majority of the phones can still be unaffected. 

An updated bootloader could be prepared and updated. A fix in the legacy DFU bootloader which turns off direct advertising requires just 1 line to be changed, one `#define`.

philips77, 2020-06-29:
Just tried this on Pixel 3 with Android 10 and nRF51 DK running HRM sample app with Buttonless Service and Legacy DFU bootloader. I was able to update the app with no problems several times in a raw. Indeed, the direct advertising packet does not show up on scanner, but when trying to connect to such device, it works immediately.
I didn't try on any other phone or Android version.

philips77, 2020-06-29:
So far it has been observed on Samsun Note9 with Android 10. All other phones I tested with work as expected with Legacy DFU. Even iPhone X with iOS 13.5.1.

philips77, 2020-07-28:
e0ab213a369982ae9cf452b55783ba0bdc5a7916 has added an option to force scanning for Legacy DFU bootloader. This requires modifying the fw on the device so that the bootloader is using address +1 and uses non-direct advertising.

sallycmp, 2020-09-16:
> The problem is, that iOS and Android device stopped reporting direct advertising to not bonded devices to apps. Direct advertising is what was using in Legacy DFU from SDK 7 until the end when buttonless service was used. To be clear, that does not violate spec and is still allowed, but perhaps there is a security issue related to it, or some newer chips rotate MAC addresses more often, as we have seen similar fixes on Android and iOS made in around the same time.
> 
> On affected phones sometimes the buttonless service cannot be used, at least not until the bootloader will be fixed. The device needs to be switched to DFU mode in some other way, like with a button (if possible). On device that have a button and supported this feature, it is usually pressing the button while toggling power on, or some other combination. For devices without a button, an old iPhone or older Android phone is needed. On Android side, I think only the latest models were updated, so majority of the phones can still be unaffected.
> 
> An updated bootloader could be prepared and updated. A fix in the legacy DFU bootloader which turns off direct advertising requires just 1 line to be changed, one `#define`. The same can be achieved by updating the app not to pass peer data to the bootloader. In this case it would behave as if started with a button, advertising undirectly using MAC+1.
> 
> In the post below I tried testing this using several phones from our library. The issue seams to be minor, at least right now. Only few phones seem to be affected and even those usually work.

Hi, what is the 1 line to be changed, one `#define`. ? where can I find that line of code? or what should I define?

closed 2023-11-20 by philips77