nordicsemi/Android-DFU-Library #235: How should I handle this error

wenhuaping, 2020-05-19:
2020-05-19 09:59:30.296 7183-14723/? E/DfuBaseService: An exception occurred while calculating file size
    java.io.IOException: The ZIP file must contain an Application, a Soft Device and/or a Bootloader.
        at no.nordicsemi.android.dfu.internal.ArchiveInputStream.<init>(Unknown Source:659)
        at no.nordicsemi.android.dfu.DfuBaseService.openInputStream(Unknown Source:15)
        at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(Unknown Source:313)
        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:76)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:176)
        at android.os.HandlerThread.run(HandlerThread.java:65)ercontent.com/9941565/82276351-9a98ee00-99b7-11ea-8605-f7393854adf7.png)

philips77, 2020-05-19:
You need to use a properly prepared zip file. Use https://github.com/NordicSemiconductor/pc-nrfutil. The zip file must contain manifest.json file. This tool will create it automatically.

wenhuaping, 2020-05-19:
> 您需要使用正确准备的zip文件。使用https://github.com/NordicSemiconductor/pc-nrfutil。压缩文件必须包含manifest.json文件。该工具将自动创建它。

wenhuaping, 2020-05-19:
> You need to use a properly prepared zip file. Use https://github.com/NordicSemiconductor/pc-nrfutil. The zip file must contain manifest.json file. This tool will create it automatically.

Thank you very much

closed 2020-05-19 by wenhuaping