nordicsemi/Android-DFU-Library #236: DFU mode, long connection time

wenhuaping, 2020-05-19:
DFU Library version: 1.10.0

1.Next interview I wrote the code：
        final boolean enablePRNs = Build.VERSION.SDK_INT < Build.VERSION_CODES.M;
        String value = String.valueOf(DfuServiceInitiator.DEFAULT_PRN_VALUE);
        int numberOfPackets;
        try {
            numberOfPackets = Integer.parseInt(value);
        } catch (final NumberFormatException e) {
            numberOfPackets = DfuServiceInitiator.DEFAULT_PRN_VALUE;
        }
        DfuServiceInitiator starter = new DfuServiceInitiator(bleAddress)
                .setDeviceName(watchName)
                .setKeepBond(false)
                .setForceDfu(false)
                .setPacketsReceiptNotificationsEnabled(enablePRNs)
                .setPacketsReceiptNotificationsValue(numberOfPackets)
                .setPrepareDataObjectDelay(400)
                .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
        starter.setZip(R.raw.lanzoom42);
        starter.start(this, DfuService.class);

2.Consulting questions
When I was using the DUF, it took a long time to connect to the firmware, about 50 seconds,I don't know what the difference is with your DEMO.

-----thanks

philips77, 2020-05-26:
The connection time does not depend on this library. It is the Android OS that connects and it does so when it receives an advertising packet from the DFU target. Perhaps your device was disabled, out of range, or has a lot of services making the service discovery very long.
Sorry, I have no idea.

closed 2020-05-26 by philips77

wenhuaping, 2020-05-27:
> The connection time does not depend on this library. It is the Android OS that connects and it does so when it receives an advertising packet from the DFU target. Perhaps your device was disabled, out of range, or has a lot of services making the service discovery very long.
> Sorry, I have no idea.

Thank you very much for your reply