nordicsemi/Android-DFU-Library #239: DFU stuck at Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/12)

Prathik-Jain, 2020-05-29:
**DFU Bootloader version (please complete the following information):**
 - SDK version: SDK 15
 - Bonding used: no
 - Library version: 1.10.0

**Device information (please complete the following information):**
 - Device: Redmi Note 5 / Poco F1
 - OS: Android 8.1.0 / Android 10

**Describe the bug**
Once the DFU process starts, it gets stuck at `Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/12)`. Works fine with nRF connect app. 

**Logs**
```
2020-05-29 15:27:49.646 15460-15675/co.iotready.iotreadyscale I/RxBle#QueueOperation: Scan operation is requested to start.
2020-05-29 15:27:49.647 15460-15675/co.iotready.iotreadyscale D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
2020-05-29 15:27:49.650 15460-15675/co.iotready.iotreadyscale D/BluetoothAdapter: isLeEnabled(): ON
2020-05-29 15:27:49.658 15460-15704/co.iotready.iotreadyscale I/DfuBaseService: Starting DFU service in foreground
2020-05-29 15:27:49.659 15460-15473/co.iotready.iotreadyscale D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2020-05-29 15:27:49.701 15460-15507/co.iotready.iotreadyscale D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(144615550) in 57 ms
2020-05-29 15:27:49.739 15460-15704/co.iotready.iotreadyscale I/DfuBaseService: Connecting to the device...
2020-05-29 15:27:49.740 15460-15704/co.iotready.iotreadyscale D/BluetoothGatt: connect() - device: CE:4E:5E:26:46:84, auto: false
2020-05-29 15:27:49.740 15460-15704/co.iotready.iotreadyscale D/BluetoothGatt: registerApp()
2020-05-29 15:27:49.740 15460-15704/co.iotready.iotreadyscale D/BluetoothGatt: registerApp() - UUID=8f8b5ea0-bbe8-4791-958f-b0dc06e1f190
2020-05-29 15:27:49.743 15460-15473/co.iotready.iotreadyscale D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-05-29 15:27:49.935 15460-15539/co.iotready.iotreadyscale D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=CE:4E:5E:26:46:84
2020-05-29 15:27:49.935 15460-15539/co.iotready.iotreadyscale I/DfuBaseService: Connected to GATT server
2020-05-29 15:27:49.936 15460-15539/co.iotready.iotreadyscale D/BluetoothGatt: discoverServices() - device: CE:4E:5E:26:46:84
2020-05-29 15:27:49.938 15460-15460/co.iotready.iotreadyscale I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2020-05-29 15:27:49.940 15460-15539/co.iotready.iotreadyscale I/DfuBaseService: Attempting to start service discovery... succeed
2020-05-29 15:27:49.945 15460-15460/co.iotready.iotreadyscale I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='CE:4E:5E:26:46:84', name=IoTReadyDFU}, rssi=-50, timestampNanos=411668463533267, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[02, 01, 06, 03, 02, 59, FE, 0C, 09, 49, 6F, 54, 52, 65, 61, 64, 79, 44, 46, 55, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]}
2020-05-29 15:27:49.947 15460-15675/co.iotready.iotreadyscale I/RxBle#CancellableDisposable: Scan operation is requested to stop.
2020-05-29 15:27:49.951 15460-15675/co.iotready.iotreadyscale D/BluetoothAdapter: isLeEnabled(): ON
2020-05-29 15:27:50.369 15460-15707/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=12 latency=0 timeout=600 status=0
2020-05-29 15:27:50.576 15460-15707/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=6 latency=0 timeout=500 status=0
2020-05-29 15:27:50.670 15460-15707/co.iotready.iotreadyscale D/BluetoothGatt: onSearchComplete() = Device=CE:4E:5E:26:46:84 Status=0
2020-05-29 15:27:50.670 15460-15707/co.iotready.iotreadyscale I/DfuBaseService: Services discovered
2020-05-29 15:27:50.677 15460-15704/co.iotready.iotreadyscale W/DfuImpl: Secure DFU bootloader found
2020-05-29 15:27:50.680 15460-15460/co.iotready.iotreadyscale I/System.out: ConnectedCE:4E:5E:26:46:84
2020-05-29 15:27:50.680 15460-15460/co.iotready.iotreadyscale I/System.out: StartingCE:4E:5E:26:46:84
2020-05-29 15:27:50.720 15460-15707/co.iotready.iotreadyscale D/BluetoothGatt: onConnectionUpdated() - Device=CE:4E:5E:26:46:84 interval=12 latency=0 timeout=600 status=0
2020-05-29 15:27:51.680 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Requesting MTU = 517
2020-05-29 15:27:51.681 15460-15704/co.iotready.iotreadyscale D/BluetoothGatt: configureMTU() - device: CE:4E:5E:26:46:84 mtu: 517
2020-05-29 15:27:51.718 15460-15707/co.iotready.iotreadyscale D/BluetoothGatt: onConfigureMTU() - Device=CE:4E:5E:26:46:84 mtu=23 status=0
2020-05-29 15:27:51.719 15460-15707/co.iotready.iotreadyscale I/DfuImpl: MTU changed to: 23
2020-05-29 15:27:51.719 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Enabling notifications...
2020-05-29 15:27:51.721 15460-15704/co.iotready.iotreadyscale D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
2020-05-29 15:27:52.751 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Resume feature disabled. Performing fresh DFU
2020-05-29 15:27:52.751 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
2020-05-29 15:27:52.785 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
2020-05-29 15:27:52.786 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
2020-05-29 15:27:52.816 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 141)
2020-05-29 15:27:52.846 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending 141 bytes of init packet...
2020-05-29 15:27:52.846 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 12-8A-01-0A-44-08-01-12-40-08-01-10-02-1A-02-A7-01-20-00-28)
2020-05-29 15:27:52.853 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 00-30-00-38-F0-FE-02-42-24-08-03-12-20-3F-BA-A9-FB-03-CE-78)
2020-05-29 15:27:52.859 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 00-4E-4D-BA-E1-8F-2D-E8-71-91-66-CD-35-31-32-E8-56-5F-CA-05)
2020-05-29 15:27:52.865 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 6E-D1-EF-5A-3D-48-00-52-04-08-01-12-00-10-00-1A-40-B1-93-F0)
2020-05-29 15:27:52.874 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = D2-9D-5D-54-FD-43-6A-3B-26-03-BE-D3-6A-3C-14-3C-3A-B4-A1-01)
2020-05-29 15:27:52.881 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 95-4B-4B-33-FA-6A-51-9A-89-E4-9C-82-D5-CA-27-FB-9E-E8-A3-F0)
2020-05-29 15:27:52.887 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 6D-5C-79-99-39-D3-B1-FE-F2-89-62-8B-78-BD-34-D0-EB-D8-B6-A1)
2020-05-29 15:27:52.893 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending init packet (Value = 88)
2020-05-29 15:27:52.902 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
2020-05-29 15:27:52.956 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Checksum received (Offset = 141, CRC = 30ADC39F)
2020-05-29 15:27:52.957 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Executing init packet (Op Code = 4)
2020-05-29 15:27:53.146 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Setting object to Data (Op Code = 6, Type = 2)
2020-05-29 15:27:53.178 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
2020-05-29 15:27:53.182 15460-15704/co.iotready.iotreadyscale I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/12)
2020-05-29 15:27:53.183 15460-15460/co.iotready.iotreadyscale I/System.out: StartedCE:4E:5E:26:46:84
2020-05-29 15:27:53.183 15460-15460/co.iotready.iotreadyscale I/System.out: DFU0
2020-05-29 15:28:13.284 15460-15465/co.iotready.iotreadyscale I/zygote64: Do partial code cache collection, code=111KB, data=78KB
[... 7 lines cut ...]
2020-05-29 15:29:58.845 15460-15704/co.iotready.iotreadyscale E/DfuBaseService: Uploading Firmware Image failed: device disconnected
2020-05-29 15:29:58.845 15460-15704/co.iotready.iotreadyscale I/DfuBaseService: Cleaning up...
```

Code: 
```java
 final DfuServiceInitiator starter = new  DfuServiceInitiator("CE:4E:5E:26:46:84")
                .setNumberOfRetries(3)
[... cut ...]
        starter.setForceDfu(true);
        starter.setDisableNotification(true);
        starter.setZip(otaFile.getAbsolutePath());

        starter.start(context, DfuService.class);

        DfuServiceListenerHelper.registerProgressListener(context, dfuProgressListener);
        // TODO: unregister after connection done / error.

        // Show notification on Oreo or above.
        if (Build.VERSION.SDK_INT > Build.VERSION_CODES.O) {
            DfuServiceInitiator.createDfuNotificationChannel(context);
        }
```

philips77, 2020-05-29:
Did you try updating the lib to 1.10.3?

philips77, 2020-05-29:
You may also set the 
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/2f61019d6da350ce3e577b5115e82b727b1c7112/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L206
to 300-400 ms.

Prathik-Jain, 2020-05-30:
Thank you @philips77 , Updating to the 1.10.3 and setting the `setPreapareDataObjectDelay`property worked. :tada:

closed 2020-05-30 by Prathik-Jain