nordicsemi/Android-DFU-Library #240: how to use Dfu ?

Serobegi, 2020-06-03:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 1.9.1]

**Device information (please complete the following information):**
 - Device: [e.g. Samsung A9]
 - OS: [e.g. Android 9]

**Your question**
What do you have problem with?

Hi. I am a beginner Android developer.
I recently want to do OTA with a smart phone using nordic nRF52810.
Of course, it was implemented with SmartPhone (Client) and nRF52810 (Server).

I tried to write with reference to "https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation".

However, the nRF Toolbox or nRF Beacon example is too difficult compared to the ReadMe.txt example.

I don't know what to see and apply. Could you please let me know if there is an easy example in short form?

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-06-03:
To do DFU you need to:
1. Scan for your devices - check out [nRF Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky)
2. Start DFU Service -> follow the [documentation](https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation#usage)

Starting the DFU service is very easy. You need to set up the dependency before:
1. Add dfu dependency in *build.gradle*,
2. Create `DfuService` class, like in the documentation. Add it to your *AndroidManifest.xml* if not added automatically.
3. If you want to display a notification with progress bar, create `NotificationActivity`, like in the documentation. Clicking the notification will start the given activity, which should create the correct activity. You may copy-paste the sample from the documentation, I think.
4. Use `DfuServiceInitializer` to start DFU. There's a bunch of config methods in there, all are documented in code.

Unfortunately we don't have any super basic example at the moment, we can think of adding a sample app to this repository, but that will have to wait.

Serobegi, 2020-06-03:
Thank you. I don't know how long it will take, but I will try again. I want to say thank you when I come back.

Serobegi, 2020-06-04:
Hi, I have a few questions.
1. What does MyActivity mean? -Main? or DFU Activity?

            final Intent intent = new Intent(this, **MyActivity**.class);
            intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
            intent.putExtras(getIntent().getExtras()); // copy all extras
            startActivity(intent);

2. Is Filehelper necessary to use DFU?

3. Is Uri essential for Start the DFU service?
Can I enter null as a parameter?

philips77, 2020-06-04:
This is the Activity you want to be open when used clicks on the notification.

philips77, 2020-06-11:
Hello, did you manage to solve your issue?

Serobegi, 2020-06-15:
^^...no! Not yet! I'm a little busy doing something else, so I'm trying again! I'm sorry to see you late for contact!

+ Google translator translated it.

philips77, 2020-06-24:
I'm closing the issue for now. If you still have questions, please ask them on [DevZone](https://devzone.nordicsemi.com/) or create another issue.

closed 2020-06-24 by philips77