nordicsemi/Android-DFU-Library #245: Device returned error after sending init packet (error 6): OPERATION FAILED

nathanlin109, 2020-06-11:
I'm having trouble implementing DFU upgrade for my bluetooth device. I was wondering if I'm following the steps correctly to update my device. My steps for trying to update the device is listed below:
1. Scan for bluetooth device
2. Connect to bluetooth device
3. Start the DfuServiceInitiator (using code given in https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation)
4. Bluetooth device disconnects, and DfuTarg becomes discoverable
5. Connect to DfuTarg
6. Start the DfuServiceInitiator again with DfuTarg
7. Disconnect from DfuTarg with this error : Device returned error after sending init packet (error 6): OPERATION FAILED

philips77, 2020-06-12:
You don't have to start initiator for the second time. The service will restart itself and will proceed with the update. Also, no need to connect to the devices, as service will connect its own client (BluetoothGatt).

The steps should be as follows:
1. Scan for Bluetooth LE device
2. Start DFU using `DfuServiceInitiator`
3. Wait for complete or fail callbacks

nathanlin109, 2020-06-13:
I've followed the 3 steps you provided by just using DfuServiceInitiator with my device, and I get this error:
E/DfuBaseService: Connection state change error: 133 newState: 0
E/DfuBaseService: An error occurred while connecting to the device:133

(My device disconnects at this point and DfuTarg becomes discoverable)

nathanlin109, 2020-06-13:
I figured out my issue, I created the zip file incorrectly.

nathanlin109, 2020-06-16:
I fixed the zip file, and it worked with nrfConnect, but when I try to update my device with the 3 steps you listed, I get this error:

E/DfuBaseService: Connection state change error: 133 newState: 0
E/DfuBaseService: An error occurred while connecting to the device:133

I forgot to include some information before. I have listed them below.
**DFU Bootloader version:**
 - SDK version: 11.0
 - Bonding used: no
 - Library version: 1.10.3

**Device information:**
 - Device: Samsung Galaxy S9
 - OS: Android 10

**Logs**
2020-06-15 16:51:31.691 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:31.691 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133
2020-06-15 16:51:31.693 8388-8768/org.aihealth.ineck D/BluetoothGatt: refresh() - device: F2:DB:E7:CB:5D:08
2020-06-15 16:51:31.694 8388-8768/org.aihealth.ineck D/BluetoothGatt: close()
2020-06-15 16:51:31.696 8388-8768/org.aihealth.ineck D/BluetoothGatt: unregisterApp() - mClientIf=12
2020-06-15 16:51:33.725 8388-8768/org.aihealth.ineck D/BluetoothAdapter: STATE_ON
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: connect() - device: F2:DB:E7:CB:5D:08, auto: false
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothAdapter: isSecureModeEnabled
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp()
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp() - UUID=c6fa3279-14ae-4dd7-a48a-dc5e8718464c
2020-06-15 16:51:33.732 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2020-06-15 16:51:38.752 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=12 device=F2:DB:E7:CB:5D:08
2020-06-15 16:51:38.760 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:38.761 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133
2020-06-15 16:51:38.763 8388-8768/org.aihealth.ineck D/BluetoothGatt: refresh() - device: F2:DB:E7:CB:5D:08
2020-06-15 16:51:38.765 8388-8768/org.aihealth.ineck D/BluetoothGatt: close()
2020-06-15 16:51:38.771 8388-8768/org.aihealth.ineck D/BluetoothGatt: unregisterApp() - mClientIf=12
2020-06-15 16:51:40.807 8388-8768/org.aihealth.ineck D/BluetoothAdapter: STATE_ON
2020-06-15 16:51:40.812 8388-8768/org.aihealth.ineck D/BluetoothGatt: connect() - device: F2:DB:E7:CB:5D:08, auto: false
2020-06-15 16:51:40.813 8388-8768/org.aihealth.ineck D/BluetoothAdapter: isSecureModeEnabled
2020-06-15 16:51:40.814 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp()
2020-06-15 16:51:40.815 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp() - UUID=5bc7b4fe-c9ea-4fd4-a2ed-c8ac47739cbe
2020-06-15 16:51:40.822 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2020-06-15 16:51:45.851 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=12 device=F2:DB:E7:CB:5D:08
2020-06-15 16:51:45.856 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:45.856 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133

closed 2020-06-18 by nathanlin109