nordicsemi/Android-DFU-Library #248: DFU Failure If Phone Screen Is Switched Off

JongChern, 2020-06-15:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
-1.7.0
-bonding used

**Device information (please complete the following information):**
- POCO F1, Nokia 6.1, Nokia 2.3
 - Android

**Your question**
Are there any known issues with using the DFU with the phone screen switched off. We've found that the DFU outright fails if we initiate it with the phone screen switched off. The error msg being DFU DEVICE DISCONNECTED.

We are using an older version of the the library 1.7.0 (ported to Xamarin) I've gone through the release logs, and dont see any new fixes which relates to this. I am not sure if this is a limitation or a bug.

**Logs**
DFU DEVICE DISCONNECTED msg received when attempting DFU with phone screen switched off

philips77, 2020-06-15:
No, I'm not aware of any issues related to this. The only thing that comes to my mind is background execution limits, but this triggers after 5 minutes, and not when the service is in background, and location permission granted only when app is running (Android 10+), which could prevent from scanning for the bootloader.

JongChern, 2020-06-24:
thanks for getting back to us so quickly. We've been using a partial wake lock (AcquireCausesWakeup ~ causing the screen to come on) to get the dfu to work, it is not ideal. out of interest is there a test case for this specific scenario?

philips77, 2020-06-24:
No, we didn't test this scenario.

yamidragut, 2020-10-07:
**DFU Bootloader version:**
 - SDK version: 
 - Bonding used: yes
 - Library version: 1.11.0

**Device information:**
 - Device: Aquaris X2
 - OS: Android 10

Hi!

I am having this same problem in Android 10, the DFU enters in a loop when the screen is switched off trying to reconnect to the BLE device after setting it in the DFU mode.
I have fixed the problem of the location permission granted only when app is running having the `setForeground(true)` in the `DfuServiceInitiator` and in the DFU's service in the manifest with the `android:foregroundServiceType="connectedDevice|location"`
This way the DFU can continue in background perfectly. 

The problem comes when switching the screen off in android 10 (probably since android 8). It isn't able to start the BLE scan for finding the BLE device after setting it in the DFU mode. In the system's logs of the phone I can see this error when trying to start the BLE scan in background: 
```
D/BtGatt.ScanManager: handling starting scan
W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 6
```
In quite a lot devices you need now to set a not empty filter in the BLE scan to be able to start it when the screen is off. In fact I was using your [Android-Scanner-Compat-Library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library) and is a problem I had also. You can read more about it [here](https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0/48079800#48079800)

So, can some filter be added to the scan after setting the DFU mode so the BLE scan can be performed with the screen off? Or is another way to overcome this problem?

Thanks a lot!

philips77, 2020-10-07:
The scan filter can be set, you may add device address in it. Don't know if that works in all devices, though.
Regarding scanning, you may also add a permission for background location https://developer.android.com/about/versions/11/privacy/location#:~:text=In%20order%20to%20enable%20background,how%20to%20Request%20background%20location.

yamidragut, 2020-10-08:
Hi, 

Sorry, maybe I didn't explained myself well enough yesterday. 

The problem is when the DFU library tries to start a ble scan for reconnecting to the ble device after setting the DFU mode, scanning for the mac address + 1. If the user starts the DFU process and then switch off the screen, the android system doesn't allow the library to perform that scan because it doesn't have any not empty filter set. This is mandatory since android 8 as you can see it also in the android documentation of [the startScan() method](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.bluetooth.le.ScanCallback)):
```
Start Bluetooth LE scan. The scan results will be delivered through callback. For unfiltered scans, scanning is stopped on screen off to save power. Scanning is resumed when screen is turned on again. To avoid this, do filetered scanning by using proper ScanFilter.
```
**So, when the screen is off, the DFU library is not able to find the ble device in DFU mode because the scan is not started**. The system shows in the adb logs the error I said: 
``` 
W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 6
```
About the problem of not finding the scanner a device when it has set some `ScanFilter` it was happening in old devices such as the one in the comment of [your code](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/07bdaa50cfc5786790bf1ac589b14931de65d099/dfu/src/main/java/no/nordicsemi/android/dfu/internal/scanner/BootloaderScannerLollipop.java#L90) that has android 5.1. In newer devices it should not happen, and in fact as you have seen in the android documentation, it is mandatory to set a not empty `ScanFilter` to be able to perform the scan when the screen is off. 
As this is only mandatory since android 8, in order to maintain fixed the problem with the old devices, **what I was proposing was to change the code in your DFU library in [here](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/07bdaa50cfc5786790bf1ac589b14931de65d099/dfu/src/main/java/no/nordicsemi/android/dfu/internal/scanner/BootloaderScannerLollipop.java#L90)** to something like

``` 
      List<ScanFilter> filters = null;
      if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
         filters = new ArrayList<>();
         filters.add(new ScanFilter.Builder().setDeviceAddress(mDeviceAddress).build());
         filters.add(new ScanFilter.Builder().setDeviceAddress(mDeviceAddressIncremented).build());
      }
      final ScanSettings settings = new ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).build();
      scanner.startScan(filters, settings, this);
```

I can't set the address in the scan's filter because is a scan that the DFU library performs internally, so it is something you need to change in the DFU library. 

About the location permission, that isn't a problem, it is working perfectly as the DFU process is done in a foreground service, with the declared `android:foregroundServiceType="location"` in the manifest item of that service. So the system allows us to use the location in the foreground service of the DFU process, the background location is not needed.

Thanks!

philips77, 2020-10-08:
Yes, i meant that your can modify the source of the Dfu lib. I'm on leave until mid January, so can't update the lib now. If you could prepare the PR I can merge it, but I'll release next year when I'm back in the office.

yamidragut, 2020-10-09:
Ooh, ok! I will do the PR then
Thanks!

PR #281 (2020-11-25): Release v1.11.1
This version introduces the following change
 * Fixes #248 by adding a ScanFilter when scanning in order to receive scan results while the screen is turned off. This change was introduced from Android 8.1.0 onwards as mentioned [here](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/319aeae6f4ebd13678b4f77375d1804978c4a1e1)
closed 2020-11-25 by roshanrajaratnam