nordicsemi/Android-DFU-Library #249: Support inline DFU file content

andrewcharnley, 2020-06-17:
Hi,

Unless I've missed it this module supports either a zip or binary file. On ReactNative bundling assets outside of the JS sphere is problematic so for including a firmware upgrade with an app upgrade (aka Google Play etc) I've base64 encoded the zip data into a json file. Here's a script I wrote to automate the packaging a bit.

```
#!/usr/bin/bash -e

perl -i -pe 's/\b(\d+)(?=\D*$)/$1+1/e' .APP_VERSION
VERSION=$(< .APP_VERSION)
./bootloader/nrfutil-linux pkg generate --hw-version 52 --application-version $VERSION --application ./app/Output/810-release/Exe/c1.hex --sd-req 0xCD --key-file ./bootloader/private.key /tmp/dfu.zip
base64 -w 0 /tmp/dfu.zip > /tmp/dfu.64
DFU64=$(< /tmp/dfu.64)
echo "{ \"payload\":\"${DFU64}\"}" > ./reactnative/src/assets/dfu.json
```
Next, the react-native-nordic-dfu only exposes a setZip() method which can be seen here:

https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/android/src/main/java/com/pilloxa/dfu/RNNordicDfuModule.java

The intention is to modify this to pass the firmware data directly, but Android-DFU-Library and the equivalent IOS-DFU-Library will need to support it first.

philips77, 2020-06-24:
Hi,
We don't have capacity to implement the feature you are requesting in near future. Please, prepare a PR and we would gladly merge it! Sorry.

andrewcharnley, 2020-07-10:
Ok thanks - it's now on my TODO radar.

closed 2023-11-20 by philips77