nordicsemi/Android-DFU-Library #250: E/DfuBaseService: Connection state change error: 133 newState: 0

nathanlin109, 2020-06-18:
Sorry for submitting a new issue, but I encountered another issue and reopened my issue from earlier, and I'm not sure if you saw it.

I have created a working zip file to upgrade my device, and I followed these steps below:
1. Scan for Bluetooth LE device
2. Start DFU using DfuServiceInitiator
3. Wait for complete or fail callbacks

I get this error below:
E/DfuBaseService: Connection state change error: 133 newState: 0
E/DfuBaseService: An error occurred while connecting to the device:133

DFU Bootloader version:

SDK version: 11.0
Bonding used: no
Library version: 1.10.3
Device information:

Device: Samsung Galaxy S9
OS: Android 10

Logs:
2020-06-15 16:51:31.691 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:31.691 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133
2020-06-15 16:51:31.693 8388-8768/org.aihealth.ineck D/BluetoothGatt: refresh() - device: F2:DB:E7:CB:5D:08
2020-06-15 16:51:31.694 8388-8768/org.aihealth.ineck D/BluetoothGatt: close()
2020-06-15 16:51:31.696 8388-8768/org.aihealth.ineck D/BluetoothGatt: unregisterApp() - mClientIf=12
2020-06-15 16:51:33.725 8388-8768/org.aihealth.ineck D/BluetoothAdapter: STATE_ON
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: connect() - device: F2:DB:E7:CB:5D:08, auto: false
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothAdapter: isSecureModeEnabled
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp()
2020-06-15 16:51:33.728 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp() - UUID=c6fa3279-14ae-4dd7-a48a-dc5e8718464c
2020-06-15 16:51:33.732 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2020-06-15 16:51:38.752 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=12 device=F2:DB:E7:CB:5D:08
2020-06-15 16:51:38.760 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:38.761 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133
2020-06-15 16:51:38.763 8388-8768/org.aihealth.ineck D/BluetoothGatt: refresh() - device: F2:DB:E7:CB:5D:08
2020-06-15 16:51:38.765 8388-8768/org.aihealth.ineck D/BluetoothGatt: close()
2020-06-15 16:51:38.771 8388-8768/org.aihealth.ineck D/BluetoothGatt: unregisterApp() - mClientIf=12
2020-06-15 16:51:40.807 8388-8768/org.aihealth.ineck D/BluetoothAdapter: STATE_ON
2020-06-15 16:51:40.812 8388-8768/org.aihealth.ineck D/BluetoothGatt: connect() - device: F2:DB:E7:CB:5D:08, auto: false
2020-06-15 16:51:40.813 8388-8768/org.aihealth.ineck D/BluetoothAdapter: isSecureModeEnabled
2020-06-15 16:51:40.814 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp()
2020-06-15 16:51:40.815 8388-8768/org.aihealth.ineck D/BluetoothGatt: registerApp() - UUID=5bc7b4fe-c9ea-4fd4-a2ed-c8ac47739cbe
2020-06-15 16:51:40.822 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2020-06-15 16:51:45.851 8388-8423/org.aihealth.ineck D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=12 device=F2:DB:E7:CB:5D:08
2020-06-15 16:51:45.856 8388-8423/org.aihealth.ineck E/DfuBaseService: Connection state change error: 133 newState: 0
2020-06-15 16:51:45.856 8388-8768/org.aihealth.ineck E/DfuBaseService: An error occurred while connecting to the device:133

philips77, 2020-06-24:
Error 133 is very common on Android. It's a generic BLE error. From the above log I conclude that your device does not behave correctly. It replies with something that is being rejected by the phone. Did you try another phone to check it?

nathanlin109, 2020-07-02:
I got the DFU to work. I talked to my firmware team making the hardware, and they changed the MAC address of DfuTarg to be the same as the device itself. Before it was +1.

philips77, 2020-07-03:
Be careful, as services may be cached now. Make sure you have Service Changed characteristic enabled, otherwise you'll have problems on iOS (and perhaps Android future versions).

I'm closing the issue now. Feel free to open a new one if you have another issue.

closed 2020-07-03 by philips77

oliver-ep, 2020-07-09:
Hi I would like to recommend re-opening this issue. I could have created a new issue, but its the same issue without a resolve for me. @philips77  if you would rather, I can log this as a new issue.

I have the exact same issue as this - and I can conclude that it is not because of the phone. I think this is related, identical or the exact same experience that developers go through to other issues others have raised:
#245
#250 
#255

and so it's worth addressing this with an updated sample code or documentation that prevents this behaviour so people don't need to go through multiple closed issues. 

Here's what I did:

1. Downloaded the nRF Tool box app from PlayStore and confirmed DFU is working with the hardware (phone and device).
2. Cloned the nRF Tool box repo, ran it locally and confirmed DFU is working.
3. Created a new project, referenced the same version of DFU in gradle
4. Copy the implementation over from the cloned nRF Toolbox on DFU feature (Updated to Kotlin, targeting recent androids)
5. Experience the error with channel needing to be setup for recent Android (#192)
6. Fix the issue with code like below (we're using Kotlin):
```
override fun onCreate(savedInstanceState: Bundle?) {
    // abbreviated
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            DfuServiceInitiator.createDfuNotificationChannel(this)
        }
    DfuServiceListenerHelper.registerProgressListener(this, dfuProgressListener)
```
7. Experience the same bug as raised by @nathanlin109 

In my case the device goes into bootloader mode, but then the connection seems to get an error. 

I'm really lost as to how the same code as the repository, would create a difference? There must be somewhere that is configuring a different behaviour.

We're using the latest Android version, firmware and SDK. 

Here's the relevant gradle dependency 
```
implementation 'no.nordicsemi.android:log:2.2.0'
implementation 'no.nordicsemi.android.support.v18:scanner:1.4.3'
implementation 'no.nordicsemi.android:dfu:1.10.3' // The DFU Library is imported automatically from jcenter:
implementation 'no.nordicsemi.android:ble-common:2.2.0' // The BLE Common Library depends on BLE Library. It is enough to include the first one.
```

As mentioned in #255 

> Regarding error 133. This is a very common error thrown for any reason. I would recommend:
>
> Trying other phone.
> In nRF Connect app, connect to your device and select "Refresh services", or "Refresh device cache". Then try again.

This did not work for me, it's consistently happening.

here's the log from nRF toolbox clone - a successful update:

```
D/ViewRootImpl@562cad5[DfuActivity]: ViewPostIme pointer 0
D/ViewRootImpl@562cad5[DfuActivity]: ViewPostIme pointer 1
I/DfuBaseService: DFU service created. Version: 1.10.3
I/DfuBaseService: Starting DFU service in foreground
I/DfuBaseService: Connecting to the device...
D/BluetoothAdapter: STATE_ON
D/BluetoothGatt: connect() - device: E2:83:BD:2F:9B:57, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=e8a71895-ac6f-45a7-b749-be1f87d9a336
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=E2:83:BD:2F:9B:57
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
I/DfuBaseService: Connected to GATT server
D/BluetoothGatt: discoverServices() - device: E2:83:BD:2F:9B:57
I/DfuBaseService: Attempting to start service discovery... succeed
D/BluetoothGatt: onConnectionUpdated() - Device=E2:83:BD:2F:9B:57 interval=6 latency=0 timeout=500 status=0
[... 340 lines cut ...]

------
The implementation is identical to the cloned repository, with the one exception that I hard-coded the `.setPacketsReceiptNotificationsValue()` (which was derived from nRFTool box implementation).

```
val starter : DfuServiceInitiator = DfuServiceInitiator(deviceId!!)
            .setDeviceName("Nordic_")
            .setKeepBond(false)
            .setForceDfu(false)
            .setPacketsReceiptNotificationsEnabled(false)
            .setPacketsReceiptNotificationsValue(12)
            .setPrepareDataObjectDelay(400)
            .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)

starter.setZip(fileStreamUri!!) // can use file path if preferred
starter.start(this, DfuService::class.java)
```

When I run the above code, the device goes into bootloader mode (nice LED that tells me that) and, well it hangs with the above message.

If I run the nRF Toolbox clone, I get 100% success rate. I run the above code, it's a 100% failure rate so far (I've only started on this today!)

I'm using Samsung Galaxy A10, Android 10. I've tried 6 times with the refresh services strategy - and the 133 error seems to come through. 

I'll switch to nRF Toolbox and it works without a hitch! What am I doing wrong?

oliver-ep, 2020-07-10:
Hi, 

Just a follow up. 

I created a ticket, and I immediately found a resolve so I'm posting it here.

My issue, which wasn't to do with Legacy DFU or anything like that, was because I was working on an activity of a larger project - and my activity I was working on wasn't asking for **Location Permission** before the DFU process started.

The DFU Library must be using a privacy sensitive method to interact with the device after it turns the Nordic device into bootloader mode.

See more details here
https://devzone.nordicsemi.com/f/nordic-q-a/63623/android-ota-dfu-not-working-with-latest-sdk-dfu-library?ReplySortBy=CreatedDate&ReplySortOrder=Descending

philips77, 2020-07-15:
True. I should have mention this in the ReadMe. Unless bonding is used, and buttonless service with bond support, Location permission is required for scanning Bluetooth LE devices.