nordicsemi/Android-DFU-Library #255: Connection state change error: 133 newState: 0 And Could not resolve project :dfu.

Serobegi, 2020-06-30:
```
        otaStartButton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                final DfuServiceInitiator starter = new DfuServiceInitiator(SwitchDeviceMac)
                        .setDeviceName(SwitchDeviceName)
                        .setKeepBond(false)
                        .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
                        .setPrepareDataObjectDelay(300L)
                        .setDisableNotification(true);
                starter.setZip(R.raw.app_dfu_package1);
                final DfuServiceController controller = starter.start(OTA.this, DfuService.class);
            }
        });
```

```
    @Override
    protected void onResume() {
        super.onResume();
        DfuServiceListenerHelper.registerProgressListener(this, dfuProgressListener);
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        DfuServiceListenerHelper.unregisterProgressListener(this, dfuProgressListener);
    }
```

```
public class NotificationActivity extends Activity {
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        // If this activity is the root activity of the task, the app is not running
        if (isTaskRoot()) {
            // Start the app before finishing
            final Intent intent = new Intent(this, MainActivity.class);
            intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
            intent.putExtras(getIntent().getExtras()); // copy all extras
            startActivity(intent);
        }

        // Now finish, which will drop you to the activity at which you were at the top of the task stack
        finish();
    }
```

```
public class DfuService extends DfuBaseService {

    @Override
    protected Class<? extends Activity> getNotificationTarget() {
        /*
          As a target activity the NotificationActivity is returned, not the MainActivity. This is because
          the notification must create a new task:

          intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);

          when you press it. You can use NotificationActivity to check whether the new activity
          is a root activity (that means no other activity was open earlier) or that some
          other activity is already open. In the latter case the NotificationActivity will just be
          closed. The system will restore the previous activity. However, if the application has been
          closed during upload and you click the notification, a NotificationActivity will
          be launched as a root activity. It will create and start the main activity and
          terminate itself.

          This method may be used to restore the target activity in case the application
[... 86 lines cut ...]
ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve project :dfu.
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Could not resolve project :dfu.
Show Details
Affected Modules: app
```

File structure

MyAppFILE
|         
|______ androidProject
|          |_____ manifest...
|          |_____ app...
|          |_____ etc ...
|          
|______ DFULibrary
          |_____ dfu
          |_____ documentation
          |_____ gradle
          |_____ etc ...          

Is this right?

philips77, 2020-07-03:
Hello,
Sorry for the late reply.

First, your configuration issues.
You should either have:
1. Importing from JCenter
```groovy
implementation 'no.nordicsemi.android:dfu:1.10.4'
```
or

2. Direct import
```groovy
implementation project(':dfu')
```
in your *build;gradle*. In the latter case, you need to clone the DFU library to the same root folder, like that:
```
StudioProjects/
  - YourApp
     - build
     - app
         - src
     - gradle
  - DFULibrary
     - build
     - dfu
         - src
     - gradle
```
or, modify the path in:
```groovy
include ':dfu'
project(':dfu').projectDir = file('../DFULibrary/dfu')
```
so that it points to the dfu module. Note that by default the DFU is cloned to *Android-DFU-Library* folder, not to *DFULibrary*. Also, in *dfu/build.gradle* remove the line with *VERSION_NAME* and the last line, pointing to a script.

But, the first approach should work, without all the hassle.

philips77, 2020-07-03:
Regarding error 133. This is a very common error thrown for any reason. I would recommend:
1. Trying other phone.
2. In nRF Connect app, connect to your device and select "Refresh services", or "Refresh device cache". Then try again.

Most popular reasons and solutions for 133:
1. Packet collision (connect request was missed by the target device) -> retry number of times
2. Bluetooth stack issue on your phone -> restart Bluetooth / restart phone / try other phone
3. Caching issue -> this trick with nRF Connect and refreshing services
4. Your device fails to reply to connect request or negotiate connection -> fix firmware
5. Rejected connection parameters -> fix firmware
6. Bonding issue -> Make sure the bond information is present either on both sides, or none of them. Remove bonding from the phone (if there was any) and reprogram the device, making sure it was wiped out first.

Serobegi, 2020-07-05:
Hello!

Thanks for the answer. I will check 133 again. As you say it's a common error, I'll probably review the causes and solutions you suggested. I haven't tried it, but the error I've seen so far seems to be a VERSION_NAME problem. I think I can solve it. my question was long and tired of reading, your answer grateful, I want to say thank you again.

closed 2020-07-28 by philips77