nordicsemi/Android-DFU-Library #256: Error on DFU process - Android 10

carmelo-ruota, 2020-07-03:
**Device information (please complete the following information):**
 - Devices: Samsung A51 - OnePlus 8 Pro
 - OS: 10

**Describe the bug**
Hi, I've been trying to complete the DFU process through your library for a while with Android 10.
Once I got the error, I tried with the NRFConnect application but I continued getting the same error on some devices.
I've attached the log obtained by NRFConnect on two smartphones with the same Android version (10) with two different behaviours: 
Samsung Doesn't work
OnePlus Works

**Log OnePlus [OK]**
```nRF Connect, 2020-06-23
No name (FB:22:BA:4D:6A:B9)
V 14:39:11.600 Connecting to FB:22:BA:4D:6A:B9...
D 14:39:11.600 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 14:39:12.578 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 14:39:12.578 Connected to FB:22:BA:4D:6A:B9
V 14:39:12.599 Discovering services...
D 14:39:12.599 gatt.discoverServices()
I 14:39:13.029 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 14:39:13.955 [Callback] Services discovered with status: 0
I 14:39:13.955 Services discovered
V 14:39:13.996 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Serial Number String [R] (0x2A25)
- Hardware Revision String [R] (0x2A27)
- Firmware Revision String [R] (0x2A26)
Unknown Service (43ba1a00-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [N R] (43ba4000-4cf5-46d8-a887-a846d9de712d)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (43ba4001-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4002-4cf5-46d8-a887-a846d9de712d)
Unknown Service (43ba1a02-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [N R] (43ba4200-4cf5-46d8-a887-a846d9de712d)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (43ba4202-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4203-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4204-4cf5-46d8-a887-a846d9de712d)
Unknown Service (43ba1a03-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4300-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4301-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4302-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4303-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [R W] (43ba4304-4cf5-46d8-a887-a846d9de712d)
Unknown Service (43ba1a04-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [N R] (43ba4400-4cf5-46d8-a887-a846d9de712d)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (43ba4401-4cf5-46d8-a887-a846d9de712d)
- Unknown Characteristic [N R] (43ba4402-4cf5-46d8-a887-a846d9de712d)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R] (43ba4403-4cf5-46d8-a887-a846d9de712d)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (43ba4404-4cf5-46d8-a887-a846d9de712d)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Nordic UART Service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
[... 484 lines cut ...]
D 14:35:58.896 [DFU] wait(1000)
D 14:35:59.928 [DFU] wait(1000)
V 14:36:00.898 [DFU] Connecting to DFU target...
D 14:36:00.905 [DFU] gatt = device.connectGatt(autoConnect = false)
E 14:36:05.938 [DFU] Connection failed (0x85): GATT ERROR
W 14:36:05.939 [DFU] Retrying...
D 14:36:05.939 [DFU] gatt.refresh() (hidden)
D 14:36:05.940 [DFU] gatt.close()
V 14:36:05.963 [DFU] DFU service started
I 14:36:05.963 [DFU] Firmware file opened successfully
D 14:36:05.963 [DFU] wait(1000)
D 14:36:07.018 [DFU] wait(1000)
V 14:36:07.965 [DFU] Connecting to DFU target...
D 14:36:07.968 [DFU] gatt = device.connectGatt(autoConnect = false)
E 14:36:13.016 [DFU] Connection failed (0x85): GATT ERROR
D 14:36:13.019 [DFU] gatt.refresh() (hidden)
D 14:36:13.020 [DFU] gatt.close()
D 14:36:13.030 [DFU] wait(600)
D 14:36:13.642 gatt.close()
D 14:36:13.653 wait(200)
V 14:36:13.856 Connecting to FB:22:BA:4D:6A:B9...
D 14:36:13.857 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 14:36:18.903 [Callback] Connection state changed with status: 133 and new state: DISCONNECTED (0)
E 14:36:18.903 Error 133 (0x85): GATT ERROR
I 14:36:18.903 Disconnected ```

philips77, 2020-07-03:
Duplicate of #234.

philips77, 2020-07-03:
Have a look at the linked issue, a specially my comments below.

closed 2020-07-03 by philips77