nordicsemi/Android-DFU-Library #259: How does the device enter DFUmode

kdwu, 2020-07-10:
How do I use this library to put the hardware into DFU mode？

andrewcharnley, 2020-07-10:
You don't. This library is for once the device is in DFU mode.

DFU mode will automatically start if the bootloader detects an incorrect checksum, which will also happen if application is missing. If correct it jumps to the application, then you have to do it another way usually by setting a temporary register which is retained on reset, such as

 err_code = sd_power_gpregret_set(0, BOOTLOADER_DFU_START);
 NVIC_SystemReset();

Remember the flow is softdevice -> bootloader (or app if not installed) -> dfu (or app if correct checksum) -> app (set bit to jump to bootloader)

philips77, 2020-07-15:
This library will automatically switch to dfu bootloader mode during the update of you have buttonless service in place. You can't use it just for switching. The app must have location permission granted in order to scan for the device advertising in dfu mode, unless bonding is used.

closed 2020-07-28 by philips77