nordicsemi/Android-DFU-Library #262: onError:4096::DFU DEVICE DISCONNECTED::0

Lloyd0577, 2020-07-24:
some phone get error:
2020-07-24 15:30:49.475 21061-22199/cn.entertech.flowtimeble E/DfuBaseService: Connection state change error: 133 newState: 0
2020-07-24 15:30:49.475 21061-22283/cn.entertech.flowtimeble E/DfuBaseService: Device not reachable. Check if the device with address C5:6A:42:98:0A:F9 is in range, is advertising and is connectable

philips77, 2020-07-27:
Which library version are you using?
In here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/852a51fa381d28ae08a4ce19dacfef073b77981e/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1310-L1314
this message should be printed when connection took at least 25 seconds. In your case only 5 passed from calling `connect()` to getting 133 error. Do you know what happens there?
Does the library try to reconnect? Is your device advertising?

philips77, 2020-07-28:
I've released 1.11.0. Could you try this version and post logs?

Lloyd0577, 2020-07-29:
Hello ,thanks for your reply, The lib i used above was old version, then i try the new release 1.11.0,and set force scan ,but it also not work,and the error was same after i closed force scan for new mac address. Also  i tried restart bluetooth and phone ,but could not be solved.I wonder if there are some problem in sdk of softdevice or something problem in chip ,now the chip model i use is nrf52832.
**My Test Phone:**
Name：Galaxy S9
Model：SM-G9600/DS
Hardware：MP0.9A
One UI Version：2.1
Android Version：10
**The Log:**
2020-07-29 10:37:36.145 9665-9665/cn.entertech.flowtimeble I/DfuBaseService: DFU service created. Version: 1.11.0
2020-07-29 10:37:36.156 9665-10822/cn.entertech.flowtimeble I/DfuBaseService: Starting DFU service in foreground
2020-07-29 10:37:36.223 9665-10822/cn.entertech.flowtimeble I/DfuBaseService: Connecting to the device...
2020-07-29 10:37:36.224 9665-10822/cn.entertech.flowtimeble D/BluetoothAdapter: STATE_ON
2020-07-29 10:37:36.226 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: connect() - device: EA:65:34:FA:A9:0A, auto: false
2020-07-29 10:37:36.226 9665-10822/cn.entertech.flowtimeble D/BluetoothAdapter: isSecureModeEnabled
2020-07-29 10:37:36.226 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: registerApp()
2020-07-29 10:37:36.226 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: registerApp() - UUID=c3aa7cd2-aac9-4966-a31d-ad136c4aed5f
2020-07-29 10:37:36.229 9665-9708/cn.entertech.flowtimeble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2020-07-29 10:37:36.232 9665-9708/cn.entertech.flowtimeble D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=EA:65:34:FA:A9:0A
2020-07-29 10:37:36.233 9665-9708/cn.entertech.flowtimeble I/DfuBaseService: Connected to GATT server
2020-07-29 10:37:36.234 9665-9708/cn.entertech.flowtimeble D/BluetoothGatt: discoverServices() - device: EA:65:34:FA:A9:0A
2020-07-29 10:37:36.234 9665-9708/cn.entertech.flowtimeble I/DfuBaseService: Attempting to start service discovery... succeed
2020-07-29 10:37:36.235 9665-9708/cn.entertech.flowtimeble D/BluetoothGatt: onSearchComplete() = Device=EA:65:34:FA:A9:0A Status=0
2020-07-29 10:37:36.235 9665-9708/cn.entertech.flowtimeble I/DfuBaseService: Services discovered
2020-07-29 10:37:36.237 9665-9665/cn.entertech.flowtimeble D/onDfuUpdate: onDfuProcessStarting
2020-07-29 10:37:37.244 9665-10822/cn.entertech.flowtimeble I/DfuImpl: Reading DFU version number...
2020-07-29 10:37:37.282 9665-10822/cn.entertech.flowtimeble I/DfuImpl: Version number read: 0.1 -> Application with Legacy buttonless update from SDK 7.0 or newer
2020-07-29 10:37:37.284 9665-10822/cn.entertech.flowtimeble W/DfuImpl: Application with legacy buttonless update found
2020-07-29 10:37:37.285 9665-10822/cn.entertech.flowtimeble I/DfuImpl: Enabling notifications...
2020-07-29 10:37:37.285 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ff41-1212-abcd-1523-785feabcd123 enable: true
2020-07-29 10:37:38.345 9665-9665/cn.entertech.flowtimeble D/onDfuUpdate: onEnablingDfuMode
2020-07-29 10:37:38.358 9665-10822/cn.entertech.flowtimeble I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2020-07-29 10:37:38.482 9665-9875/cn.entertech.flowtimeble D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=10 device=EA:65:34:FA:A9:0A
2020-07-29 10:37:38.483 9665-9708/cn.entertech.flowtimeble D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=11 device=EA:65:34:FA:A9:0A
2020-07-29 10:37:38.488 9665-9708/cn.entertech.flowtimeble W/DfuBaseService: Target device disconnected with status: 19
2020-07-29 10:37:38.490 9665-10822/cn.entertech.flowtimeble W/ech.flowtimebl: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (greylist, reflection, allowed)
2020-07-29 10:37:38.490 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: refresh() - device: EA:65:34:FA:A9:0A
2020-07-29 10:37:38.492 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ╔════════════════════════════════════════════════════════════════════════════════════════
2020-07-29 10:37:38.493 9665-10822/cn.entertech.flowtimeble I/DfuBaseService: Refreshing result: true
2020-07-29 10:37:38.493 9665-10822/cn.entertech.flowtimeble I/DfuBaseService: Cleaning up...
2020-07-29 10:37:38.493 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: cancelOpen() - device: EA:65:34:FA:A9:0A
2020-07-29 10:37:38.497 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ║ Thread: Binder:9665_3
2020-07-29 10:37:38.497 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ╟────────────────────────────────────────────────────────────────────────────────────────
2020-07-29 10:37:38.498 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ║ DeviceActivity$addDisConnectListener$1.invoke  (DeviceActivity.kt:28)
2020-07-29 10:37:38.498 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ║    DeviceActivity$addDisConnectListener$1.invoke  (DeviceActivity.kt:67)
2020-07-29 10:37:38.498 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ╟────────────────────────────────────────────────────────────────────────────────────────
2020-07-29 10:37:38.498 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ║ connect failure
2020-07-29 10:37:38.498 9665-9875/cn.entertech.flowtimeble D/PRETTYLOGGER: ╚════════════════════════════════════════════════════════════════════════════════════════
[... cut ...]
2020-07-29 10:38:00.342 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: cancelOpen() - device: EA:65:34:FA:A9:0A
2020-07-29 10:38:00.362 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: close()
2020-07-29 10:38:00.367 9665-10822/cn.entertech.flowtimeble D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-07-29 10:38:00.973 9665-9665/cn.entertech.flowtimeble D/onDfuUpdate: onError:133::GATT ERROR::1
2020-07-29 10:38:01.015 9665-9665/cn.entertech.flowtimeble I/DfuBaseService: DFU service destroyed

philips77, 2020-07-29:
The issue you have is the same as #234. The problem is, that on some phones, mainly Samsung with recent Android versions, the direct advertising performed by the bootloader misses the phone, and in result the phone never connects. From the log above I see, that your bootloader is advertising with the same address as in the app mode, and as you use Legacy DFU, most probably (unless you change something in the fw), it's using direct advertising. For me this looks like a bug on those phones. Perhaps they use the same controller (Bluetooth chip), or the root cause is somewhere in the stack.
For you to make it work you need to:
1. Use other phone, e.g. from Pixel family.
2. Looks like your device is not bonded. In my opinion, that will work when you use bonding, as then connection to devices advertising directly should work.
3. To allow updating from Samsung phone, you need to modify your app or bootloader on the device. Either you make the app (buttonless service) not to pass peer data to the bootloader, or you modify the bootloader to always use address +1. When you modify one of them, use the new API available in 1.11.0: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/d1de47b24ca600f4273afc8ed1cb48dca45e9b4d/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L289-L310
That will make the library scan for the bootloader. I guess you already did it.

Hint how to modify the buttonless service on your device is below. You need to comment out the `dfu_app_peer_data_set(conn_handle);`:

```C
/**@brief Function for preparing the reset, disabling SoftDevice, and jumping to the bootloader.
 *
 * @param[in] conn_handle Connection handle for peer requesting to enter DFU mode.
 */
static void bootloader_start(uint16_t conn_handle)
{
    uint32_t err_code;
    uint16_t sys_serv_attr_len = sizeof(m_peer_data.sys_serv_attr);

    err_code = sd_ble_gatts_sys_attr_get(conn_handle,
                                         m_peer_data.sys_serv_attr,
                                         &sys_serv_attr_len,
                                         BLE_GATTS_SYS_ATTR_FLAG_SYS_SRVCS);
    if (err_code != NRF_SUCCESS)
    {
        // Any error at this stage means the system service attributes could not be fetched.
        // This means the service changed indication cannot be sent in DFU mode, but connection
        // is still possible to establish.
    }

    m_reset_prepare();

    err_code = sd_power_gpregret_set(BOOTLOADER_DFU_START);
    APP_ERROR_CHECK(err_code);

    err_code = sd_softdevice_disable();
    APP_ERROR_CHECK(err_code);

    err_code = sd_softdevice_vector_table_base_set(NRF_UICR->NRFFW[0]);
    APP_ERROR_CHECK(err_code);

    // Prevent application from sharing Peer Data with Bootloader
    //dfu_app_peer_data_set(conn_handle);

    NVIC_ClearPendingIRQ(SWI2_IRQn);
    interrupts_disable();
    bootloader_util_app_start(NRF_UICR->NRFFW[0]);
}
```

If you already have devices in production with the default implementation, you need to allow users to update it either with a button (when buttonless service is not used, the bootloader will advertise with address+1, so scanning for *DfuTarg* device is required. Start DFU for the new found device), or use a phone that works. When I tested it, it worked 100% on some phones and only sometimes failed on others, but I don't have Samsung S9 with me.

philips77, 2020-07-29:
Duplicate of #234

Lloyd0577, 2020-07-30:
Oh My God,I comment out this line ,and it worded. Thank you so so so  much , I love you baby!

closed 2020-07-30 by philips77

Lloyd0577, 2020-07-31:
hi, i am really sorry to bother you again. there are a new trouble, after changed the mac address in device, iphone could not connect to the bootloader because the uuid was changed, and i wonder what the new uuid is after device into bootloader, or is there any other methods to connect bootloader by iphone, thanks.

philips77, 2020-07-31:
There is this https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/374 PR, which introduces the same flag as I added recently to Android DFU Library. For now you may use this branch, I'll try to release the new version for iOS today.

Lloyd0577, 2020-08-03:
Thanks , it worked for iOS !!!

deniske, 2021-01-16:
Hi @philips77 ,

    where can i modify dfu_app_peer_data_set(conn_handle) in react-native-nordic-dfu? Because I met the same issue in react-native-nordic-dfu

philips77, 2021-01-20:
Hello,

The `dfu_app_peer_data_set(conn_handle)` is on the bootloader side, not mobile. On mobile side you need to force scanning after jumping to bootloader (see comments above). If you can't find that method in react native implementation, please create an issue [here](https://github.com/Pilloxa/react-native-nordic-dfu/issues). We do not maintain that repo.