nordicsemi/Android-DFU-Library #263: Two issues about DFU library.

Joey780606, 2020-07-27:
We find two issues in the DFU library and also find the solution.  Maybe Nordic can review and join them if they are actually useful.

1. During our testing, we find out dfu library will be hanged up if the mobile phone turn off Bluetooth during OTA.
After the mobile phone turn on Bluetooth, dfu library still can't back to normal status.
dfu will keep hanging up status unless app restarts.

After our testing, we find if we add "mLock.notifyAll()" code in the mBluetoothStateBroadcastReceiver below.
This question will be solved.

```
private final BroadcastReceiver mBluetoothStateBroadcastReceiver = new BroadcastReceiver() {
	@Override
	public void onReceive(final Context context, final Intent intent) {
		final int state = intent.getIntExtra(BluetoothAdapter.EXTRA_STATE, BluetoothAdapter.STATE_OFF);
		final int previousState = intent.getIntExtra(BluetoothAdapter.EXTRA_PREVIOUS_STATE, BluetoothAdapter.STATE_ON);
		logw("Action received: android.bluetooth.adapter.action.STATE_CHANGED [state: " + state + ", previous state: " + previousState + "]");
		if (previousState == BluetoothAdapter.STATE_ON
				&& (state == BluetoothAdapter.STATE_TURNING_OFF || state == BluetoothAdapter.STATE_OFF)) {
			sendLogBroadcast(LOG_LEVEL_WARNING, "Bluetooth adapter disabled");
			mConnectionState = STATE_DISCONNECTED;
			if (mDfuServiceImpl != null)
				mDfuServiceImpl.getGattCallback().onDisconnected();

			// Notify waiting thread
			synchronized (mLock) {
				mLock.notifyAll();
			}
		}
	}
};
```

2. In the Nordic BLE library, connectGatt command have different parameters in different android version, like below (I remove some log information):
```
	if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
		final int preferredPhy = connectRequest.getPreferredPhy();
		bluetoothGatt = device.connectGatt(context, false, gattCallback,
				BluetoothDevice.TRANSPORT_LE, preferredPhy/*, handler*/);
	} else if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
		bluetoothGatt = device.connectGatt(context, false, gattCallback,
				BluetoothDevice.TRANSPORT_LE);
	} else {
		bluetoothGatt = device.connectGatt(context, false, gattCallback);
	}
```

  But in the DFU library, connectGatt command only use one method.

```
   final BluetoothGatt gatt = device.connectGatt(this, false, mGattCallback);
```

  During out testing, some Android mobile sometimes can't connect successfully if we only use the DFU connectGatt command. (Like HTC U11).  Maybe you can review it and consider to alert these parts.

philips77, 2020-07-27:
Thank you for your contribution! I'll fix the issues soon.

yakdere, 2020-07-27:
@philips77 is there a rollout date, so we can expect to get these changes as a consumer? Thank you in advance.

philips77, 2020-07-28:
Let's roll out today.

philips77, 2020-07-28:
1.11.0 should be available for use.

philips77, 2020-07-28:
Released.

closed 2020-07-28 by philips77