nordicsemi/Android-DFU-Library #267: DFU to already connected device

trix0, 2020-08-19:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: yes
 - Library version: [e.g. 1.11.0]

**Device information (please complete the following information):**
 - Device:Lenovo TB-X505L
 - OS:  Android 9

**Your question**
Hi, I have downloaded your nRF Toolbox app for android. The way you do DFU is that you select the pacakage, select the device and then click upload. The App will connect to the device and updates it.

The flow of my App works little bit different. I have a list of devices, I connect to them and I can perform various tasks to the device over BLE. Is it possible to do the DFU update to already connected device?

philips77, 2020-08-20:
Yes it is. On Android and iOS you may have multiple clients (`BluetoothGatt`) connected to the same device. Each of them is not aware of any others, has to connect as if it was in fact establishing physical connection. They all get all the notifications sent from the device, but are unaware of any write or read requests. Also, the physical connection ends when the last one disconnects, or the connection is terminated from the peripheral side. So you may start dfu to already connected device, but expect it to disconnect at least once during that time. Done send any data during dfu.

trix0, 2020-08-20:
@philips77  Thanks :)

closed 2020-08-20 by trix0