nordicsemi/Android-DFU-Library #269: How to reduce the time of firmware upgrade using the Nordic library in Android.

KaushikDelphiansystems, 2020-09-01:
I have integrated the latest version (DFU Library Version 1.10.5 (SDK 15.0.0)) of the Nordic library into my android project but still not reduce the time of firmware upgrade as the iOS app takes less time (around 2 minutes) of firmware upgrade using the Nordic library. My firmware file size is only 242k. Let me know, if there is any solution or require any settings into the Nordic library so I can add and fix the time issue

**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 1.9.1]

**Device information (please complete the following information):**
 - Device: [e.g. Samsung S10]
 - OS: [e.g. Android 9]

**Describe the bug**
A clear and concise description of what the bug is.

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-09-01:
Hello,
We need more details. How long does it take on Android and on iOS? Latest version of the library is 1.11.0. Check your logs from Android and iOS, with timestamps and compare. Where do you see Android being slower? What mtu do you use?

philips77, 2023-11-20:
Closing due to 3 years of inactivity.

closed 2023-11-20 by philips77