nordicsemi/Android-DFU-Library #271: Android 7 DFU update cannot be performed

alphaYuto, 2020-09-10:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: 
 - Bonding used: no
 - Library version: 1.9.1

**Device information (please complete the following information):**
 - Device:
 - OS: Android7

**Your question**
Hello
I implemented a DFU update on Android 7.0, but I couldn't run DFU and got an error. The content of the error is "java.lang.NoClassDefFoundError: Failed resolution of: Landroid / app / Notification Channel". Is there a solution?
Attach the log.

**Logs**
[error_log_20200909.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/5199231/error_log_20200909.txt)

Add any other context or logs describing the problem here.

philips77, 2020-09-10:
You need to set the notification channels on Android 8 or newer only.

alphaYuto, 2020-09-10:
In Android 8 and earlier, the process of setting up a notification channel is not described.
Unless you explain the process of disabling the notification channel setting
Is the notification channel set automatically?

philips77, 2020-09-10:
Notification Channels were added in Android 8: https://developer.android.com/training/notify-user/channels
For older platforms you just don't call this method and everything should work.
Also, in your code, you set the channels for 8+, but also disable notifications, so channels are not needed, but you keep them enabled for pre-8.

roshanrajaratnam, 2020-11-26:
Closing due to inactivity. Please feel free to re-open if the issue persists.

closed 2020-11-26 by roshanrajaratnam