nordicsemi/Android-DFU-Library #272: DFU - DFU Service not found

Maartinsh, 2020-09-22:
Hi all, 
Before I tried to run my code, I tested DFU device services in nRF Connect app:

Device: Huawei P20, Samsuns S9 (Android 10s)

https://i.imgur.com/eoStwLQ.png

My code:

```
return DfuServiceInitiator(device.address)
                .setDeviceName(device.name)
                .setKeepBond(true)
                .setNumberOfRetries(5)
                .setZip(fileStreamUri, null)
```

Logcat:

```
Attempt: 1
Refreshing result: true
Cleaning up...
Restarting the service
Starting DFU service in foreground
Connecting to the device...
Connected to GATT server
Attempting to start service discovery... succeed
Action received: android.bluetooth.device.action.ACL_CONNECTED
I/DfuBaseService: Services discovered
**W/DfuBaseService: DFU Service not found.**
I/DfuBaseService: Disconnecting from the device...
I/DfuBaseService: Disconnected from GATT server
I/DfuBaseService: Refreshing result: true
I/DfuBaseService: Cleaning up...
I/DfuBaseService: Starting DFU service in foreground
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

Do I have wrong configuration of DFUService initiator object?

philips77, 2020-09-22:
Looks like old services were cashed. Try nRF Connect, connect and select "Refresh Device cache" in top-right menu or restart your Bluetooth and try again.

Maartinsh, 2020-09-22:
I connected to device and refreshed Services as you requests + turned on/off bluetooth.
Now logs show that first multiple connections failed (Connection error after: 30013 ms) and (Device not reachable. Check if the device with address E6:F6:05:7E:6A:5D is in range, is advertising and is connectable) even though device is next to me. Then after few minutes, it succeeds connecting to device, but issue still persist: [Services discovered
2020-09-22 15:35:34.710 7680-8223/com.scope.lifeDriveBLE W/DfuBaseService: DFU Service not found](url).

philips77, 2020-09-23:
Can you see the DFU service in the nRF Connect app?

Maartinsh, 2020-09-23:
> Can you see the DFU service in the nRF Connect app?

https://i.imgur.com/eoStwLQ.png

philips77, 2020-09-23:
Could you share logs from nRF Connect here?

Maartinsh, 2020-09-23:
https://i.imgur.com/qgpNckv.png
https://i.imgur.com/jM8vAq0.png
https://i.imgur.com/hTxOOrD.png
https://i.imgur.com/rpdH9nm.png
https://i.imgur.com/BnlrlDz.png

philips77, 2020-09-24:
But.. but what's not working then? It works in nRF Connect, but doesn't work in your app?

Maartinsh, 2020-09-24:
> But.. but what's not working then? It works in nRF Connect, but doesn't work in your app?

It works in nRF Connect, but it does not in my app when I use DFU-library. As you can see at the top of this discussion I posted my configuration of library and what logs I see in my app (it says it cannot find DFU Service for the file - even tho nRF Connect has no problem of uploading the very same file. I assume nRF Connect uses the same DFU-library underneath, so question is - could be configuration of DFU initializer could be wrong and causing these problems?)

philips77, 2020-09-24:
Yes, it uses the same library. Are you sure you're not using 
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/UuidHelper.java which can change UUID, so the lib would look for another ones?

Maartinsh, 2020-09-24:
Issue found. Somehow lost devices between connection and it caused me to connect to wrong device. It's working now. Thanks.

closed 2020-09-24 by Maartinsh

jazarforcite, 2021-07-27:
Hello @philips77 i'm having the same problem explained above, except that i am using the modified MAC address of the device. The same code works perfectly on all Android versions earlier than 11. I noticed that the problem is due to the fact that Android 11 does not refresh the list when the DfuServiceInitiator tries to connect. But if i open the BT settings on the phone and refresh the list and check if the device is there, it refreshes the list and then afterwards the consecutive connections work perfectly.

Have you seen this issue before? is there a known fix for it?

Any help would be appreciated.

Thanks,
Joseph

philips77, 2021-07-27:
No, I have not performed DFU tests on Android 11 or 12 yet. I will be refreshing the DFU library in coming weeks (mostly dependency and Maven Central related issues), so I'll try it.
Did you try nRF Connect for Android? Could you paste the logs from the app here?

jazarforcite, 2021-07-27:
nRF Connect works perfectly because you open it after you switch OTA Mode and you're not connected to it so it manages to find the device normally. But our app is connected to the device when you switch to OTA mode, it then disconnects and tries to find the device with the new MAC address which then fails to work. It works perfectly on any Android other than 11, we haven't tried it on 12 yet either. I'm still digging, trying to find a solution for it. i'll get some logs shortly and paste them here.

philips77, 2021-07-27:
nRF Connect can also switch to the bootloader mode, assuming you have the default implementation of the buttonless service. If you change the MAC in bootloader mode, there should be no caching issues, unless you also reuse the same address in some other situations, applications or something else. Do you have Service Changed characteristic in your app and bootloader mode? Are you using bonding?

jazarforcite, 2021-07-27:
Our device does not implement the buttonless option, you switch to OTA mode using a long button press and yes the MAC is incremented by 1 in OTA mode which works fine.
i've included below some of the log:
D/SystemManagementService: Bluetooth BroadcastReceiver: ACTION_ACL_DISCONNECTED (Controller)
D/UpdateFirmwareHBCActivity: SystemManagementService successfully bound to activity. Proceeding to flash HBC...
D/UpdateFirmwareHBCActivity: Controller MAC Address: [E1:F3:65:2C:ED:70] => [E1:F3:65:2C:ED:71] (Modified)
D/SystemManagementService: Controller stopping...
D/BLEManager: Stopping BLEManager instance...
D/BluetoothGatt: cancelOpen() - device: E1:F3:65:2C:ED:70
D/BLEManager: Stopped BLEManager instance.
D/SystemManagementService: Controller destroyed successfully.
    Controller stopped.
V/SystemManagementService: Sending event to activity => SystemEvents
D/MainActivity: Synchronising UI...
D/UpdateFirmwareHBCActivity: Controller successfully stopped. Commencing controller firmware flashing...
W/: Verification of void com.XXXX.app.XXXX.Library.FirmwareHBCManager.processFlashingState() took 138.032ms (738.95 bytecodes/s) (3744B approximate peak alloc)
D/FirmwareHBCManager: State Change: NONE => DOWNLOAD_FILE_FROM_URL
D/UpdateFirmwareHBCActivity: Step Changed to: DOWNLOAD_FILE_FROM_URL
D/FirmwareHBCManager: Downloading firmware package from URL: 
D/UpdateFirmwareHBCActivity: Progress: 100
D/FirmwareHBCManager: State Change: DOWNLOAD_FILE_FROM_URL => INITIATE_DFU_FLASHING
D/UpdateFirmwareHBCActivity: Step Changed to: INITIATE_DFU_FLASHING
D/FirmwareHBCManager: Flashing BIN file to HBC...
    Starting DFU...
D/FirmwareHBCManager: Started DFU.
I/DfuBaseService: DFU service created. Version: 1.11.0
W/DfuBaseService: Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
    Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - device: E1:F3:65:2C:ED:71, auto: false
    registerApp()
D/BluetoothGatt: registerApp() - UUID=25469dba-cecb-4610-85a0-ef9e2c744368
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=E1:F3:65:2C:ED:71
E/DfuBaseService: Connection state change error: 133 newState: 0
I/DfuBaseService: Connection error after: 30015 ms
E/DfuBaseService: Device not reachable. Check if the device with address E1:F3:65:2C:ED:71 is in range, is advertising and is connectable
I/DfuBaseService: Attempt: 1
W/: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (greylist, reflection, allowed)
D/BluetoothGatt: refresh() - device: E1:F3:65:2C:ED:71
I/DfuBaseService: Refreshing result: true
    Cleaning up...
D/BluetoothGatt: cancelOpen() - device: E1:F3:65:2C:ED:71
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=9
I/DfuBaseService: Restarting the service
W/DfuBaseService: Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
    Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - device: E1:F3:65:2C:ED:71, auto: false
    registerApp()
D/BluetoothGatt: registerApp() - UUID=885a5a86-c065-416c-9d10-5e6e174a6301
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=E1:F3:65:2C:ED:71
E/DfuBaseService: Connection state change error: 133 newState: 0
I/DfuBaseService: Connection error after: 30013 ms
E/DfuBaseService: Device not reachable. Check if the device with address E1:F3:65:2C:ED:71 is in range, is advertising and is connectable
I/DfuBaseService: Attempt: 2
D/BluetoothGatt: refresh() - device: E1:F3:65:2C:ED:71
I/DfuBaseService: Refreshing result: true
    Cleaning up...
D/BluetoothGatt: cancelOpen() - device: E1:F3:65:2C:ED:71
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=9
I/DfuBaseService: Restarting the service
W/DfuBaseService: Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
    Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - device: E1:F3:65:2C:ED:71, auto: false
    registerApp()
D/BluetoothGatt: registerApp() - UUID=49fb2926-4f14-4ee0-b76c-b53ec5480d05
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9

What you can see from the log is that the DFU service is trying to connect to the new mac address and not able to find it and eventually OTA mode times out and the device is turned off after a minute or so.
not sure i understood what you mean by service changed characteristic?

Thank you for taking the time to reply in either case.

philips77, 2021-07-27:
Is the app scanning for the device after triggering the bootloader more, or you just pass the new, calculated address to the `DFUServiceInitiator`? The library simply calls `connectGatt` on a `BluetoothDevice` instance obtained like this:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b5d587bfbfaa5b1eacebce3a8846746a9d3a9793/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1531-L1532

**Important:**
Android needs to scan for a device before connecting to it. Otherwise, as the MAC address given is only 48-bit long, the system cannot determine the "type" of the address (public or private) and because it's unlikely that one would know a random MAC address before scanning for it, it assumes public one. So then, Android scans for a device advertising with the same MAC but of public type (it's the 49th bit of address) before connecting to it. After 30 sec it times out.

**Solution:**
After switching the device to bootloader mode start LE scan until you find a device with required MAC address. Only then start DFU as you go right now. Android caches address type of all scanned devices, so after scanning connection should succeed.

**How to test:**
You may also trigger bootloader mode and switch to nRF Connect. On Scanner tab just scan for it, without connecting, go back to your app and proceed with DFU.

jazarforcite, 2021-07-27:
yep tried that, i added a step to scan for the device before connecting to it but it didn't work. i'll try it again and see what happens.

jazarforcite, 2021-07-27:
yep i guess i tried to scan for BLE devices incorrectly, i removed the code and re-wrote it from scratch and it works! thanks for the suggestion @philips77 appreciate the help!

archie94, 2023-01-03:
Hello!

I am facing an issue on some user devices which is similar to this one. On some devices the peripheral never goes into Bootloader mode as evident from the logs 

```
1672653564, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] DFU service started
1672653564, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Opening file...
1672653564, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Firmware file opened successfully
1672653564, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
1672653564, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
1672653594, BleManager, e, address = EF:31:55:E9:31:9C; msg = [DFU] Error 133: Connection timeout
1672653594, BleManager, w, address = EF:31:55:E9:31:9C; msg = [DFU] Retrying...
1672653594, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.refresh() (hidden)
1672653594, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.disconnect()
1672653594, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.close()
1672653594, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] DFU service started
1672653594, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Firmware file opened successfully
1672653594, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653595, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653596, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
1672653596, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
1672653626, BleManager, e, address = EF:31:55:E9:31:9C; msg = [DFU] Error 133: Connection timeout
1672653626, BleManager, w, address = EF:31:55:E9:31:9C; msg = [DFU] Retrying...
1672653626, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.refresh() (hidden)
1672653626, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.disconnect()
1672653626, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.close()
1672653626, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] DFU service started
1672653626, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Firmware file opened successfully
1672653626, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653627, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653628, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
1672653628, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
1672653658, BleManager, e, address = EF:31:55:E9:31:9C; msg = [DFU] Error 133: Connection timeout
1672653658, BleManager, w, address = EF:31:55:E9:31:9C; msg = [DFU] Retrying...
1672653658, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.refresh() (hidden)
1672653658, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.disconnect()
1672653658, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.close()
1672653658, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] DFU service started
1672653658, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Firmware file opened successfully
1672653658, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653659, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653660, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
1672653660, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
1672653680, BleManager, e, address = EF:31:55:E9:31:9C; msg = [DFU] Connection failed (0x85): GATT ERROR
1672653680, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.refresh() (hidden)
1672653680, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.disconnect()
1672653680, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.close()
1672653680, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(600)
1672653681, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] DFU service started
1672653681, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Firmware file opened successfully
1672653681, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653682, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
1672653683, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
1672653683, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
1672653685, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
1672653685, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Connected to EF:31:55:E9:31:9C
1672653685, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Discovering services...
1672653685, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.discoverServices()
1672653686, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Services discovered
1672653686, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(1000)
[... cut ...]
1672653724, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.close()
1672653724, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] wait(600)
```

We do scan for the device before initiating DFU. The flow is scan for the peripheral, check if firmware update is required. If required then go ahead with the DFU service.

Lib version - 1.12.1-beta01

Is there any known solution to this? Will upgrading the lib help? Haven't because we do not compile with sdk 33 yet. Let me know if you would want a separate issue for this

archie94, 2023-01-10:
Hey @philips77 when you are available can you please share your thoughts on this?

philips77, 2023-01-11:
Hi @archie94,

#### Observations

It's a bit hard to say what is the root cause of this. 
Based on the logs I see that:
1. You're using Secure DFU (8ec90003-f315-4f60-9fb8-838830daea50 is the Secure DFU Buttonless Charactersitic UUID)
2. The library sends *Jump to bootloader* command:
   ```
   1672653688, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] Enter bootloader sent (Op Code = 1)
   1672653688, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] Response received (Op Code = 1; Status = 1)
   ```
   but is unable to reconnect to the device in bootloader mode:
   ```
   1672653688, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Scanning for the DFU Bootloader...
   1672653693, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] DFU Bootloader not found. Trying the same address...
   ```
3. After that it tries to reconnect to the device with the same MAC address, but does it twice, which indicates an issue:
   ```
   1672653696, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Connecting to DFU target...
   1672653696, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt = device.connectGatt(autoConnect = false; TRANSPORT_LE; preferredPhy = LE_1M | LE_2M)
    // Connection event is received twice:
   1672653696, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Connected to EF:31:55:E9:31:9C
   1672653696, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Discovering services...
   1672653696, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.discoverServices()
    // Here as well:
   1672653696, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Connected to EF:31:55:E9:31:9C
   1672653696, BleManager, v, address = EF:31:55:E9:31:9C; msg = [DFU] Discovering services...
   1672653696, BleManager, d, address = EF:31:55:E9:31:9C; msg = [DFU] gatt.discoverServices()
   // And service is not found
   1672653724, BleManager, i, address = EF:31:55:E9:31:9C; msg = [DFU] Services discovered
   1672653724, BleManager, w, address = EF:31:55:E9:31:9C; msg = [DFU] DFU Service not found
   ```

#### Conclusions

We did address similar issue with in #356 or #345. The issue is that on some Android devices switching to bootlaoder mode doesn't trigger disconnection. The phone still thinks it's connected. Some phones actually keep the connection active for some time. When the lib is searching for an incremented MAC it is still not ready. Then, after 2 seconds it tries to reconnect to the old MAC (which is still connected), so this succeeds, or at least the phone thinks so. Service discovery may fail, as the device has perhaps reset.

Those 2 PRs try to fix this issue by calling `disconnect()` to make "sure" Android gets that, allowing setting "reset time" (delay before the lib starts scanning) or settings scan period (to increase from the default 2 seconds).

#### Recommendation

I would recommend trying updating using the [DFU app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.dfu) and trying different settings on the Settings screen. Check if you're able to make it work on those phones before applying changes to your app. If that works, I recommend to migrate to the latest version or doing cherry-picking important commits (you need to search for them on your own).

#### Questions

1. Does it work on some phones and fails on some other?
2. Does it work sometimes on those phones, or never?

archie94, 2023-01-18:
Thanks for your reply @philips77 

Sorry for the late revert. 

>     1. Does it work on some phones and fails on some other?
Yes only few users have reported this.

>     2. Does it work sometimes on those phones, or never?
It seems DFU never works on those phones. Although we have had a handful users only facing this issue.