nordicsemi/Android-DFU-Library #273: Can not import the library as a module due to gradle-bintray-push file broken reference

OlgaDery, 2020-10-05:
Hello,
I have tried to import the library to my Android project following the steps described in the documentation here - https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation. It failed to build due to the following error - "Could not read script '/Users/olgadery/AndroidStudioProjects/orpyxbluetoothsdk/gradle/gradle-bintray-push.gradle' as it does not exist." Looks like this file is located in the "gradle" folder that is not being imported if I am following the listed steps.
Are there any ways to fix this issue? Thanks in advance.

philips77, 2020-10-06:
Comment this line out.

OlgaDery, 2020-10-06:
Works, thanks a lot.

closed 2020-10-06 by OlgaDery