nordicsemi/Android-DFU-Library #274: Hex to Bin (Android vs iOS)

mikebertiean, 2020-10-08:
Hello

I am using the HexInputStream class (java) to convert a HEX file to BIN so I can write it to my peripheral.

IntelHex2BinConverter is being used in an iOS project.

Trying to convert the same HEX file using these two files gives us different results.
(see attached image)
[image]

The left side is the iOS conversion, while the right side is the Android conversion.

As you can see, the Android side is missing two 0's, causing the rest of the packets to not line up properly.

I am not quite sure what might be causing this. Any ideas on how it can be resolved so they are the same?

philips77, 2020-10-08:
Hi,
Could you share the relevant lines of the hex file? Few lines before, few after.

mikebertiean, 2020-10-09:
:10108000000002C016028498A00B66CE0000000A81
:101090008408E62C006C0AF315006801FF0B1A3077
:0A10A00048A16B038800000001**A8**BE
:0810AC00**AC**F0FC7F0100000024
:1010B40001006203223344556677010000000000FA
:1010C4000200000014000000C49112000A00000095

mikebertiean, 2020-10-09:
The issue lies in the two short lines ... doing the conversion manually it appears that it is indeed converting it correctly ... 

... 0, 1, **168**, **172**, 240 ...

I guess what I need to try and figure out now is why on the iOS side it is padding 2 zeros in-between 168 and 172.

The weird part is that on the iOS side, the DFU works well, while on Android it seems to brick the peripheral.

mikebertiean, 2020-10-09:
It appears that we should be filling in the missing address locations with 0's

So the last address only has 10 bytes of data ... but then then next line is not exactly consecutive to it (i.e. not right below the next 10), therefore, we need to fill those two empty spots with 0's.

closed 2020-10-09 by mikebertiean