nordicsemi/Android-DFU-Library #276: DFU problems on Huawei devices  Android 9

xiaochenMate, 2020-10-22:
Hello, I use your DFU SDK on Huawei devices  Android 9. When an interrupt occurs during the upgrade process, the exception "device returned error after sending file (error 6): operation failed" is thrown. The specific log is as follows. This problem must occur on Android 9 Huawei devices. Can you solve it? Thanks！

2020-10-22 14:57:30.859 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Connecting to the device...
2020-10-22 14:57:30.859 9408-10163/com.wiparking.lockwiparking.piciwebota I/BluetoothDevice: connectGatt
2020-10-22 14:57:30.860 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: connect() - device: E8:A1:16:A4:A4:CB, auto: false
2020-10-22 14:57:30.860 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: registerApp()
2020-10-22 14:57:30.861 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: registerApp() - UUID=5bd6b2d5-7c7c-42c2-887b-9786143997f8
2020-10-22 14:57:30.877 9408-9465/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2020-10-22 14:57:31.198 9408-9465/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=E8:A1:16:A4:A4:CB
2020-10-22 14:57:31.198 9408-9465/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Connected to GATT server
2020-10-22 14:57:31.204 9408-9465/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: discoverServices() - device: E8:A1:16:A4:A4:CB
2020-10-22 14:57:31.206 9408-9465/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Attempting to start service discovery... succeed
2020-10-22 14:57:31.208 9408-9408/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2020-10-22 14:57:31.664 9408-9436/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onConnectionUpdated() - Device=E8:A1:16:A4:A4:CB interval=24 latency=0 timeout=400 status=0
2020-10-22 14:57:31.900 9408-9436/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onConnectionUpdated() - Device=E8:A1:16:A4:A4:CB interval=6 latency=0 timeout=500 status=0
2020-10-22 14:57:32.101 9408-9436/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onSearchComplete() = Device=E8:A1:16:A4:A4:CB Status=0
2020-10-22 14:57:32.102 9408-9436/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Services discovered
2020-10-22 14:57:32.156 9408-9436/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onConnectionUpdated() - Device=E8:A1:16:A4:A4:CB interval=12 latency=0 timeout=400 status=0
2020-10-22 14:57:32.517 9408-9436/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onConnectionUpdated() - Device=E8:A1:16:A4:A4:CB interval=12 latency=0 timeout=400 status=0
2020-10-22 14:57:33.122 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Reading DFU version number...
2020-10-22 14:57:33.177 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
2020-10-22 14:57:33.180 9408-10163/com.wiparking.lockwiparking.piciwebota W/DfuImpl: Legacy DFU bootloader found
2020-10-22 14:57:34.182 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Enabling notifications...
2020-10-22 14:57:34.185 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
2020-10-22 14:57:35.227 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2020-10-22 14:57:35.261 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending image size array to DFU Packet (0b, 0b, 64472b)
2020-10-22 14:57:40.706 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
2020-10-22 14:57:40.750 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending 14 bytes of init packet
2020-10-22 14:57:40.750 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending init packet (Value = FF-FF-FF-FF-FF-FF-FF-FF-01-00-64-00-BA-24)
2020-10-22 14:57:40.754 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
2020-10-22 14:57:40.840 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending the number of packets before notifications (Op Code = 8, Value = 12)
2020-10-22 14:57:40.885 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending Receive Firmware Image request (Op Code = 3)
2020-10-22 14:57:40.941 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Uploading firmware...
2020-10-22 14:57:41.471 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Response received (Op Code = 16, Req Op Code = 3, Status = 6)
2020-10-22 14:57:41.472 9408-10163/com.wiparking.lockwiparking.piciwebota E/DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
2020-10-22 14:57:41.473 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuImpl: Sending Reset command (Op Code = 6)
2020-10-22 14:57:41.535 9408-9465/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=9 device=E8:A1:16:A4:A4:CB
2020-10-22 14:57:41.536 9408-9465/com.wiparking.lockwiparking.piciwebota W/DfuBaseService: Target device disconnected with status: 19
2020-10-22 14:57:41.538 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: refresh() - device: E8:A1:16:A4:A4:CB
2020-10-22 14:57:41.540 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Refreshing result: true
2020-10-22 14:57:41.540 9408-10163/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Cleaning up...
2020-10-22 14:57:41.545 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: close()
2020-10-22 14:57:41.545 9408-10163/com.wiparking.lockwiparking.piciwebota D/BluetoothGatt: unregisterApp() - mClientIf=9
2020-10-22 14:57:41.561 9408-9408/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2020-10-22 14:57:42.176 9408-9408/com.wiparking.lockwiparking.piciwebota I/DfuBaseService: DFU service destroyed

philips77, 2020-10-23:
Which version of the library are you using?

xiaochenMate, 2020-11-06:
There are problems with both old and new versions. The version I used about two years ago had problems, and then I updated it to the latest version. The same error was reported. Only Huawei's Android 9 and some Android Q had problems. Everything  was ok with other mobile devices，thx for your attention

QuentinFarizon, 2020-12-08:
I have the same issue on a Huawei P20 pro CLT-L09, Android 10
First connection works, then the dfu library does not send any progress and eventually disconnects.

I have tried with dfu library versions 1.11.0 and 1.11.1

I'm not sure how to get the logs of DfuImpl like above, I don't have them in logcat by default

QuentinFarizon, 2020-12-30:
I get the same issue also with a Samsung 10, SM-A750FN, Android 10.

glailton-ia, 2021-03-23:
I have the same issue on a Huawei nova 3i, Android 8.1.0
In my case I am trying to update using a .hex and a .bin file, sometimes the .bin file works fine, but when starts the .hex file always fails.
I am using dfu library 1.11.1.