nordicsemi/Android-DFU-Library #278: DFU update finished but onDfuCompleted() is not called

kinber95, 2020-11-05:
**DFU Bootloader version (please complete the following information):**
 - Bonding used: [yes]
 - Library version: [1.11.0]

**Device information (please complete the following information):**
 - Device: [Redmi 7A]
 - OS: [Android 9]

**Describe the bug**
When I connect and **don't** bond to my device, the dfu _onDfuCompleted()_ callback method would not be called, but the dfu update finished successfully. But when I connect and immediately bond my device all works fine. 

My DfuServiceInitiator Settings - > 

`   starter = new DfuServiceInitiator(bleDevice.getAddress())
                .setDeviceName(bleDevice.getName())
                .setKeepBond(false)
                .setPacketsReceiptNotificationsEnabled(true)
                .setPacketsReceiptNotificationsValue(6)
                .setZip(file.getAbsolutePath());
        starter.start(getContext(), DfuService.class);` 

**Logs**
2020-11-05 13:33:57.744 24693-24693/de.haveltec.ishareit I/DfuBaseService: DFU service created. Version: 1.11.0
2020-11-05 13:33:57.802 24693-25342/de.haveltec.ishareit I/DfuBaseService: Starting DFU service in foreground
2020-11-05 13:33:57.855 24693-25342/de.haveltec.ishareit I/DfuBaseService: Connecting to the device...
2020-11-05 13:33:57.856 24693-25342/de.haveltec.ishareit D/BluetoothGatt: connect() - device: E3:F5:2D:CA:FF:4D, auto: false
2020-11-05 13:33:57.856 24693-25342/de.haveltec.ishareit D/BluetoothGatt: registerApp()
2020-11-05 13:33:57.856 24693-25342/de.haveltec.ishareit D/BluetoothGatt: registerApp() - UUID=69b5ecc2-14f7-4240-ab69-a9d03d5796f1
2020-11-05 13:33:57.858 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onClientRegistered() - status=0 clientIf=16
2020-11-05 13:33:57.860 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=E3:F5:2D:CA:FF:4D
2020-11-05 13:33:57.860 24693-24736/de.haveltec.ishareit I/DfuBaseService: Connected to GATT server
2020-11-05 13:33:57.861 24693-24736/de.haveltec.ishareit D/BluetoothGatt: discoverServices() - device: E3:F5:2D:CA:FF:4D
2020-11-05 13:33:57.861 24693-24736/de.haveltec.ishareit I/DfuBaseService: Attempting to start service discovery... succeed
2020-11-05 13:33:57.870 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onSearchComplete() = Device=E3:F5:2D:CA:FF:4D Status=0
2020-11-05 13:33:57.871 24693-24736/de.haveltec.ishareit I/DfuBaseService: Services discovered
2020-11-05 13:33:57.876 24693-24693/de.haveltec.ishareit D/de.haveltec.ishareit.screens.firmware_update.FirmwareUpdateFragment: onDeviceConnected: 
2020-11-05 13:33:57.876 24693-24693/de.haveltec.ishareit D/de.haveltec.ishareit.screens.firmware_update.FirmwareUpdateFragment: onDfuProcessStarting: 
2020-11-05 13:33:58.895 24693-25342/de.haveltec.ishareit I/DfuImpl: Reading DFU version number...
2020-11-05 13:33:59.116 24693-25342/de.haveltec.ishareit I/DfuImpl: Version number read: 0.1 -> Application with Legacy buttonless update from SDK 7.0 or newer
2020-11-05 13:33:59.120 24693-25342/de.haveltec.ishareit W/DfuImpl: Application with legacy buttonless update found
2020-11-05 13:33:59.121 24693-25342/de.haveltec.ishareit I/DfuImpl: Enabling notifications...
2020-11-05 13:33:59.122 24693-25342/de.haveltec.ishareit D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
2020-11-05 13:33:59.993 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onConnectionUpdated() - Device=E3:F5:2D:CA:FF:4D interval=6 latency=0 timeout=500 status=0
2020-11-05 13:33:59.994 24693-24723/de.haveltec.ishareit D/BluetoothGatt: onConnectionUpdated() - Device=E3:F5:2D:CA:FF:4D interval=6 latency=0 timeout=500 status=0
2020-11-05 13:34:00.953 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onConnectionUpdated() - Device=E3:F5:2D:CA:FF:4D interval=78 latency=0 timeout=400 status=0
2020-11-05 13:34:00.953 24693-24723/de.haveltec.ishareit D/BluetoothGatt: onConnectionUpdated() - Device=E3:F5:2D:CA:FF:4D interval=78 latency=0 timeout=400 status=0
2020-11-05 13:34:01.239 24693-24693/de.haveltec.ishareit D/de.haveltec.ishareit.screens.firmware_update.FirmwareUpdateFragment: onEnablingDfuMode: 
2020-11-05 13:34:01.247 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2020-11-05 13:34:05.274 24693-24723/de.haveltec.ishareit D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=15 device=E3:F5:2D:CA:FF:4D
2020-11-05 13:34:05.275 24693-24736/de.haveltec.ishareit D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=16 device=E3:F5:2D:CA:FF:4D
2020-11-05 13:34:05.275 24693-24736/de.haveltec.ishareit W/DfuBaseService: Target device disconnected with status: 8
2020-11-05 13:34:05.278 24693-25342/de.haveltec.ishareit I/DfuBaseService: Cleaning up...
2020-11-05 13:34:05.279 24693-25342/de.haveltec.ishareit D/BluetoothGatt: cancelOpen() - device: E3:F5:2D:CA:FF:4D
2020-11-05 13:34:05.279 24693-24723/de.haveltec.ishareit D/BluetoothGatt: close()
2020-11-05 13:34:05.279 24693-24723/de.haveltec.ishareit D/BluetoothGatt: unregisterApp() - mClientIf=15
2020-11-05 13:34:05.282 24693-25342/de.haveltec.ishareit D/BluetoothGatt: close()
2020-11-05 13:34:05.282 24693-25342/de.haveltec.ishareit D/BluetoothGatt: unregisterApp() - mClientIf=16
2020-11-05 13:34:05.283 24693-25342/de.haveltec.ishareit I/DfuImpl: Starting service that will connect to the DFU bootloader
2020-11-05 13:34:05.287 24693-24693/de.haveltec.ishareit I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2020-11-05 13:34:07.321 24693-24693/de.haveltec.ishareit D/de.haveltec.ishareit.screens.firmware_update.FirmwareUpdateFragment: onDeviceConnecting: 
2020-11-05 13:34:07.341 24693-25342/de.haveltec.ishareit I/DfuBaseService: Connecting to the device...
2020-11-05 13:34:07.343 24693-25342/de.haveltec.ishareit D/BluetoothGatt: connect() - device: E3:F5:2D:CA:FF:4D, auto: false
2020-11-05 13:34:07.343 24693-25342/de.haveltec.ishareit D/BluetoothGatt: registerApp()
2020-11-05 13:34:07.343 24693-25342/de.haveltec.ishareit D/BluetoothGatt: registerApp() - UUID=a9f5dfed-626a-4313-b6a3-0c4d47657a52
2020-11-05 13:34:07.347 24693-24723/de.haveltec.ishareit D/BluetoothGatt: onClientRegistered() - status=0 clientIf=15
2020-11-05 13:34:08.044 24693-24723/de.haveltec.ishareit D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=15 device=E3:F5:2D:CA:FF:4D
2020-11-05 13:34:08.044 24693-24723/de.haveltec.ishareit E/DfuBaseService: Connection state change error: 133 newState: 0
2020-11-05 13:34:08.045 24693-25342/de.haveltec.ishareit I/DfuBaseService: Connection error after: 706 ms
2020-11-05 13:34:08.045 24693-25342/de.haveltec.ishareit E/DfuBaseService: An error occurred while connecting to the device:133
2020-11-05 13:34:08.048 24693-25342/de.haveltec.ishareit I/DfuBaseService: Attempt: 1
[... 37 lines cut ...]
2020-11-05 13:34:15.252 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2020-11-05 13:34:15.301 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending image size array to DFU Packet (0b, 0b, 49220b)
2020-11-05 13:34:17.252 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
2020-11-05 13:34:17.311 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending 14 bytes of init packet
2020-11-05 13:34:17.311 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending init packet (Value = FF-FF-FF-FF-FF-FF-FF-FF-01-00-80-00-7F-E9)
2020-11-05 13:34:17.317 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
2020-11-05 13:34:17.373 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending the number of packets before notifications (Op Code = 8, Value = 6)
2020-11-05 13:34:17.431 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending Receive Firmware Image request (Op Code = 3)
2020-11-05 13:34:17.511 24693-25342/de.haveltec.ishareit I/DfuImpl: Uploading firmware...
2020-11-05 13:34:45.181 24693-25342/de.haveltec.ishareit I/DfuImpl: Response received (Op Code = 16, Req Op Code = 3, Status = 1)
2020-11-05 13:34:45.182 24693-25342/de.haveltec.ishareit I/DfuImpl: Transfer of 49220 bytes has taken 27632 ms
2020-11-05 13:34:45.182 24693-25342/de.haveltec.ishareit I/DfuImpl: Sending Validate request (Op Code = 4)
2020-11-05 13:34:45.301 24693-25342/de.haveltec.ishareit I/DfuImpl: Response received (Op Code = 16, Req Op Code = 4, Status = 1)
[... cut ...]
2020-11-05 13:34:45.440 24693-25342/de.haveltec.ishareit I/DfuBaseService: Cleaning up...
2020-11-05 13:34:45.440 24693-25342/de.haveltec.ishareit D/BluetoothGatt: cancelOpen() - device: E3:F5:2D:CA:FF:4D
2020-11-05 13:34:45.442 24693-25342/de.haveltec.ishareit D/BluetoothGatt: close()
2020-11-05 13:34:45.442 24693-25342/de.haveltec.ishareit D/BluetoothGatt: unregisterApp() - mClientIf=15
2020-11-05 13:34:45.449 24693-25342/de.haveltec.ishareit I/BluetoothDevice: removeBond() for device E3:F5:2D:CA:FF:4D called by pid: 24693 tid: 25342

kinber95, 2020-11-12:
Update:

So I can tell clearly what the problem occurs because of the keeBond(false) setting. When I start my dfu without a bonding, the nordic library creates a bond for me but after the dfu update finished, it has problems to remove the bonding. When I remove it manually, the onDfuCompleted() callback is called.

dinesharjani, 2026-08-21:
Marking as closed since a related issue was marked as closed. If the issue has not been resolved and needs resolving, please reopen or file as a new Issue.

closed 2026-08-21 by dinesharjani