nordicsemi/Android-DFU-Library #282: Target device disconnected with status: 8 on android 11

azakharenka1, 2020-12-04:
**Information**
After starting dfu mode, library cannot connect to it on android 11. On other android versions/devices it works. 

**DFU Bootloader version:**
 - Library version: 1.11.1

**Device information:**
 - Device: pixel 4a
 - OS: 11

**Describe the bug**
After starting device firmware update, library cannot connect to it on android 11. On other versions all works. 

**Logs**
From device:
12:20:49.669 06:15.187 i> ble_svc: <<< API REQUEST: START_DFU_AND_REBOOT len 1 handle 212:20:49.670 06:15.188 i> ble_msg: >>> API RESPONSE: START_DFU_AND_REBOOT len 19 handle: 212:20:49.671 06:15.190 i> storage: FDS write complete for FIRMWARE INFORMATION12:20:49.756 06:15.238 i> reboot: Reset (only) with DFU (long) started...12:20:49.759 06:15.238 i> storage: FDS del_rec complete for NEW_CALIBRATION12:20:49.763 06:15.241 i> storage: FDS write complete for NEW_CALIBRATION12:20:49.766 06:15.288 i> reboot: FDS in IDLE state12:20:54.779 Connection lost

From mobile:
2020-12-04 12:29:40.233 D/BluetoothGatt: refresh() - device: D9:BA:60:5B:0C:AC
2020-12-04 12:29:40.238 D/UpdateLockInteractor$searchDfuDevice: Refreshing result: true
2020-12-04 12:29:40.272 I/DfuBaseService: DFU service created. Version: 1.11.1
2020-12-04 12:29:40.297 I/DfuBaseService: Starting DFU service in foreground
2020-12-04 12:29:40.349 I/DfuBaseService: Connecting to the device...
2020-12-04 12:29:40.350 D/BluetoothGatt: connect() - device: D9:BA:60:5B:0C:AC, auto: false
2020-12-04 12:29:40.350 D/BluetoothGatt: registerApp()
2020-12-04 12:29:40.350 D/BluetoothGatt: registerApp() - UUID=dfa1ad40-bdfe-4ea5-a2f4-3396a1a12109
2020-12-04 12:29:40.352 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2020-12-04 12:29:40.353 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=D9:BA:60:5B:0C:AC
2020-12-04 12:29:40.353 I/DfuBaseService: Connected to GATT server
2020-12-04 12:29:40.354 D/BluetoothGatt: discoverServices() - device: D9:BA:60:5B:0C:AC
2020-12-04 12:29:40.354 I/DfuBaseService: Attempting to start service discovery... succeed
2020-12-04 12:29:44.411 D/BluetoothGatt: onConnectionUpdated() - Device=D9:BA:60:5B:0C:AC interval=6 latency=0 timeout=500 status=8
2020-12-04 12:29:44.411 D/BluetoothGatt: onConnectionUpdated() - Device=D9:BA:60:5B:0C:AC interval=6 latency=0 timeout=500 status=8
2020-12-04 12:29:44.413 D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=9 device=D9:BA:60:5B:0C:AC
2020-12-04 12:29:44.413 D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=11 device=D9:BA:60:5B:0C:AC
2020-12-04 12:29:44.413 W/DfuBaseService: Target device disconnected with status: 8
2020-12-04 12:29:44.414 I/DfuBaseService: Connection error after: 4065 ms
2020-12-04 12:29:44.414 E/DfuBaseService: An error occurred while connecting to the device:8
2020-12-04 12:29:44.415 I/DfuBaseService: Attempt: 1
2020-12-04 12:29:44.416 D/BluetoothGatt: refresh() - device: D9:BA:60:5B:0C:AC
2020-12-04 12:29:44.417 I/DfuBaseService: Refreshing result: true
2020-12-04 12:29:44.417 I/DfuBaseService: Cleaning up...
2020-12-04 12:29:44.417 D/BluetoothGatt: cancelOpen() - device: D9:BA:60:5B:0C:AC
2020-12-04 12:29:44.418 D/BluetoothGatt: close()
2020-12-04 12:29:44.418 D/BluetoothGatt: unregisterApp() - mClientIf=11
2020-12-04 12:29:44.419 I/DfuBaseService: Restarting the service
2020-12-04 12:29:44.422 D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000301-4899-489f-a301-fbee544b1db0 enable: false
2020-12-04 12:29:44.423 D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000101-4899-489f-a301-fbee544b1db0 enable: false
2020-12-04 12:29:44.425 D/BluetoothGatt: close()

closed 2021-01-12 by azakharenka1

shy1008, 2021-05-18:
Did you solve it?

fluxxion82, 2022-01-27:
So I'm seeing this issue on an LG Thinq v40 Android 10 while using the latest versions of the library (1.12.0, 1.12.1-beta01). If I use version 1.11.0, I see the same status but the dfu process is able to continue. I'm not seeing any issues on a pixel 6 with android 12.