nordicsemi/Android-DFU-Library #284: Write opcode failed?

BaiduSu, 2020-12-11:
Write opcode operation occasionally errors, this error result in intent service thread always in wait status. Need to add timeout  for write opcode operation.

I have a method to implement timeout mechanism, but feel too clumsy. I want to hear about your ideas. 

Thanks!!!

philips77, 2021-01-19:
Well.. that would indeed be nice to have feature. I'll keep it open here and try to fix it in the future, but I'm busy with other tasks right now, so it may take a while.
Where exactly do you get that, and why is your device failing to send the response? Could you provide some logs?

Nuclear6, 2021-01-20:
check HCI log of Android phone, not find my write opcode data, It feels like a mobile phone compatibility issue.

philips77, 2024-10-25:
Hi,
I'm releasing new version: 2.7.0. Please try it and reopen the issue pasting logs from LogCat if the error is stil there.

closed 2024-10-25 by philips77