nordicsemi/Android-DFU-Library #285: Can't use DfuProgressListenerAdapter on my Unity Project

knightLin0211, 2020-12-24:
Hi,
I want to use DFU for my Unity project. But when I want to upload the file. Unity gives me an error message.
Please tell me the solution.
Best regards.

Unity's error msg:
AndroidPlayer(ADB@127.0.0.1:34999) UnityEngine.AndroidJavaException: java.lang.NoClassDefFoundError: Failed resolution of: Lno/nordicsemi/android/dfu/DfuProgressListenerAdapter;
  at UnityEngine.AndroidJNISafe.CheckException () [0x0008d] in /Users/bokken/buildslave/unity/build/Modules/AndroidJNI/AndroidJNISafe.cs:24 
  at UnityEngine.AndroidJNISafe.CallVoidMethod (System.IntPtr obj, System.IntPtr methodID, UnityEngine.jvalue[] args) [0x0000f] in /Users/bokken/buildslave/unity/build/Modules/AndroidJNI/AndroidJNISafe.cs:372 
  at UnityEngine.AndroidJavaObject._Call (System.String methodName, System.Object[] args) [0x0002d] in /Users/bokken/buildslave/unity/build/Modules/AndroidJNI/AndroidJava.cs:412 
  at UnityEngine.AndroidJavaObject.Call (System.String methodName, System.Object[] args) [0x00001] in /Users/bokken/buildslave/unity/build/Modules/AndroidJNI/AndroidJava.cs:239 
  at AndroidPlugin.OnUpdateClicked () [0x00002] in D:\texray_svn\DFU\Unity\DFU\Assets\Plugins\Android\AndroidPlugin.cs:29

My scripts:
 public class MainActivity extends UnityPlayerActivity {
    public void OnUpdateClicked()//Unity will call this method
    {
        if(connectedDevice == null)
        {
            UnityPlayer.UnitySendMessage("Canvas", "GetMsg", "Please connect device");
            return;
        }
       Progress progress = new Progress(this); =>this will be error
       DfuServiceListenerHelper.registerProgressListener(this, progress .dfuProgressListener);

//        final DfuServiceInitiator starter = new DfuServiceInitiator(connectedDevice.getAddress())
//                .setDeviceName(connectedDevice.getName())
//                .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
//            starter.setZip(mFileStreamUri, mFilePath);//file type已固定為DfuService.TYPE_AUTO所以直接設定
//            if (mScope != null)
//                starter.setScope(mScope);
//        Log.i(LOGTAG, starter.toString());
//        starter.start(this, DfuService.class);
    }
}

public class Progress {

    public Progress(Context context)
    {
        Log.i(LOGTAG, context.getPackageName());
    }

    public DfuProgressListener dfuProgressListener = new DfuProgressListenerAdapter() {
        @Override
        public void onDeviceConnecting(final String deviceAddress) {
            UnityPlayer.UnitySendMessage("Canvas", "GetMsg", "Connecting...");
            Log.i(LOGTAG, "Connecting...");
        }
    };
}

philips77, 2020-12-26:
I'll be back in the office in mid January, but it may be a while before I look into this.

knightLin0211, 2020-12-28:
Hi,
I think I solve some parts, but I don't understand why?

I put the java files in the DFU library into my own project. And build again. When I make a plugin for Unity, follow the steps I must delete BuildConfing.class in classes.jar to create the apk file. In this way, the BuildConfing.class cannot be deleted from classes.jar (In order to delete BuildConfing.class, so I set DfuProgressListener as a separate Java file), but I still tried to create the APK file, but Unity can create it!!!! And Unity will not display error messages.
WHY??!!

Later, I tried again. I directly placed the DfuProgressListener method in my MainActivity.java according to the document. Of course, it could not delete BuildConfing.class. I also tried to create an APK file. Unity could create it, but it crashed and unusable.

Put the Java files in the DFU library into the file hierarchy of my project
[image]
MainActivity.java 

  public class MainActivity extends UnityPlayerActivity {

    public void OnUpdateClicked()//Unity will call this method
    {
        if(connectedDevice == null)
        {
            UnityPlayer.UnitySendMessage("Canvas", "GetMsg", "Please connect device");
            return;
        }
       Progress progress = new Progress(this); =>It successfully returned the Connecting... message

      // DfuServiceListenerHelper.registerProgressListener(this, progress .dfuProgressListener);

       //  final DfuServiceInitiator starter = new DfuServiceInitiator(connectedDevice.getAddress())
       //  .setDeviceName(connectedDevice.getName())
       //  .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
       //  starter.setZip(mFileStreamUri, mFilePath);//file type已固定為DfuService.TYPE_AUTO所以直接設定
       //   if (mScope != null)
       //   starter.setScope(mScope);
       //   Log.i(LOGTAG, starter.toString());
       //   starter.start(this, DfuService.class);
    }
}

Progress.java

public class Progress {

    public Progress(Context context)
    {
        Log.i(LOGTAG, context.getPackageName());
    }

    public DfuProgressListener dfuProgressListener = new DfuProgressListenerAdapter() {
        @Override
        public void onDeviceConnecting(final String deviceAddress) {
            UnityPlayer.UnitySendMessage("Canvas", "GetMsg", "Connecting...");
            Log.i(LOGTAG, "Connecting...");
        }
    };
}

DfuProgressListener method in my MainActivity.java
[image]

Can't delete  BuildConfing.class
[image]

If there is new progress, I will come back to inform

philips77, 2021-01-19:
Sorry, we don't have experience with Unity. Did you try asking your question on StackOverflow or some Unity forum?

knightLin0211, 2021-01-25:
I haven't solved this problem. I searched the forum and didn't see similar problems. So that's why I came here.

abkoradiya, 2021-09-28:
@knightLin0211 have you made plugin for unity?

knightLin0211, 2021-09-29:
@abkoradiya Sorry, the project has been stopped until a solution is found.

closed 2021-09-29 by philips77