nordicsemi/Android-DFU-Library #287: Context.startForegroundService() did not then call Service.startForeground()

VinayTShetty, 2021-01-21:
```
public void dfuupdate(){
        final DfuServiceInitiator starter = new DfuServiceInitiator("D4:A6:CB:43:B6:70")
                .setDeviceName("test");

        starter.setPrepareDataObjectDelay(300L);

        if (0 == DfuService.TYPE_AUTO)
            starter.setZip(fileStreamUri, null);
        else {
            //  starter.setBinOrHex(0, mFileStreamUri, mFilePath).setInitFile(mInitFileStreamUri, mInitFilePath);
        }
       starter.start(this, DfuService.class);

    }
```

philips77, 2021-01-21:
What error?

VinayTShetty, 2021-01-21:
[image]
Please find the screen shot of the error showing

philips77, 2021-01-22:
Check the first error you got. It's thrown here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b5d587bfbfaa5b1eacebce3a8846746a9d3a9793/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1113-L1116
(there should be "or", not "of", it's a typo).
Looks like the `fileStreamUri` you give is *null*.

VinayTShetty, 2021-01-23:
@philips77 
Thanks for helping me to resolve the issue and make the DFU working.
Appreciate it Mate.
As you were right i was not giving the file path.

**_### Erorr 2:-Context.startForegroundService() did not then call Service.startForeground()_**
This issue was resolved by adding 
     .setForeground(false) to theDfuServiceInitiator.
     
     For future reference for other i have addded the code below.
     
   ```
    final DfuServiceInitiator starter = new DfuServiceInitiator("D4:A6:CB:43:B6:70")
                .setForeground(false)
                .setDeviceName("Succorfish SC2");

        starter.setPrepareDataObjectDelay(300L);

        if (0 == DfuService.TYPE_AUTO)
            starter.setZip(fileStreamUri, filepath);
        else {
              starter.setBinOrHex(0, uri, filepath).setInitFile(uri, filepath);
        }
       starter.start(this, DfuService.class);
```

Inside if condition i have hard coded the file type.
Use it as per your requirements.

closed 2021-01-23 by VinayTShetty