nordicsemi/Android-DFU-Library #298: DFU problems on Huawei devices to Upload .hex files

glailton-ia, 2021-04-06:
**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: yes
 - Library version: 1.11.1

**Device information (please complete the following information):**
 - Device: Huawei nova 3i
 - OS: Android 8.1.0

**Your question**
I have an issue trying to update the firmware using Huawei devices, Android 8.1.0
In my case I am trying to update using a .hex and a .bin file, sometimes the .bin file works fine, but when starts the .hex file always fails.
I am using dfu library 1.11.1.

Is there any solution for this issue?

**Logs**
2021-04-06 08:40:54.864 W/DfuCtrl.Refresh: stopScanProcess.mIsScanning
2021-04-06 08:40:54.864 W/DfuCtrl.Switch: stopScanProcess.already
2021-04-06 08:40:54.864 W/NormalModeController: removeCommandListener.SwitchToDfuModeListener.AlreadyRemoved: com. controller.dfumode.DfuSwitchController$2@88977c6
2021-04-06 08:40:54.865 I/DfuCtrl.Download: setupInit()@(Handler=Handler (android.os.Handler) {d9752b4}, Retry=(false, 0/3, 20000), Ready=true, Activity=com. wearable_template_app.activity.FirmwareUpdatingActivity@36a6024, Listener=com. controller.dfumode.DfuMasterController$1@fd93edd, Address=, File=, Result=(UndefinedUnknown, ), MainService=(com. controller.MainService@c10c632), Index=(0/2))=true
2021-04-06 08:40:55.948 I/DfuCtrl.Master: startSwitchProcedure(Index=0, Type=1)
2021-04-06 08:40:55.957 I/DfuCtrl.Switch: setupInit()@(Handler=Handler (android.os.Handler) {f75c64e}, Retry=(false, 0/8, 15000), Ready=true, Activity=com. wearable_template_app.activity.FirmwareUpdatingActivity@36a6024, Listener=com. controller.dfumode.DfuMasterController$1@fd93edd, Address=, File=, Result=(UndefinedUnknown, ), MainService=(com. controller.MainService@c10c632), Ref=(CB:00:C1:B8:12:DE/01F75CC4), Mode=(DfuBle, 00001530-1212-efde-1523-785feabcd123))=true
2021-04-06 08:40:55.958 W/DfuCtrl.Switch: stopScanProcess.already
2021-04-06 08:40:55.958 W/NormalModeController: removeCommandListener.SwitchToDfuModeListener.AlreadyRemoved: com. controller.dfumode.DfuSwitchController$2@88977c6
2021-04-06 08:40:55.960 D/MainService: MainService - BleCommand - Switch DFU Mode
2021-04-06 08:40:55.961 D/CMD.Builder.Base: CommandCode:112, CommandName:SwitchToDfuMode, CommandObject:DfuSwitch(1=BluetoothRadioStack), Data:8070D417A68401
2021-04-06 08:40:56.342 D/NormalModeController: onCharacteristicChanged.State(Multi=false, Seq=0, Cmd=112=SwitchToDfuMode, Len=2)
2021-04-06 08:40:56.343 D/CMD.Builder.Base: getObject(112=SwitchToDfuMode, 8070) = BaseObj(112=SwitchToDfuMode, 0=Acknowledgement)
2021-04-06 08:40:56.344 I/DfuCtrl.Switch: mSwitchToDfuListener.onReceive(BaseObj(112=SwitchToDfuMode, 0=Acknowledgement))
2021-04-06 08:41:10.979 D/MainService: MainService - BleCommand - Switch DFU Mode
2021-04-06 08:41:10.980 D/CMD.Builder.Base: CommandCode:112, CommandName:SwitchToDfuMode, CommandObject:DfuSwitch(1=BluetoothRadioStack), Data:8070D417A68401
2021-04-06 08:41:17.269 I/DfuCtrl.Switch: mConnectionStateListener.onReceiveConnectionStatusChange = false
2021-04-06 08:41:17.300 D/DfuCtrl.Switch: startScanProcess()
2021-04-06 08:41:17.455 D/DfuCtrl.Switch: stopScanProcess()
2021-04-06 08:41:17.456 W/DfuCtrl.Switch: stopScanProcess.already
2021-04-06 08:41:17.458 I/DfuCtrl.Master: onSwitchComplete.Accept CB:00:C1:B8:12:DF for (Index=0/4, Type=1)
2021-04-06 08:41:18.466 I/DfuCtrl.BLE: setupInit()@(Handler=Handler (android.os.Handler) {4436fbd}, Retry=(false, 0/4, 15000), Ready=true, Activity=com. wearable_template_app.activity.FirmwareUpdatingActivity@36a6024, Listener=com. controller.dfumode.DfuMasterController$1@fd93edd, Address=CB:00:C1:B8:12:DF, File=/data/user/0/package/files/dfutemp/MLRadio063017.0.1.07.hex, Result=(UndefinedUnknown, ), Receiver=(com. bluetoothcontroller.DfuBleController$1@83a3b14))=true
2021-04-06 08:41:18.484 I/DfuBaseService: DFU service created. Version: 1.11.1
2021-04-06 08:41:18.497 I/DfuBaseService: Starting DFU service in foreground
2021-04-06 08:41:18.505 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] DFU service started)
2021-04-06 08:41:18.505 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Opening file...)
2021-04-06 08:41:18.511 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Firmware file opened successfully)
2021-04-06 08:41:18.512 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Connecting to DFU target...)
2021-04-06 08:41:18.513 I/DfuBaseService: Connecting to the device...
2021-04-06 08:41:18.514 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M))
2021-04-06 08:41:30.116 I/DfuBaseService: Connected to GATT server
2021-04-06 08:41:30.117 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Connected to CB:00:C1:B8:12:DF)
2021-04-06 08:41:30.121 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Discovering services...)
2021-04-06 08:41:30.121 I/DfuBaseService: Attempting to start service discovery... succeed
2021-04-06 08:41:30.121 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.discoverServices())
2021-04-06 08:41:30.125 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2021-04-06 08:41:30.127 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED)
2021-04-06 08:41:30.710 I/DfuBaseService: Services discovered
2021-04-06 08:41:30.713 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Services discovered)
2021-04-06 08:41:30.736 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] wait(1000))
2021-04-06 08:41:31.737 I/DfuImpl: No DFU Version characteristic found -> Bootloader from SDK 6.1 or older
2021-04-06 08:41:31.739 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(10 @ [DFU] DFU Version characteristic not found)
2021-04-06 08:41:31.742 W/DfuImpl: Legacy DFU bootloader found
2021-04-06 08:41:31.743 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] wait(1000))
2021-04-06 08:41:32.743 I/DfuImpl: Enabling notifications...
2021-04-06 08:41:32.746 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123)
2021-04-06 08:41:32.746 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true))
2021-04-06 08:41:32.746 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true))
2021-04-06 08:41:32.752 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00))
2021-04-06 08:41:32.784 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00)
2021-04-06 08:41:32.785 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123)
2021-04-06 08:41:32.786 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(10 @ [DFU] Notifications enabled)
2021-04-06 08:41:32.787 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] wait(1000))
2021-04-06 08:41:33.494 W/DfuCtrl.BLE: mRetryRunnable.isAlreadyRunning
[... 22 lines cut ...]
2021-04-06 08:41:52.703 I/DfuImpl: Transfer of 46760 bytes has taken 17283 ms
2021-04-06 08:41:52.704 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(10 @ [DFU] Response received (Op Code = 3, Status = 1))
2021-04-06 08:41:52.704 I/DfuImpl: Sending Validate request (Op Code = 4)
2021-04-06 08:41:52.704 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(10 @ [DFU] Upload completed in 17283 ms)
2021-04-06 08:41:52.705 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(1 @ [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123)
2021-04-06 08:41:52.705 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123))
2021-04-06 08:41:52.732 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 04)
2021-04-06 08:41:52.732 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(10 @ [DFU] Validate request sent)
2021-04-06 08:41:52.793 I/DfuImpl: Response received (Op Code = 16, Req Op Code = 4, Status = 1)
2021-04-06 08:41:52.793 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(5 @ [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-04-01)
[... cut ...]
2021-04-06 08:41:52.925 I/DfuBaseService: Refreshing result: true
2021-04-06 08:41:52.925 I/DfuBaseService: Cleaning up...
2021-04-06 08:41:52.926 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] gatt.disconnect())
2021-04-06 08:41:52.933 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2021-04-06 08:41:52.934 I/DfuCtrl.BLE: mBleDfuReceiver.onReceive.Log(0 @ [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED)
2021-04-06 08:41:54.375 I/DfuBaseService: DFU service destroyed

mkotzjan, 2021-05-06:
Did you also set your init file? Take a look at this line:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b5d587bfbfaa5b1eacebce3a8846746a9d3a9793/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L743
This is required when using a .bin or .hex file for updating

philips77, 2021-05-06:
Hello,
Sorry for the delay, I was busy with other tasks. 

Have a look at this log message:
```
2021-04-06 08:41:31.737 I/DfuImpl: No DFU Version characteristic found -> Bootloader from SDK 6.1 or older
```
So you're using DFU from SDK 6.1 or older. By counting number of services, the library assumes that the device is in DFU bootloader mode, which is correct. Looks like there's only Legacy DFU service.

Before SDK 7 the init packet was optional, so your HEX gets accepted and transfer starts. Without the init packet you cannot, however, ensure, that the device you're connected to will support the fw you're sending. There no any validation, except from checking bytes number and comparing to one sent before the firmware. It may be, that you're sending a fw that is not compatible with the SoftDevice installed, or for completely other device. That's why it fails.

If the bin file works sometime, that means there may be an issue with how the hex file is parsed. Could you check if the bin file that matches the hex file is 46760 bytes long? This is how much bytes the lib can take from the hex file and send.

dinesharjani, 2026-08-21:
Marking as closed due to lack of follow-up.

closed 2026-08-21 by dinesharjani