nordicsemi/Android-DFU-Library #302: Attempt to invoke virtual method 'void android.os.Bundle.unparcel()' on a null object reference

Strife91, 2021-04-23:
OnePlus 7T Pro
Android 11

Gettings this crash at intent.putExtras(getIntent().getExtras()); 

What am I missing?

`public class NotificationActivity extends Activity {
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        // If this activity is the root activity of the task, the app is not running
        if (isTaskRoot()) {
            // Start the app before finishing
            final Intent intent = new Intent(this, MainActivity.class);
            intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
            intent.putExtras(getIntent().getExtras()); // copy all extras
            startActivity(intent);
        }

        // Now finish, which will drop you to the activity at which you were at the top of the task stack
        finish();
    }
}`

philips77, 2021-04-26:
Well, for some reason `getIntent()` returns *null*.
Try:
```java
public class NotificationActivity extends Activity {
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        // If this activity is the root activity of the task, the app is not running
        if (isTaskRoot()) {
            // Start the app before finishing
            final Intent intent = new Intent(this, MainActivity.class);
            intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
            final Intent i = getIntent();
            if (i != null) {
                intent.putExtras(i.getExtras()); // copy all extras
            }
            startActivity(intent);
        }

        // Now finish, which will drop you to the activity at which you were at the top of the task stack
        finish();
    }
}
```

philips77, 2021-04-26:
The `NotificationActivity` should be started with this intent:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b5d587bfbfaa5b1eacebce3a8846746a9d3a9793/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1858-L1863

closed 2021-04-26 by philips77

Strife91, 2021-04-26:
The problem was when I created NotificationActivity it was automatically added to manifest as launch activity. 

Still not getting any progress notifications, but that's another problem.  I think you can close this issue.