nordicsemi/Android-DFU-Library #303: Frequent GATT errors

sjmerel, 2021-04-28:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: don't know; the firmware was provided by a 3rd party.
 - Bonding used: no
 - Library version: 1.11.1

**Device information (please complete the following information):**
 - Device: various (including Pixel 2, OnePlus 3)
 - OS: various (including 9.0, 11.0)

**Describe the bug**
Firmware updates frequently fail on some devices, when using library version 1.11.1.  
DfuProgressListener.onError() is called with error=133, errorType=1, message="GATT ERROR"

This happens more frequently on some devices than on others. Note that this seems to be a new issue in library version 1.11.1; I have not been able to reproduce it with versions 1.11.0 or 1.10.4.

I ran multiple firmware updates one after the other on several devices to test:
On a Pixel 2, the error happened 5/10 times.
On a OnePlus 3, it happened 1/12 times.
One a Samsung Galaxy S10 it happened 0/10 times.

**Logs**

In the log below, note the line:

> Scanning for new address finished with: null

The address is only null when the error occurs.

> 2021-04-27 18:21:32.134 8831-8831/io.focusmotion.tkr I/DfuBaseService: DFU service created. Version: 1.11.1
> 2021-04-27 18:21:32.153 8831-9435/io.focusmotion.tkr I/DfuBaseService: Starting DFU service in foreground
> 2021-04-27 18:21:32.160 8831-9435/io.focusmotion.tkr W/DfuBaseService: getNotificationTarget() should not return null if the service is to be started as a foreground service
> 2021-04-27 18:21:32.238 8831-9435/io.focusmotion.tkr I/DfuBaseService: Connecting to the device...
> 2021-04-27 18:21:32.239 8831-9435/io.focusmotion.tkr D/BluetoothGatt: connect() - device: EC:5E:29:62:B5:E8, auto: false
> 2021-04-27 18:21:32.239 8831-9435/io.focusmotion.tkr D/BluetoothGatt: registerApp()
> 2021-04-27 18:21:32.240 8831-9435/io.focusmotion.tkr D/BluetoothGatt: registerApp() - UUID=cd0731c0-ae04-40cf-be2c-4f897b4cff57
> 2021-04-27 18:21:32.242 8831-9034/io.focusmotion.tkr D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
> 2021-04-27 18:21:32.243 8831-9034/io.focusmotion.tkr D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=EC:5E:29:62:B5:E8
> 2021-04-27 18:21:32.243 8831-9034/io.focusmotion.tkr I/DfuBaseService: Connected to GATT server
> 2021-04-27 18:21:32.244 8831-9034/io.focusmotion.tkr D/BluetoothGatt: discoverServices() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:21:32.245 8831-9034/io.focusmotion.tkr I/DfuBaseService: Attempting to start service discovery... succeed
> 2021-04-27 18:21:32.249 8831-9034/io.focusmotion.tkr D/BluetoothGatt: onSearchComplete() = Device=EC:5E:29:62:B5:E8 Status=0
> 2021-04-27 18:21:32.249 8831-9034/io.focusmotion.tkr I/DfuBaseService: Services discovered
> 2021-04-27 18:21:32.252 8831-9435/io.focusmotion.tkr I/DfuImpl: Buttonless service without bond sharing found -> SDK 13 or newer
> 2021-04-27 18:21:33.257 8831-9435/io.focusmotion.tkr I/DfuImpl: Enabling indications...
> 2021-04-27 18:21:33.267 8831-9435/io.focusmotion.tkr D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
> 2021-04-27 18:21:34.263 8831-9034/io.focusmotion.tkr D/BluetoothGatt: onConnectionUpdated() - Device=FA:E2:8E:ED:62:77 interval=18 latency=0 timeout=400 status=0
> 2021-04-27 18:21:34.302 8831-9435/io.focusmotion.tkr I/DfuImpl: Sending Enter Bootloader (Op Code = 1)
> 2021-04-27 18:21:34.352 8831-9435/io.focusmotion.tkr I/DfuImpl: Response received (Op Code = 1, Status = 1)
> 2021-04-27 18:21:35.860 8831-8970/io.focusmotion.tkr V/FA: Inactivity, disconnecting from the service
> 2021-04-27 18:21:38.384 8831-9034/io.focusmotion.tkr D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=10 device=EC:5E:29:62:B5:E8
> 2021-04-27 18:21:38.385 8831-9034/io.focusmotion.tkr W/DfuBaseService: Target device disconnected with status: 8
> 2021-04-27 18:21:38.386 8831-8851/io.focusmotion.tkr D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=11 device=EC:5E:29:62:B5:E8
> 2021-04-27 18:21:38.389 8831-9435/io.focusmotion.tkr W/focusmotion.tk: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (greylist, reflection, allowed)
> 2021-04-27 18:21:38.389 8831-9435/io.focusmotion.tkr D/BluetoothGatt: refresh() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:21:38.391 8831-8831/io.focusmotion.tkr D/BluetoothGatt: setCharacteristicNotification() - uuid: 1cac0003-656e-696c-4b5f-6e6572726157 enable: false
> 2021-04-27 18:21:38.391 8831-9435/io.focusmotion.tkr I/DfuBaseService: Refreshing result: true
> 2021-04-27 18:21:38.391 8831-9435/io.focusmotion.tkr I/DfuBaseService: Cleaning up...
> 2021-04-27 18:21:38.395 8831-9435/io.focusmotion.tkr D/BluetoothGatt: cancelOpen() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:21:38.397 8831-8831/io.focusmotion.tkr I/FM: FM INFO: io.focusmotion.ptsdk.Device$Companion: "Sensoria-C1-1D43" connected: false
> 2021-04-27 18:21:38.400 8831-9435/io.focusmotion.tkr D/BluetoothGatt: close()
> 2021-04-27 18:21:38.400 8831-9435/io.focusmotion.tkr D/BluetoothGatt: unregisterApp() - mClientIf=10
> 2021-04-27 18:21:38.401 8831-9435/io.focusmotion.tkr I/DfuImpl: Restarting to bootloader mode
> 2021-04-27 18:21:38.402 8831-8831/io.focusmotion.tkr I/FM: FM INFO: DeviceService: DeviceService: device 0: Sensoria-C1-1D43
> 2021-04-27 18:21:38.407 8831-8831/io.focusmotion.tkr I/FM: FM INFO: DeviceService:   storing serial DE35017C12E84212
> 2021-04-27 18:21:38.411 8831-9435/io.focusmotion.tkr D/BluetoothAdapter: isLeEnabled(): ON
[... 42 lines cut ...]
> 2021-04-27 18:22:47.559 8831-9435/io.focusmotion.tkr I/DfuBaseService: Cleaning up...
> 2021-04-27 18:22:47.561 8831-9435/io.focusmotion.tkr D/BluetoothGatt: cancelOpen() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:22:47.562 8831-9435/io.focusmotion.tkr D/BluetoothGatt: close()
> 2021-04-27 18:22:47.563 8831-9435/io.focusmotion.tkr D/BluetoothGatt: unregisterApp() - mClientIf=10
> 2021-04-27 18:22:47.563 8831-9435/io.focusmotion.tkr I/DfuBaseService: Restarting the service
> 2021-04-27 18:22:47.579 8831-9435/io.focusmotion.tkr I/DfuBaseService: Starting DFU service in foreground
> 2021-04-27 18:22:47.580 8831-9435/io.focusmotion.tkr W/DfuBaseService: getNotificationTarget() should not return null if the service is to be started as a foreground service
> 2021-04-27 18:22:49.598 8831-9435/io.focusmotion.tkr I/DfuBaseService: Connecting to the device...
> 2021-04-27 18:22:49.603 8831-9435/io.focusmotion.tkr D/BluetoothGatt: connect() - device: EC:5E:29:62:B5:E8, auto: false
> 2021-04-27 18:22:49.605 8831-9435/io.focusmotion.tkr D/BluetoothGatt: registerApp()
> 2021-04-27 18:22:49.608 8831-9435/io.focusmotion.tkr D/BluetoothGatt: registerApp() - UUID=3cd80b94-97e9-4637-8f20-204c4d187d2e
> 2021-04-27 18:22:49.616 8831-8851/io.focusmotion.tkr D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
> 2021-04-27 18:23:19.620 8831-8851/io.focusmotion.tkr D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=EC:5E:29:62:B5:E8
> 2021-04-27 18:23:19.620 8831-8851/io.focusmotion.tkr E/DfuBaseService: Connection state change error: 133 newState: 0
> 2021-04-27 18:23:19.620 8831-9435/io.focusmotion.tkr I/DfuBaseService: Connection error after: 30024 ms
> 2021-04-27 18:23:19.621 8831-9435/io.focusmotion.tkr E/DfuBaseService: Device not reachable. Check if the device with address EC:5E:29:62:B5:E8 is in range, is advertising and is connectable
> 2021-04-27 18:23:19.621 8831-9435/io.focusmotion.tkr I/DfuBaseService: Attempt: 3
> 2021-04-27 18:23:19.623 8831-9435/io.focusmotion.tkr D/BluetoothGatt: refresh() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:23:19.623 8831-9435/io.focusmotion.tkr I/DfuBaseService: Refreshing result: true
> 2021-04-27 18:23:19.623 8831-9435/io.focusmotion.tkr I/DfuBaseService: Cleaning up...
> 2021-04-27 18:23:19.624 8831-9435/io.focusmotion.tkr D/BluetoothGatt: cancelOpen() - device: EC:5E:29:62:B5:E8
> 2021-04-27 18:23:19.625 8831-9435/io.focusmotion.tkr D/BluetoothGatt: close()
> 2021-04-27 18:23:19.625 8831-9435/io.focusmotion.tkr D/BluetoothGatt: unregisterApp() - mClientIf=10
> 2021-04-27 18:23:20.301 8831-8831/io.focusmotion.tkr I/DfuBaseService: DFU service destroyed
>

sjmerel, 2021-05-05:
Is this library actively maintained?  Seems like there hasn't been much response to the most recent bug reports. @philips77 ?

mkotzjan, 2021-05-06:
Error 133 does happen frequently on android, it's a known issue. See here for some more informations: https://devzone.nordicsemi.com/f/nordic-q-a/33313/android-gatt-133-error

philips77, 2021-05-06:
Hi, sorry for the delay. The library is maintained, but I'm busy with other tasks, so it waits for its time...

Looks like the issue is related to how the phone filters scan packets. Have a look at this change in the latest version: https://github.com/NordicSemiconductor/Android-DFU-Library/commit/72c01758f68a1462113e8485289f5a13c1895c2b#diff-da467c50d8806bbe969e81aba215e9b1fc4b34735e13aa46f665fb7ff565aaf6R97
I'm surprised it works only sometimes. It may be, that with the device address set as filter the phone scans for public MAC addresses. This is actually explicitly stated in Android 12: https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/le/ScanFilter.java#590
For me it looks, like if the device has been scanned before, Android will know the device type, so it may set type random. If it hadn't, it will assume public type and scan will fail.

philips77, 2021-05-06:
>Error 133 does happen frequently on android, it's a known issue. See here for some more informations: https://devzone.nordicsemi.com/f/nordic-q-a/33313/android-gatt-133-error

This 133 here happens after 30+ sec, so it's a timeout. Android scans for public device address, not random, so it ignores the advertising bootloader.

philips77, 2021-05-06:
As filter is not mandatory to scan in background, I think the only thing that could be done is set a different filter. That means we may need to use the "alternative advertising name" feature made with iOS in mind (DFU from SDK 14+). Or set a different filter, e.g. with Secure DFU Service UUID, but that would need to be user configurable setting.

sjmerel, 2021-05-06:
Background scanning seems like it's probably not a very common requirement, whereas the behavior with the current scan filter breaks DFU updates on some devices, so requiring some configuration to support background scanning seems like a reasonable tradeoff.

sjmerel, 2021-05-06:
For what it's worth, in my own Bluetooth code I had found that scan filters worked unreliably on some of my testing devices, so I stopped using them (though again, we don't need to scan in the background).

philips77, 2021-05-06:
Yes, we know it's unreliable. I'll check if adding an empty filter works in bg.

QuentinFarizon, 2021-08-13:
Hello @philips77 , any news on this ? Encountering this also on some customer phones (Oppo,  Xiaomi, ...)

QuentinFarizon, 2021-10-08:
I've rollbacked to 1.11.0 since mid-august and did not encounter any customer issue since.
In particular it fixed the issue on a customer's Oppo phone.

philips77, 2021-10-08:
The scan filter was added, so that scanning was possible in background. Without fillers, scan doesn't return results. I'll release an updated version of this lib soon and will look into this issue.

> Is there any reason this app doesn't use the latest DFU library version? Reason I ask is one version considered more stable that the most recent release, or maybe its not having enough time to update all the apps utilizing this library.

We're working on a new version, written in kotlin. Perhaps we'll extract dfu from nRF Toolbox and move it to a standalone app, eg. a sample app for dfu project, to make releasing own apps easier. But yes, we just don't have resources to develop new things and maintain all older apps :(

QuentinFarizon, 2022-01-17:
@philips77 Any news about this ? Does the 1.12.0 fixes it ?

philips77, 2022-01-19:
Sorry, I didn't have time to look into this yet. All the GATT errors are originiating in the Android system or are coming from the Bluetooth controller, so there's not much we can fix in the library. As a general rule each next version should have less bugs of have more featuers than the older one, so I would recommend upgrading. 

You may find the change log here: https://github.com/NordicSemiconductor/Android-DFU-Library/releases

QuentinFarizon, 2022-01-19:
Well, as mentionned before in this issue, I was using 1.11.1 in production for a few months, and several customers had this issue.
I rollbacked to 1.11.0, and it fixed the issue for those customers.

So this is definitely an issue of this library introduced in 1.11.1. You said it yourself : "Looks like the issue is related to how the phone filters scan packets."

So my question still stands : is this corrected in the 1.12.0 ?

philips77, 2022-01-20:
The change you're talking about was introduced in https://github.com/NordicSemiconductor/Android-DFU-Library/commit/72c01758f68a1462113e8485289f5a13c1895c2b (apart from changing tabs to spaces, which makes it unreadable...).
The change is still present in the latest version. Also, version 1.12+ target API 31 and your app would also need to target this Android API to use it.
Could you try replacing the filters with device address and incremented device address with a single, but empty one?
```java
scanner.startScan(Collections.singletonList(new ScanFilter.Builder().build()), settings, this);
```
The device addresses are still being checked later in the callback, but perhaps an empty filter can solve both your issues and with scanning in background.

If you target API 30 or lower, I can release a 1.11.2 version with requested change, but you may also use 1.11.0 if it works for you.

QuentinFarizon, 2022-01-20:
> The change you're talking about was introduced in 72c0175

It's difficult to test because its happening only on specific devices (1%
of our customers).

QuentinFarizon, 2023-01-02:
Hello @philips77 any new on this issue ? I'd like to update to latest version of the library, but if the original issue has not been fixed some customers will again encounter the same issues

philips77, 2023-01-03:
I'm still in Christmas break until next week. As far as I remember, nothing was changed so far with filters. I'll look into it when I'm back.

QuentinFarizon, 2023-08-22:
@philips77 We have updated, and as expected it started to break for some customers, with error GATT error (Xiaomi Redmi Note 9 Pro)

I don't understand how this issue can have stayed open for 2 years with such a clear bug report.

With [1.11.0 being unpublished](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/392), we are now stuck with broken releases and can't reliably provide updates to our customers

QuentinFarizon, 2023-08-22:
If you don't want to fix the core issue, maybe just ignoring the filters for Redmi phones here would be a wordaround : https://github.com/NordicSemiconductor/Android-DFU-Library/commit/72c01758f68a1462113e8485289f5a13c1895c2b#diff-da467c50d8806bbe969e81aba215e9b1fc4b34735e13aa46f665fb7ff565aaf6R95

Linked issues with Redmi phones: 
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/385
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/387
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/384
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/381
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/376

QuentinFarizon, 2023-09-21:
I think passing an empty filter (instead of no filter) will fix the issue of scanning with screen off :

Replacing lines 99 to 111 https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/lib/dfu/src/main/java/no/nordicsemi/android/dfu/internal/scanner/BootloaderScannerLollipop.java#L99-L111 by : 
```
scanner.startScan(new ScanFilter.Builder().build(), settings, this);
```

See https://github.com/don/cordova-plugin-ble-central/pull/989/files#diff-b2f24d93764e23dc03ea7b651077b6a2535e2125158fe53dfb827b6769b999abR1259-R1261

PR #403 (2023-10-03): Adding an empty scan filter
This PR fixes #303 and perhaps #387, #384, #376.
closed 2023-10-03 by philips77