nordicsemi/Android-DFU-Library #307: Duplicate class android support library.

bingerz, 2021-07-20:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 1.9.1]

**Device information (please complete the following information):**
 - Device: [e.g. Samsung S10]
 - OS: [e.g. Android 9]

**Describe the bug**
A clear and concise description of what the bug is.

Our company's customers use Android-DFU-Library and encounter the following error:
Duplicate class android.support.v4.app.INotificationSideChannel found in modules core-1.5.0-alpha01-runtime.jar (androidx.core:core:1.5.0-alpha01) and support-v4-23.4.0-runtime.jar (com.android.support:support-v4:23.4.0)

Android IDE: Eclipse;
Their project does not enable AndroidX, and the project relies on com.android.support:support-v4:23.4.0. In this case, is there any good way to avoid this conflict without changing the client project’s dependency library? 

**Logs**
Add any other context or logs describing the problem here.

philips77, 2021-07-21:
Latest version of this library not migrated to AndroidX is 1.8.1.

bingerz, 2021-07-21:
> Latest version of this library not migrated to AndroidX is 1.8.1.

Thank you.

closed 2021-07-22 by philips77