nordicsemi/Android-DFU-Library #311: Error "Could not find no.nordicsemi.android:dfu:1.11.0" when building cordova-android@10 app

jacywang, 2021-08-29:
** Could not find no.nordicsemi.android:dfu:1.11.0 **
Error "Could not find no.nordicsemi.android:dfu:1.11.0" showed up in Ionic Cordova Android build. 
I'm not sure if this is the right place to ask. If not, it would be great if someone can point me in the right direction. Really appreciate it. 

I have been using [cordova-plugin-dfu-update](https://github.com/EinfachHans/cordova-plugin-dfu-update) and it worked well with cordova-android@9.0. But once I upgraded to [cordova-android@10.0](https://cordova.apache.org/announcements/2021/07/20/cordova-android-10.0.0.html), the error `Could not find no.nordicsemi.android:dfu:1.11.0` showed up. cordova-android@9.0 is using Gradle 6.5 and cordova-android@10.0 is using Gradle 7.2. 
I'm wondering if this library is not uploaded to certain places, such as maven apache shown in the error logs below. 

**Logs**
> Task :app:checkDebugAarMetadata FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find no.nordicsemi.android:dfu:1.11.0.
     Searched in the following locations:
       - **https://dl.google.com/dl/android/maven2/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - https://repo.maven.apache.org/maven2/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom**
     Required by:
         project :app

philips77, 2021-08-29:
Did you include
```
mavenCentral()
```
in you main build.gradle file?

philips77, 2021-08-29:
Also, the latest version is 1.11.1.

jacywang, 2021-08-30:
> Did you include
> 
> ```
> mavenCentral()
> ```
> 
> in you main build.gradle file?

Yes, `mavenCentral` is included. Upgrade from 1.11.0 to 1.11.1 fixed the issue. Thanks! Really appreciate it.
```
ext.repos = {
    google()
    mavenCentral()
}
```

closed 2021-08-30 by jacywang

QuentinFarizon, 2022-01-17:
The issue is that 1.11.0 is not available on mavenCentral anymore, which is an issue.

I can't upgrade on 1.11.1 because of https://github.com/NordicSemiconductor/Android-DFU-Library/issues/303

@philips77 Could you push the 1.11.0 on mavenCentral again ? Thanks !