nordicsemi/Android-DFU-Library #321: Fatal Exception: java.lang.IllegalStateException - Not allowed to start service Intent (app is in background)

ReneFerrari, 2022-01-25:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 31
 - Library version: 1.12.0

**Device information (please complete the following information):**
 - Device: Any device running Android 8 or higher

**Describe the bug**
From time to time we are facing this crash (Android 8 and up due to changes with services). Basically the App is in background and tries to start a service. 

The affected LOC is: [no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent (DfuBaseService.java:1413)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1413)

I think changing it to:

```
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
    context.startForegroundService(intent);
} else {
    context.startService(intent);
}
```
would solve it. 

Is there anything speaking against it?

junebeans, 2023-02-06:
To add a note - we need to add permission `FOREGROUND_SERVICE` on the consumer to start the foreground service. Here are the additional reference https://developer.android.com/guide/components/foreground-services#request-foreground-service-permissions. It would be nice if we have permission added in the library 👀 

Ran this on v2.3.0 at android 13.

philips77, 2023-03-23:
> So startForegroundService won't work there, rather WorkManager should be used

Indeed, `WorkManager` should be used. The library was created long before the first WorkManager existed in a way that it would have to be rewritten separating transport and runner.

You may also disable starting service as foreground at all by setting this to false:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/9123f8b68e65fb45e42146219724ab0c88c17ba3/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L152