nordicsemi/Android-DFU-Library #324: Unable to build the project

ap8866, 2022-03-07:
Hello,

I tried checking out the main and develop branch of the project. Still getting these errors, please check the attached image.

Any help is really appriciated.
[image]

ap8866, 2022-03-08:
Thank you so much for your response!

My dependencies looks like:
dependencies {
    implementation 'androidx.core:core:1.7.0'
    implementation 'androidx.localbroadcastmanager:localbroadcastmanager:1.0.0'
    implementation 'androidx.annotation:annotation:1.2.0'
    implementation 'no.nordicsemi.android:dfu:1.12.1-beta01'
    implementation 'com.google.code.gson:gson:2.8.8'
}

I also get a different error in the main branch, please have a look at the attached image
[image]
.

philips77, 2022-04-06:
Hi, did you try after latest updates?

closed 2022-04-08 by philips77