nordicsemi/Android-DFU-Library #327: .zip fails on Android 12

MikeInMinnesota, 2022-03-21:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]  15.3
 - Bonding used: [yes/no]  no
 - Library version: [e.g. 1.9.1] 1.10.4

**Device information (please complete the following information):**
 - Device: [e.g. Samsung S10]  Pixel 3XL
 - OS: [e.g. Android 9] Android 12

**Describe the bug**
A clear and concise description of what the bug is.

Illegal operation due to not having file Uri.

A .zip was selected as the update source and an exception occurs.  It looks like the problem is that the Uri was a "context" type instead of a "file" type resource.  This appears to be a change from Android 11 to Android 12.

E/DFU_FILE_MANAGER: Error during creation file from uri.
    java.lang.IllegalArgumentException: Uri lacks 'file' scheme: content://com.android.providers.downloads.documents/document/msf%3A2019
        at androidx.core.net.UriKt.toFile(Uri.kt:43)
        at no.nordicsemi.dfu.profile.main.data.DFUFileManager.createFromFile(DFUFileManager.kt:33)
        at no.nordicsemi.dfu.profile.main.data.DFUFileManager.createFile(DFUFileManager.kt:20)
        at no.nordicsemi.dfu.profile.main.repository.DFURepository.setZipFile(DFURepository.kt:29)
        at no.nordicsemi.dfu.profile.main.viewmodel.DFUViewModel.onZipFileSelected(DFUViewModel.kt:146)
        at no.nordicsemi.dfu.profile.main.viewmodel.DFUViewModel.onEvent(DFUViewModel.kt:132)
        at no.nordicsemi.dfu.profile.main.view.DFUScreenKt$DFUScreen$onEvent$1.invoke(DFUScreen.kt:28)
        at no.nordicsemi.dfu.profile.main.view.DFUScreenKt$DFUScreen$onEvent$1.invoke(DFUScreen.kt:28)
        at no.nordicsemi.dfu.profile.main.view.DFUSelectFileViewKt$DFUNotSelectedFileView$launcher$1$1.invoke(DFUSelectFileView.kt:48)
        at no.nordicsemi.dfu.profile.main.view.DFUSelectFileViewKt$DFUNotSelectedFileView$launcher$1$1.invoke(DFUSelectFileView.kt:47)
        at androidx.activity.compose.ActivityResultRegistryKt$rememberLauncherForActivityResult$1$1.onActivityResult(ActivityResultRegistry.kt:106)
        at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:409)
        at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:366)
        at androidx.activity.ComponentActivity.onActivityResult(ComponentActivity.java:712)
        at android.app.Activity.dispatchActivityResult(Activity.java:8382)
        at android.app.ActivityThread.deliverResults(ActivityThread.java:5297)
        at android.app.ActivityThread.handleSendResult(ActivityThread.java:5343)
        at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2214)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7842)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

**Logs**
Add any other context or logs describing the problem here.

sylwester-zielinski, 2022-03-21:
Hello,

Thank you for reporting the issue.

I've checked the code and this exception is not a FATAL exception. Actually loading file has two scenarios.
First, a file has "file" scheme and can be directly opened, because operation happens in try catch block so an exception is printed.
If the file lacks "file" scheme then, in catch block, conentResolver is used to open the file.

So my question is: if the file was successfully loaded and displayed in UI? I will remove logging this exception as I can see it is a little confusing.

MikeInMinnesota, 2022-03-21:
Yes, you are correct that it isn’t fatal to execution.  And you are also correct that there are 2 attempts, the first as a file and then in the catch block.  The first attempt fails and the 2nd attempt is what creates the exception.

No, the file isn’t successfully loaded.

The result of the exception and not loading isn’t really felt until later, when the zipfile object is used.  The DFU then fails at that point and the target device disconnects.

I believe the library is used in the nRF Toolbox on Android, because we also see the same DFU failure and disconnect behavior in the DFU utility.

Regards,
Mike

philips77, 2022-03-24:
Hello @MikeInMinnesota,
Did you create the ZIP file with [*nrfutil*](https://github.com/NordicSemiconductor/pc-nrfutil)? Can you share it, or show what files are inside? Does it have *manifest.json* and a BIN and DAT file, or a pair of them, directly in the root, not in a subfolder?

MikeInMinnesota, 2022-03-25:
Sure.  Attached is a representative .zip.  We use nrfutil for generating the file.  All the files are in the root.  We have successfully used this process for several years with the DFU utility in the Toolbox from both Android and IOS devices.  The target device is a nRF52840 running SDK 15.3 with a BLE bootloader.

The Android side is a Pixel 3XL that was updated to Android 12 last fall.  I don’t think it was used for updating since then, but I think it was able to update devices before that.

I don’t know if I mentioned this, but I was able to use this .zip to update devices through pc-nrfutil and a PCA10040 from a PC.

I’ll check and see what other phones I have laying around and see if they behave differently.

Regards,
Mike

MikeInMinnesota, 2022-03-25:
I tried a Pixel XL with that .zip file and had no problems.  It’s Android 10 with DFU Library version 1.10.3.

Regards,
Mike

philips77, 2022-03-25:
Could you send us the ZIP at mag@nordicsemi.no? Seems like you were replying with an email and attachments don't seem to be supported on GitHub.

sylwester-zielinski, 2022-03-28:
Hello Mike,

Thank you for sending the file.
Can you please send us en error code which is thrown? It would be the best if you can upload the whole logs.

philips77, 2022-04-06:
Hello,
We're trying to understand what exactly is the issue that you're facing. The ZIP file that you've sent us "works". I mean it has been accepted by the library (and DFU app), and the DFU process started. Of course we did get INVALID OBJECT error, as we're testing with a debug key set, not yours, so our DK has rejected the init packet due to not matching signature, but that was expected.
We've tried it on Android 12.

MikeInMinnesota, 2022-04-11:
> Yes, I think you are correct about the file selection itself not being a problem.
>
> Attached are two logs, one from an Android 10 unit that communicates successfully and one from an Android 12 unit that wasn’t able to communicate.
>
> I haven’t looked in any detail except to note it looks like the Android 12 had an almost immediate disconnect from the GATT Server.

Resending because original attachments were too large as .pdf format.

Regards,
Mike

2022-04-11 12:32:53.357 5737-5737/? I/emi.android.df: Late-enabling -Xcheck:jni
2022-04-11 12:32:53.455 5737-5737/? E/emi.android.df: Unknown bits set in runtime_flags: 0x8000
2022-04-11 12:32:53.577 5737-5737/no.nordicsemi.android.dfu W/ActivityThread: Application no.nordicsemi.android.dfu is waiting for the debugger on port 8100...
2022-04-11 12:32:53.578 5737-5737/no.nordicsemi.android.dfu I/System.out: Sending WAIT chunk
2022-04-11 12:32:54.580 5737-5737/no.nordicsemi.android.dfu I/System.out: Debugger has connected
2022-04-11 12:32:54.580 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:54.781 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:54.981 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:55.182 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:55.383 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:55.584 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:55.785 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:55.986 5737-5737/no.nordicsemi.android.dfu I/System.out: waiting for debugger to settle...
2022-04-11 12:32:56.187 5737-5737/no.nordicsemi.android.dfu I/System.out: debugger has settled (1336)
2022-04-11 12:32:58.283 5737-5737/no.nordicsemi.android.dfu I/Choreographer: Skipped 35 frames!  The application may be doing too much work on its main thread.
2022-04-11 12:33:00.103 5737-5737/no.nordicsemi.android.dfu W/emi.android.df: Class androidx.compose.runtime.snapshots.SnapshotStateList failed lock verification and will run slower.
2022-04-11 12:33:00.104 5737-5737/no.nordicsemi.android.dfu W/emi.android.df: Common causes for lock verification issues are non-optimized dex code
2022-04-11 12:33:00.104 5737-5737/no.nordicsemi.android.dfu W/emi.android.df: and incorrect proguard optimizations.
2022-04-11 12:33:00.436 5737-6131/no.nordicsemi.android.dfu I/Adreno: QUALCOMM build                   : 4a00b69, I4e7e888065
    Build Date                       : 04/09/19
    OpenGL ES Shader Compiler Version: EV031.26.06.00
    Local Branch                     : mybranche95ae4c8-d77f-f18d-a9ef-1458d0b52ae8
    Remote Branch                    : quic/gfx-adreno.lnx.1.0
    Remote Branch                    : NONE
    Reconstruct Branch               : NOTHING
2022-04-11 12:33:00.436 5737-6131/no.nordicsemi.android.dfu I/Adreno: Build Config                     : S L 8.0.5 AArch64
2022-04-11 12:33:00.439 5737-6131/no.nordicsemi.android.dfu I/Adreno: PFP: 0x005ff110, ME: 0x005ff066
2022-04-11 12:33:00.485 5737-6131/no.nordicsemi.android.dfu W/Gralloc3: mapper 3.x is not supported
2022-04-11 12:33:00.494 5737-6131/no.nordicsemi.android.dfu I/OpenGLRenderer: Davey! duration=2802ms; Flags=1, IntendedVsync=70240071805, Vsync=70823405115, OldestInputEvent=9223372036854775807, NewestInputEvent=0, HandleInputStart=70833538563, AnimationStart=70833783355, PerformTraversalsStart=70834555126, DrawStart=73007976801, SyncQueued=73032883626, SyncStart=73033189355, IssueDrawCommandsStart=73033285657, SwapBuffers=73039318054, FrameCompleted=73043123262, DequeueBufferDuration=159000, QueueBufferDuration=533000, 
2022-04-11 12:33:00.553 5737-5737/no.nordicsemi.android.dfu I/Choreographer: Skipped 113 frames!  The application may be doing too much work on its main thread.
2022-04-11 12:33:01.237 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
2022-04-11 12:33:01.238 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
2022-04-11 12:33:01.238 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
2022-04-11 12:33:01.239 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->getLong(Ljava/lang/Object;J)J (greylist,core-platform-api, linking, allowed)
2022-04-11 12:33:01.240 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
2022-04-11 12:33:01.244 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->objectFieldOffset(Ljava/lang/reflect/Field;)J (greylist,core-platform-api, linking, allowed)
2022-04-11 12:33:01.245 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2022-04-11 12:33:01.245 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
2022-04-11 12:33:01.245 5737-6325/no.nordicsemi.android.dfu W/emi.android.df: Accessing hidden method Lsun/misc/Unsafe;->putLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
[... cut ...]
2022-04-11 12:38:30.094 15044-15162/no.nordicsemi.android.dfu I/DfuBaseService: Cleaning up...
2022-04-11 12:38:30.095 15044-15162/no.nordicsemi.android.dfu D/BluetoothGatt: cancelOpen() - device: D0:00:00:50:03:69
2022-04-11 12:38:30.097 15044-15162/no.nordicsemi.android.dfu D/BluetoothGatt: close()
2022-04-11 12:38:30.098 15044-15162/no.nordicsemi.android.dfu D/BluetoothGatt: unregisterApp() - mClientIf=11
2022-04-11 12:38:31.010 15044-15044/no.nordicsemi.android.dfu I/DfuBaseService: DFU service destroyed

philips77, 2022-08-08:
The line printing this non-issue has been removed.

closed 2022-08-08 by philips77