nordicsemi/Android-DFU-Library #328: Requires long bootloader timeout

obasille, 2022-03-22:
**Information**
This about the DFU Application 2.0.

**DFU Bootloader version (please complete the following information):**
 - SDK version: SDK 17
 - Bonding used: no
 - Library version: don't know

**Device information (please complete the following information):**
 - Device: Nokia G10
 - OS: Android 11

**Describe the bug**
I'm trying to have shortest possible timeout for the bootloader, so the firmware is quickly stared on device reset.

Using the DFU feature of nRF Toolbox for iOS, I can have a 0.8s timeout. In the app, I scan for devices, reset my device once and the device bootloader advertisement is shown. I select it. Then I start the DFU update. Of course by that time the bootloader has already timed out and started the firmware, but resetting the device does the trick, and the nRF Toolbox app is able to connect to the bootloader and push the update.

The same process works with nRF Toolbox for Android. Unfortunately it requires a timeout of 3s for the bootloader (see this [issue](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/93#issuecomment-375207208) for the reasons why it takes longer on Android).

But I can't get the same behavior with the DFU App 2.0.
If I press start after the bootloader times out, the DFU will just fail even if I restart my device.

I can get DFU to work only I have bootloader timeout long enough so I can select the device and start DFU before the timeout expires.

As a side note, the reason of failure is not displayed.

**Logs**
Sorry, I don't know how to get the app log.

philips77, 2022-04-06:
Hello,

Could you send us your fw, that we could try on nRF52832 or nRF52840 DK?

Indeed, that Android library has some long initial delays, that have been added long time ago, and made the connection more reliable on the phones we had back then. As the Bluetooth LE on modern phones is improved, I bet most of the delays are no longer required. We will try without them and make an option in the initializer to remove it in the initial connection (the lib tried to connect for 2 or 3 times).

philips77, 2022-04-06:
Could you also write what does the DFU app show, instead of an error? Perhaps a screen shot, or behavior description?

obasille, 2022-04-07:
Thank you for looking at this!

Here is a screenshot, the zip file I use for DFU, and the hex image grabbed from the device.

[image]

[firmware_03_14_22_sdk17.zip](https://github.com/NordicSemiconductor/Android-DFU-Library/files/8443016/firmware_03_14_22_sdk17.zip)

[nRF_Connect_Programmer_1649336734273.zip](https://github.com/NordicSemiconductor/Android-DFU-Library/files/8443038/nRF_Connect_Programmer_1649336734273.zip)

The source code for our bootloader:
https://github.com/GameWithPixels/DiceBootloader
(except that I've changed `CUSTOM_BL_DFU_INACTIVITY_TIMEOUT_MS` from 3000 to 1000 in `sdk_config.h`).

And for the firmware:
https://github.com/GameWithPixels/DiceFirmware/

philips77, 2022-08-09:
Hi, could you try #350? All the delays have been removed, but I'm not sure that it didn't break functionality for older devices. After all, the delays have been added at some point.

closed 2022-09-09 by philips77