nordicsemi/Android-DFU-Library #329: Taks: Make the initial delays in connection optional

philips77, 2022-04-06:
The DFU service has some artificial delays, added at some points in the history, e.g.:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/9d953c0fc3e082a94279b2b792749ebbf3f8e528/lib_dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1285-L1290
which were added to improve connection stability. They may not be required on mored phones or latest Android versions.

Task:
- [ ] Remove the `waitFor` callas and check it DFU still works.
- [ ] Test on different phones with older Android versions, compare behavior.
- [ ] If they seem to be needed, make the `waitFor` optional in the DFU initializer (leaving enabled by default for backwards compatibility), or remove them in the initial connection, leaving only for the retries.

philips77, 2022-04-06:
This issue is to solve #328.

obasille, 2022-05-13:
Hi! Thanks for creating this issue, is it something that you think will be look at in the near future?

philips77, 2022-05-13:
Yes, that is the plan. Also, depends on the definition of "near future".

PR #350 (2022-09-09): DFU speed improvement (delays removed)
This PR removes number of `waitFor(... ms)` introduced in some places to support different phones.
Years passed and perhaps the devices where the issues were observed no longer need them. As an effect, this should make the DFU faster. Some delays have been left unchanged.

This PR fixes #329.

This change will not go to 2.1, but I'll release 2.2-beta01 just after 2.1 to allow testing.
closed 2022-09-09 by philips77