nordicsemi/Android-DFU-Library #333: DFU upload fails with unsecure bootloader

Montvydas, 2022-05-05:
**Information**
I am observing an issue with the DFU service whenever working with an unsecure bootloader (aka legacy bootloader such as [Adafruit](https://github.com/adafruit/Adafruit_nRF52_Bootloader)) on nRF52840. I have tried using the SDK as is (although on Flutter using the dedicated Flutter plugin to work with the latest 2.0.2 DFU library) and also the  nRF Connect app. On both occasions using the latest versions of the SDK and the app the uploads would fail, although in slightly different way depending on Android version. On Android 12 (e.g. Pixel 6) the upload starts and would progress by jumping in large steps e.g. would upload 24%, then wait 5 seconds, then 49%, then wait 5 seconds, then 70%... Until it would reach 100%, at which point the upload would stall and would never complete. Using versions below Android 12 (e.g. OnePlus 3T, Android 9) the percetage would progress by systematically incrementing 1% until it would reach 100% but again would stuck. I tried waiting for a very long time however the update would never complete and if I cancel it, the MCU doesn't boot back into firmware..

I also tested the nrf Connect app versions on both of my phones and it seems that the last version that still worked for me on **both** phones was 4.24.3. New versions 4.25.X would have the described problem. I understand that the cause for all this might be that the latest DFU library isn't compatible with the unsecure bootloader, however I decided to reach out to see if this was intentional or was done by accident. The closest similar problem that I could find was the issue #278, however that one also seemed slightly different..

**DFU Bootloader version (please complete the following information):**
 - Library version: 2.0.2

**Device information (please complete the following information):**
 - Device: Pixel 6, OnePlus 3T
 - OS: Android 12, Android 9

philips77, 2022-08-05:
Hi, sorry for the late reply. Do you have any logs?

philips77, 2022-08-05:
The lib and nRF Connect should be backwards compatible with all DFU versions (based on nRF5 SDK, not nRF Connect SDK (NCS)).

philips77, 2022-08-05:
I just tested legacy DFU with different configurations and seems to be working.

closed 2022-08-29 by philips77

Montvydas, 2023-02-12:
Hi @philips77, so the problem is how the Nordic-DFU-Library handles legacy bootloaders. It seems to set the MTU value to the maximum allowed value set within the bootloader, even for a legacy bootloader, where it requires MTU of 23. I figured this by finding that with my bootloader the Nordic-DFU-Library logs would state that it is setting MTU to 247 (`I/DfuImpl ( 9193): MTU changed to: 247`) even though it is a legacy bootloader. I understand a legacy bootloader requires MTU of 23, is that correct? After setting the max MTU value on my bootlaoder to 23 everything was working perfectly though, so that's a win! I wonder though, is there a way to force MTU to be 23 using the Nordic-DFU-Library within Android App? I don't have an option of switching the bootloader on my device, thus wanted to simply set the MTU value to something that is OK.