nordicsemi/Android-DFU-Library #335: ANDROID DfuProgressListenerAdapter changes?(override issue)

techker, 2022-05-11:
**Device information (please complete the following information):**
 - Device: [e.g. Samsung A20]  PIXEL 6PRO
 - OS: [e.g. Android 6]       OS12

**Your question**
Updated from 1.8.2 to 2.02

Dfu class was using 

private val _dfuProgressListener = object : DfuProgressListenerAdapter() {
        override fun onProgressChanged(deviceAddress: String?, percent: Int, speed: Float, avgSpeed: Float, currentPart: Int, partsTotal: Int) {
            super.onProgressChanged(deviceAddress!!, percent, speed, avgSpeed, currentPart, partsTotal)
            _progressObservable.onNext(percent.toDouble())
            Timber.d("_dfuProgressListener - onProgressChanged")
        }

        override fun onDeviceDisconnected(deviceAddress: String?) {
            super.onDeviceDisconnected(deviceAddress!!)
            addLog("_dfuProgressListener", "Device Disconnected")
            Timber.d("_dfuProgressListener - onDeviceDisconnected")
        }

        override fun onDeviceConnecting(deviceAddress: String?) {
            super.onDeviceConnecting(deviceAddress!!)
            addLog("_dfuProgressListener", "Device Connecting")
        }

        override fun onDeviceConnected(deviceAddress: String?) {
            super.onDeviceConnected(deviceAddress!!)
            addLog("_dfuProgressListener", "Device Connected")
        }

        override fun onEnablingDfuMode(deviceAddress: String?) {
            super.onEnablingDfuMode(deviceAddress!!)
            addLog("_dfuProgressListener", "Enabling Dfu Mode")

        }

        override fun onDfuCompleted(deviceAddress: String?) {
            super.onDfuCompleted(deviceAddress!!)
            addLog("_dfuProgressListener", "Dfu Completed")
            disposeDfuServiceController()
            _dfuObservableEmitter.onNext(true)
            _dfuObservableEmitter.onComplete()
            Timber.d("_dfuProgressListener - onDfuCompleted")
        }

        override fun onFirmwareValidating(deviceAddress: String?) {
            super.onFirmwareValidating(deviceAddress!!)
            addLog("_dfuProgressListener", "Validating Firmware")
        }

        override fun onDfuProcessStarted(deviceAddress: String?) {
            super.onDfuProcessStarted(deviceAddress!!)
            _updateStateObservable.onNext(UPLOADING_FIRMWARE)
            addLog("_dfuProgressListener", "DFU Process Started")
        }

        override fun onError(deviceAddress: String?, error: Int, errorType: Int, message: String?) {
            super.onError(deviceAddress!!, error, errorType, message)
            addLog("_dfuProgressListener", "Error - [Message-${message
                    ?: "-EMPTY-"}] [Error:$error] [ErrorType:$errorType]")
            Timber.d("_dfuProgressListener - onError")
            //We wait 2 seconds before handling the error
            Thread.sleep(2000)
            disposeDfuServiceController()
            _dfuObservableEmitter.onError(DfuException(message ?: "", error))
        }

        override fun onDfuAborted(deviceAddress: String?) {
            super.onDfuAborted(deviceAddress!!)
            addLog("_dfuProgressListener", "Dfu Aborted")
        }
    }

We now get an error  -> overrides nothing

What changed?

techker, 2022-05-11:
ok odd seems like when i remove the override fun onDeviceDisconnected(deviceAddress: String?) to override fun onDeviceDisconnected(deviceAddress: String) works..odd

philips77, 2022-05-11:
This commit https://github.com/NordicSemiconductor/Android-DFU-Library/pull/170/commits/ba2814a5d41c121f650a13bab2e68470743e430f added `@NonNull` annotations to the Java code, so the params are no longer nullable.

closed 2022-05-11 by philips77